espressif/arduino-esp32 #213: WifiUDPClient example: Keep receiving error could not send data

mjs513, 2017-02-16:
I tired running the sample WifiUDPclient sketch and I keep receiving the following error message:

> [E][WiFiUdp.cpp:160] endPacket(): could not send data: 22

I also receive a similar error message from a custom sketch that works fine with a ESP8266.  I receive on 4210 and send on 4211.  I listed that code for your reference.

Any help would be appreciated.
Mike

```
/**
 * FreeIMU library serial communication protocol
*/
//These are optional depending on your IMU configuration

#include <ADXL345.h>
#include <HMC58X3.h>
#include <LSM303.h>
#include <LPS.h> 
#include <L3G.h>
#include <ITG3200.h> //note LPS library must come before ITG lib
#include <bma180.h>
#include <MS561101BA.h> //Comment out for APM 2.5
#include <BMP085.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <AK8975.h>
#include <AK8963.h>
//#include <SparkFunLSM9DS1.h>  // Uncomment for LSM9DS1
#include <SFE_LSM9DS0.h>  // Uncomment for LSM9DS0 Chosse one or the othe ST IMUs
#include <BaroSensor.h>
//#include <AP_Baro_MS5611.h>  //Uncomment for APM2.5

//These are mandatory
#include <AP_Math_freeimu.h>
#include <Butter.h>    // Butterworth filter
#include <iCompass.h>
#include <MovingAvarageFilter.h>

#include <Wire.h>
#include <SPI.h>

//#define DEBUG
#include "DebugUtils.h"
#include "FreeIMU.h"
#include "DCM.h"
#include "FilteringScheme.h"
#include "RunningAverage.h"

//Intel Edison, Arduino 101, Arduino Due, Arduino Zero: no eeprom 
#if defined(__SAMD21G18A__) || defined(__SAM3X8E__) || defined(__ARDUINO_ARC__) || defined(__SAMD21G18A__) || defined(ESP32)
  #define HAS_EEPPROM 0
#else
  #include <EEPROM.h>
  #define HAS_EEPPROM 1
#endif

#define M_PI 3.14159
#define BaudRate 115200

float q[4];
int raw_values[11];
float ypr[3]; // yaw pitch roll
char str[128];
float val[14];
float val_array[21]; 

// Set the FreeIMU object and LSM303 Compass
FreeIMU my3IMU = FreeIMU();

//#include <ESP8266WiFi.h>  //use for ESP8266
#include <WiFi.h>       // use for ESP32
//#include <WiFiUdp.h>
#include <elapsedMillis.h>

elapsedMillis sendData1;
#define sendInterval 100

const char* ssid = "xxxxxx";
const char* password = "xxxxxxx";
WiFiUDP Udp;
WiFiUDP UdpS;

unsigned int localUdpPort = 4210;  // local port to listen on
unsigned int localUdpPortS = 4211;
char incomingPacket[255];  // buffer for incoming packets
String payload;

//The command from the PC
char tempCorr;
String cmd;

void setup() {
  Serial.begin(BaudRate);
  Wire.begin();

  //#if HAS_MPU6050()
  //    my3IMU.RESET();
  //#endif
  
  my3IMU.init(true);
  
  Serial.println();
  Serial.printf("Connecting to %s ", ssid);

  // delete old config
  WiFi.disconnect(true);
  
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");
  Udp.begin(localUdpPort);
  UdpS.begin(localUdpPortS);
  Serial.printf("Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPort);
  Serial.printf("Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPortS);
  WiFi.localIP().toString().c_str(), localUdpPort;
  WiFi.localIP().toString().c_str(), localUdpPortS;
  
  // LED
  //pinMode(13, OUTPUT);  //
  pinMode(5, OUTPUT);  //for eps32 thing from sparkfun
}

void loop() {

  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);
	
	cmd = incomingPacket[0];
    if(cmd=="v") {
      sprintf(str, "FreeIMU library by %s, FREQ:%s, LIB_VERSION: %s, IMU: %s", FREEIMU_DEVELOPER, FREEIMU_FREQ, FREEIMU_LIB_VERSION, FREEIMU_ID);
      Serial.print(str);
      Serial.print('\n');
    }
    else if(cmd=="1"){
      my3IMU.init(true);
    }
    else if(cmd=="2"){
      my3IMU.RESET_Q();           
    }
    else if(cmd=="g"){
      my3IMU.initGyros();
      //my3IMU.zeroGyro();      
    }
    else if(cmd=="t"){
      //available opttions temp_corr_on, instability_fix
      my3IMU.setTempCalib(1);   
    }
    else if(cmd=="f"){
      //available opttions temp_corr_on, instability_fix
      my3IMU.initGyros();
      my3IMU.setTempCalib(0);
    }
        payload = "Got it!!";
        int payloadLen = payload.length();
        byte message[payloadLen]; 
  }

	//if(sendData1 > sendInterval){
		float val_array[20] = {0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};
		my3IMU.getQ(q, val);
        val_array[15] = my3IMU.sampleFreq;        
        //my3IMU.getValues(val);       
        val_array[7] = (val[3] * M_PI/180);
        val_array[8] = (val[4] * M_PI/180);
        val_array[9] = (val[5] * M_PI/180);
        val_array[4] = (val[0]);
        val_array[5] = (val[1]);
        val_array[6] = (val[2]);
        val_array[10] = (val[6]);
        val_array[11] = (val[7]);
        val_array[12] = (val[8]);
        val_array[0] = (q[0]);
        val_array[1] = (q[1]);
        val_array[2] = (q[2]);
        val_array[3] = (q[3]);
        //val_array[15] = millis();
        val_array[16] = val[9];
        val_array[18] = val[11];
        val_array[19] = val[12];
        
        #if HAS_PRESS()
           // with baro
           val_array[17] = val[10];
           val_array[13] = (my3IMU.getBaroTemperature());
           val_array[14] = (my3IMU.getBaroPressure());
        #elif HAS_MPU6050()
           val_array[13] = (my3IMU.DTemp/340.) + 35.;
        #elif HAS_MPU9150()  || HAS_MPU9250()
           val_array[13] = ((float) my3IMU.DTemp) / 333.87 + 21.0;
        #elif HAS_LSM9DS0()
            val_array[13] = 21.0 + (float) my3IMU.DTemp/8.; //degrees C
        #elif HAS_ITG3200()
           val_array[13] = my3IMU.rt;
        #elif HAS_CURIE()
           val_array[13] = (my3IMU.DTemp/512.0) + 23.0;
        #endif

		    payload = "";
		    serialPayloadFloatArr(val_array, 20);
		    payload += "\r\n";
  
		    int payloadLen = payload.length();
		    byte message[payloadLen];  
		    payload.getBytes(message,payloadLen);

        UdpS.beginPacket(UdpS.remoteIP(), UdpS.remotePort());
        UdpS.write(message,sizeof(message));
        UdpS.endPacket();
       
        sendData1 = 0;
	//}

}

//==========================================================
void serialPayloadFloatArr(float * arr, int length) {
  for(int i=0; i<length; i++) {
    serialPayloadPrint(arr[i]);
    payload += ",";
  }
}

void serialPayloadPrint(float f) {
  byte * b = (byte *) &f;
  for(int i=0; i<4; i++) {
    
    byte b1 = (b[i] >> 4) & 0x0f;
    byte b2 = (b[i] & 0x0f);
    
    char c1 = (b1 < 10) ? ('0' + b1) : 'A' + b1 - 10;
    char c2 = (b2 < 10) ? ('0' + b2) : 'A' + b2 - 10;
    
    payload += (c1);
    payload += (c2);
  }
}

```

me-no-dev, 2017-02-16:
UdpS.beginPacket(UdpS.remoteIP(), UdpS.remotePort());
should be
UdpS.beginPacket(Udp.remoteIP(), Udp.remotePort());
UdpS did not get the packet. Also you do not need to UdpS.begin(localUdpPortS); since you are not listening :) Try those and report

me-no-dev, 2017-02-16:
And why not just Udp.beginPacket(UdpS.remoteIP(), UdpS.remotePort()); and so on, no need for two UDPs. Does not help you in any way in this case. UDP is not like TCP, there isn't really a connection that is kept ;) just you know where the packet should go and that's it.

mjs513, 2017-02-16:
@me-no-dev I tried what you suggested in your first post and received the following error:

> Connecting to CyberPalin .......... connected
> Now listening at IP 192.168.1.254, UDP port 4210
> Now listening at IP 192.168.1.254, UDP port 4211
> Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x4010fcae  PS      : 0x00060830  A0      : 0x8010fccc  A1      : 0x3ffdd040  
> A2      : 0x3ffc266c  A3      : 0x00000044  A4      : 0x00000000  A5      : 0x0000ff00  
> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
> A10     : 0x00000001  A11     : 0x000000ff  A12     : 0x000000ff  A13     : 0x0000ff00  
> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
> EXCVADDR: 0x00000000  LBEG    : 0x40001699  LEND    : 0x400016aa  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x4010fcae:0x3ffdd040 0x4010fccc:0x3ffdd060 0x400d1446:0x3ffdd080 0x40108945:0x3ffdd1d0
> 
> CPU halted.

me-no-dev, 2017-02-16:
Try this:
```cpp
/**
 * FreeIMU library serial communication protocol
 */
//These are optional depending on your IMU configuration

#include <ADXL345.h>
#include <HMC58X3.h>
#include <LSM303.h>
#include <LPS.h> 
#include <L3G.h>
#include <ITG3200.h> //note LPS library must come before ITG lib
#include <bma180.h>
#include <MS561101BA.h> //Comment out for APM 2.5
#include <BMP085.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <AK8975.h>
#include <AK8963.h>
//#include <SparkFunLSM9DS1.h>  // Uncomment for LSM9DS1
#include <SFE_LSM9DS0.h>  // Uncomment for LSM9DS0 Chosse one or the othe ST IMUs
#include <BaroSensor.h>
//#include <AP_Baro_MS5611.h>  //Uncomment for APM2.5

//These are mandatory
#include <AP_Math_freeimu.h>
#include <Butter.h>    // Butterworth filter
#include <iCompass.h>
#include <MovingAvarageFilter.h>

#include <Wire.h>
#include <SPI.h>

//#define DEBUG
#include "DebugUtils.h"
#include "FreeIMU.h"
#include "DCM.h"
#include "FilteringScheme.h"
#include "RunningAverage.h"

//Intel Edison, Arduino 101, Arduino Due, Arduino Zero: no eeprom 
#if defined(__SAMD21G18A__) || defined(__SAM3X8E__) || defined(__ARDUINO_ARC__) || defined(__SAMD21G18A__) || defined(ESP32)
#define HAS_EEPPROM 0
#else
#include <EEPROM.h>
#define HAS_EEPPROM 1
#endif

#define M_PI 3.14159
#define BaudRate 115200

float q[4];
int raw_values[11];
float ypr[3]; // yaw pitch roll
char str[128];
float val[14];
float val_array[21]; 

// Set the FreeIMU object and LSM303 Compass
FreeIMU my3IMU = FreeIMU();

//#include <ESP8266WiFi.h>  //use for ESP8266
#include <WiFi.h>       // use for ESP32
//#include <WiFiUdp.h>
#include <elapsedMillis.h>

elapsedMillis sendData1;
#define sendInterval 100

[... 139 lines cut ...]
    val_array[14] = (my3IMU.getBaroPressure());
#elif HAS_MPU6050()
    val_array[13] = (my3IMU.DTemp/340.) + 35.;
#elif HAS_MPU9150()  || HAS_MPU9250()
    val_array[13] = ((float) my3IMU.DTemp) / 333.87 + 21.0;
#elif HAS_LSM9DS0()
    val_array[13] = 21.0 + (float) my3IMU.DTemp/8.; //degrees C
#elif HAS_ITG3200()
    val_array[13] = my3IMU.rt;
#elif HAS_CURIE()
    val_array[13] = (my3IMU.DTemp/512.0) + 23.0;
#endif
    
    payload = "";
    serialPayloadFloatArr(val_array, 20);
    payload += "\r\n";
    
    Udp.beginPacket(remoteIp, remotePort);
    Udp.print(payload);
    Udp.endPacket();
    
    sendData1 = 0;
    //}
}
```

mjs513, 2017-02-16:
Ok.  Gave it a try and it worked for a few data transfers and then it gave me about 20 or 30 of the following:

> [E][WiFiUdp.cpp:160] endPacket(): could not send data: 12

and then froze up.   By the way are the following lines really 0:
static IPAddress remoteIp = 0;
static uint16_t remotePort = 0;

or am I suppose to put mine own in.  PS.  if I leave them as it it won't compile complains about the IPaddress.

mjs513, 2017-02-17:
By the way I am using a Sparkfun ESP32 Thing with Arduino IDE 1.8.1 on a windows 10 machine.  I am new to wifi so pardon my ignorance.  I did get the same sketch working on an LinkNode D1 which is an ESP8266 with no problem so I am not sure where the problem lies.

me-no-dev, 2017-02-17:
well you are running on a two core 240 MHz CPU :) and you are sending packet after packet, which leaves little time for the other core to do it's chores. Try putting a delay of 1ms or so at the end of the loop. As for ```static IPAddress remoteIp = 0;``` you can set it to ```static IPAddress remoteIp = IPAddress();``` and it should compile. It should be 0 because the value is filled once a packet is received and you know the address of the other end (because you do not specify it nowhere). ESP8266 is single core and it can not get in a race condition with the other core (where the network actually runs)

mjs513, 2017-02-17:
Thanks for the info on chip and the possible race conditions.  I did the following:
```
static IPAddress remoteIp = IPAddress(); 
static remotePort = 0;
```

and added a delay of 10ms at the end of the loop.  It sends back 3 packets after I transmits from the pc and then I get the error: 

> E (20097) wifi: lmac.c lmacProcessTxRtsError 1670

I also tried to set` remotePort = 4210 `and received the following error after about 4 packets were transmitted:

> E (26161) wifi: lmac.c lmacProcessTxRtsError 1670

mjs513, 2017-02-17:
@me-no-dev I have a question that may be related to this.  what I am reading is accel, gyro and magnetomer readings via i2c from a Adafruit lsm9ds0.  When I do a test of just reading raw data (no wifi) the values are fixed no change ( using the Sparkfun lib for the lsm9ds0).  If I run the same test using a teensy 3.6 reads values no problem.  I am setting the i2c bus to 400khz with the setclock function.

mjs513, 2017-02-18:
I got the sketch to work to a point.  But to do that I had to edit a couple of lines:
```
static IPAddress remoteIp = IPAddress(); 
static remotePort = 4210;
```
and
`Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());`

It does work for quite awhile but i keep getting the 

> [E][WiFiUdp.cpp:160] endPacket(): could not send data: 12

message.  I have a tried delays of 1, 10, 20 and 100ms.  It seems for me 100 works well.  There should be a way to test if the chip is ready to send the message.  If not just wait.  Don't know if it is possible.

me-no-dev, 2017-02-19:
you are probably hitting another bug. Watch for the update that I will push tomorrow with updated libs and give it another go

mjs513, 2017-02-19:
Will check it tomorrow thanks

mjs513, 2017-02-20:
Downloaded the changes that you made and now having a whole different set of errors;

Case 1,  ESP32 Dev module selected:  FAILS
From the serial screen I receive the following:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0008,len:8
> load:0x3fff0010,len:1848
> load:0x40078000,len:6712
> load:0x40080000,len:252
> entry 0x40080034
> E (1053) wifi: wifi_init 1147 ret=4363
> 
> Connecting to CyberPalin /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
> abort() was called at PC 0x4008352b
> Guru Meditation Error: Core  1 panic'ed (abort)
> 
> Backtrace: 0x40008155:0x3ffcf7a0 0x40007d16:0x3ffcf7c0 0x4009721c:0x3ffcf800 0x400f83b4:0x3ffcf850 0x400f214d:0x3ffcf870 0x400d40c8:0x3ffcf8a0 0x400d1234:0x3ffcf930 0x4010a5f2:0x3ffcf970
> 
> CPU halted.
> 

From the ESP Debug tool:
> Decoding 12 results
> 0x40080000: _WindowOverflow4 at ?? line ?
> 0x40080034: _WindowOverflow4 at ?? line ?
> 0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
> 0x4009721c: pp_post at ?? line ?
> 0x400f83b4: ieee80211_ioctl at ?? line ?
> 0x400f214d: esp_wifi_set_config at ?? line ?
> 0x400d40c8: WiFiSTAClass::disconnect(bool) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.cpp line 557
> 0x400d1234: setup at C:\Users\CyberPalin\Documents\Arduino\ESP32SerialFreeimuTest/ESP32SerialFreeimuTest.ino line 98
> 0x4010a5f2: loopTask(void*) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)

Case 2.  Since I am using a Sparkfun ESP32Thing I selected that board. FAILS
From Serial window:

> Connecting to CyberPalin /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
> abort() was called at PC 0x4008352b
> Guru Meditation Error: Core  1 panic'ed (abort)
> 
> Backtrace: 0x40008155:0x3ffcf7a0 0x40007d16:0x3ffcf7c0 0x4009721c:0x3ffcf800 0x400f83b4:0x3ffcf850 0x400f214d:0x3ffcf870 0x400d40c8:0x3ffcf8a0 0x400d1234:0x3ffcf930 0x4010a5f2:0x3ffcf970
> 
> CPU halted.

From Decoder:

> Decoding 18 results
> 0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
> 0x4009721c: pp_post at ?? line ?
> 0x400f83b4: ieee80211_ioctl at ?? line ?
> 0x400f214d: esp_wifi_set_config at ?? line ?
> 0x400d40c8: WiFiSTAClass::disconnect(bool) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.cpp line 557
> 0x400d1234: setup at C:\Users\CyberPalin\Documents\Arduino\ESP32SerialFreeimuTest/ESP32SerialFreeimuTest.ino line 98
> 0x4010a5f2: loopTask(void*) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)
> 0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
> 0x4009721c: pp_post at ?? line ?
> 0x400f83b4: ieee80211_ioctl at ?? line ?
> 0x400f214d: esp_wifi_set_config at ?? line ?
> 0x400d40c8: WiFiSTAClass::disconnect(bool) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.cpp line 557
> 0x400d1234: setup at C:\Users\CyberPalin\Documents\Arduino\ESP32SerialFreeimuTest/ESP32SerialFreeimuTest.ino line 98
> 0x4010a5f2: loopTask(void*) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)

Case 3,  UDPCient Example:  FAILS:
from serial:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0008,len:8
> loáäº°ø³¦¦¦°°±°¬¬¥®º±¸´¸�Š¬ïáäº°ø´°°·¸°°°,len:6712
> load:0x40080000,len:252
> entry 0x40080034
> E (1005) wifi: wifi_init 1147 ret=4363
> Connecting to WiFi network: CyberPalin
> /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
> abort() was called at PC 0x4008352b
> Guru Meditation Error: Core  1 panic'ed (abort)
> 
> Backtrace: 0x40008155:0x3ffcef80 0x40007d16:0x3ffcefa0 0x4009721c:0x3ffcefe0 0x400f2fbc:0x3ffcf030 0x400ecd51:0x3ffcf050 0x400d12fc:0x3ffcf080 0x400d0e7b:0x3ffcf110 0x400d0ec0:0x3ffcf150 0x40104f9a:0x3ffcf170
> 
> CPU halted.

Decoder:

> Decoding 12 results
> 0x40080000: _WindowOverflow4 at ?? line ?
> 0x40080034: _WindowOverflow4 at ?? line ?
> 0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
> 0x4009721c: pp_post at ?? line ?
> 0x400f2fbc: ieee80211_ioctl at ?? line ?
> 0x400ecd51: esp_wifi_set_config at ?? line ?
> 0x400d12fc: WiFiSTAClass::disconnect(bool) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.cpp line 557
> 0x400d0e7b: connectToWiFi(char const*, char const*) at C:\Users\CYBERP~1\AppData\Local\Temp\arduino_modified_sketch_483612/WiFiUDPClient.ino line 50
> 0x400d0ec0: setup at C:\Users\CYBERP~1\AppData\Local\Temp\arduino_modified_sketch_483612/WiFiUDPClient.ino line 29
> 0x40104f9a: loopTask(void*) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)
> 

This is leading me to believe that there is some bug in the library pushes (9:13 am, Monday)
I am also constantly getting the following messages, I just reload the sketch and it will load fine:

> A fatal error occurred: Timed out waiting for packet header
> A fatal error occurred: Timed out waiting for packet header

me-no-dev, 2017-02-20:
all errors point to the same thing. I will do a test case and report to idf, but for you, just remove the first call to WiFi.begin(); and then to AiFi.disconnect(true); and your wifi will work as expected. Calling WiFi.begin(ssid,pass) already overwrites the default setting.

igrr, 2017-02-20:
Unrelated note... I think the exception decoder grabs a few values from the bootloader output (like 0x40080000).

me-no-dev, 2017-02-20:
@igrr great for noticing that! Will adjust the plugin in the next update.

mjs513, 2017-02-20:
Check the sketches and there is no WiFi.begin() only WiFi.begin(ssid,pass).  I commented out the lines for WiFi.disconnect(true) and am still receiving the same errors.  Used the UDP example.

mjs513, 2017-02-20:
Ok.  Decided to run the WifiScan example and am getting similar errors.

> Decoding 13 results
> 0x40080000: _WindowOverflow4 at ?? line ?
> 0x40080034: _WindowOverflow4 at ?? line ?
> 0x4008346b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
> 0x4009710c: pp_post at ?? line ?
> 0x400edd88: ieee80211_ioctl at ?? line ?
> 0x400e7947: esp_wifi_start at ?? line ?
> 0x400d0ec8: WiFiGenericClass::getMode() at c:\local programs\arduino-1.8.1\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 923
> :  (inlined by) WiFiGenericClass::getMode() at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.cpp line 259
> 0x400d0ee2: WiFiGenericClass::mode(wifi_mode_t) at c:\local programs\arduino-1.8.1\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 923
> 0x400d0c7a: setup at C:\Users\CYBERP~1\AppData\Local\Temp\arduino_modified_sketch_467225/WiFiScan.ino line 15
> 0x400ffb96: loopTask(void*) at C:\Local Programs\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)

Same type of error.

me-no-dev, 2017-02-20:
this is my output of the WiFi scan sketch (no modifications)
```text
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6712
load:0x40080000,len:252
entry 0x40080034
Setup done
scan start
scan done
5 networks found
1: nbis-test (-41)*
2: your-ssid (-47)*
3: HP-Print-EE-ficeto (-50) 
4: hristian19 (-78)*
5: www.networx.bg (-79) 
```
Also see if it might be connected to [this](https://github.com/espressif/arduino-esp32/issues/207)

mjs513, 2017-02-20:
Yep erasing the flash seemed to do the trick to get rid of the errors I posted earlier today.

me-no-dev, 2017-02-20:
everything working now?

mjs513, 2017-02-20:
Know you don't want to hear this but not really.  Since er AZ see false I can connect but I keep getting different errors.  I tried a couple of different sketches. I have to lay ou a couple of cas es S for you but decided to take a break .. am getting a headache. Smile. Will get back to it tonight.

mjs513, 2017-02-20:
Ok Here it goes.  I had written a simple test sketch based on the example to send and receive from Network like the example sketch:

```
#include <WiFi.h>
#include <WiFiUdp.h>
const char* ssid = "xxx";
const char* password = "xxxx";
WiFiUDP Udp;
unsigned int localUdpPort = 4210;  // local port to listen on
char incomingPacket[255];  // buffer for incoming packets
char  replyPacekt[] = "Hi there! Got the message :-)";  // a reply string to send back
void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");
Udp.begin(localUdpPort);
  Serial.printf("Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPort);
}
void loop()
{
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);
// send back a reply, to the IP address and port we got the packet from
    Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
    Udp.print(replyPacekt);
    Udp.endPacket();
  }
}
```
Loads fine and connects right away.  I send one message and it acknowledges receipt.  When I send another message it crashed and hangs.  This is the only thing that is returned on the second attempt:
Received 8 bytes from 192.168.1.5, port 55056

> Gr eiainErr oe 0 ai'd(Itrutwttmoto P0)
> eitrdm:
> C   :0406 P    x0003 A    x80f7 A    x39 
> 2   :03f4 A    x3b0 A    x4068 A    x0000 
> 6   :00000 A    x0000 A    x806 A    x36 
> 1   :0ffffffff A1   xffffffff A2   x0000 A3   x003 
> 1   :00001 A5   x0001 SR   x0001 ECAS x0000 
> XVDR:00000 LE   x0000 LN   x0000 LON  x0000 
> 
> Bctae0406:xffcc70 x0dc8:xffcc7
> 
> P atd
> 

If I hit the reset never wants to reconnect. Have to unplug.  When I do it again same thing happens.  Next test is with the full sketch that we worked on earlier that you modified:

> 1. When it starts it returns gibberish and then says it connecting to CyberPalin but never connects.  Have to unplug and plug back into serial to get it to connect.
> 2. When it does connect:
> E (12615) wifi: lmac.c lmacProcessTxRtsError 1676
> 
> Task watchdog got triggered. The following tasks did not feed the watchdog in time:
> Tasks currently running:
> CPU 0: ppT
> CPU 1: IDLE
> Task watchdog got triggered. The following tasks did not feed the watchdog in time:
> Tasks currently running:
> CPU 0: ppT
> CPU 1: IDLE
> Task watchdog got triggered. The following tasks did not feed the watchdog in time:
> Tasks currently running:
> CPU 0: ppT
> CPU 1: IDLE
> Task watchdog got triggered. The following tasks did not feed the watchdog in time:
> Tasks currently running:
> CPU 0: ppT
> CPU 1: IDLE

Then error message keeps repeating.   If I reset it will reconnect but just hangs with no error messages (no data is received).  If I power off/power on it will respond with an acknowledge but hangs with no data returned.  I checked to make sure that connections are good and the ESP32 recognizes the I2C addresses.  When I remove the setClock(400000L) call same error happens.

Sorry to cause you such problems.

Thanks for all your help and patience.

mjs513, 2017-02-20:
Just one more bit of info.  I tried running the same sketch on a 8266 and it crashed as well.  I have a open issue over on that side as well: [https://github.com/esp8266/Arduino/issues/2983].  May be related - not sure why when I did have it working before.  Beginning to wonder if it an i2c issue.

me-no-dev, 2017-02-22:
copy-pasted the sketch above, bombarded with packets, no issue at all. This is something on your end :) not necessarily the ESP. Could be your network/AP. I tested with 3 devices ;)

closed 2017-02-22 by me-no-dev

mjs513, 2017-02-23:
Do you have any suggestions on how to figure it what the problem is.  By the way what board are you using.

mjs513, 2017-02-26:
@me-no-dev.  Just wanted to let you know I picked up a different board (ESP32 dev board by Onehorse on Tindie, it uses dout and DFU) and uploaded the same sketch and it works fine with no problem.  I tried it again on the ESP32 Thing by Sparkfun and it hangs after it sends back about 10-20 packets.

ImanolGo, 2017-06-18:
I think I might have the exact same problem as @me-no-dev described. I'll try to upload the code to another ESP32 dev board and see.

miomir1981, 2017-12-06:
I apologize for posting on this old and closed issue, but...
I spent three days trying to figure out what is wrong with UDP on ESP32. I was working with Sparkfun esp32 thing board and UDP was misbehaving. After few days I decided to test with other ESP32 board, and guess what? It works like a charm :) It is probably some issue with Sparkfun board or ESP32 chips they are using on their board (probably early silicon rev.). Anyways, try to avoid Spurkfun board if you intend to use UDP in any form.

mjs513, 2017-12-07:
Same here before I switched boards. Pretty much gave up on the sparkfun board.

me-no-dev, 2017-12-11:
sparkfun use 26MHz crystals and all other boards use 40MHz... maybe that is the issue (somehow)