espressif/arduino-esp32 #214: Hardware Timers

williamhemmingsen, 2017-02-17:
Are there any examples of hardware timers for Arduino IDE?  I have used ESP32 hw timers a lot, but can't make since of them in the ESP32.

I need to sample the adc a a specified rate.  I would like to attachinterrupt to and ISR.  Here is what I'm doing in the ESP8266.  How do I do this in the ESP32?

void timer_init(void) {
 timer1_isr_init();
  timer1_attachInterrupt(sample_isr);
  timer1_enable(TIM_DIV1, TIM_EDGE, TIM_LOOP);
  timer1_write(8333); //9600 samples/s
}

void ICACHE_RAM_ATTR sample_isr() {
...do some stuff
}

Thanks!

me-no-dev, 2017-02-17:
```cpp
hw_timer_t * timer = NULL;

void onTimer(){
    Serial.print("timer ");
    Serial.println(millis());
}

// start/stop the timer
void toggleTimer(){
    if(timer){
        timerEnd(timer);
        timer = NULL;
    } else {
        timer = timerBegin(3, 80, 1);//div 80
        timerAttachInterrupt(timer, &onTimer, 1);
        timerAlarmWrite(timer, 1000000, true);//1000ms
        timerAlarmEnable(timer);
    }
}
```

arcao, 2017-02-17:
There is example which call `onTimer` every second:
```cpp
hw_timer_t * timer = NULL;

void onTimer(){
  static unsigned int counter = 1;
  
  Serial.print("onTimer ");
  Serial.print(counter);
  Serial.print(" at ");
  Serial.print(millis());
  Serial.println(" ms");

  if (counter == 10)
    endTimer();

  counter++;
}

void startTimer() {
  timer = timerBegin(0, 80, true); // timer_id = 0; divider=80; countUp = true;
  timerAttachInterrupt(timer, &onTimer, true); // edge = true
  timerAlarmWrite(timer, 1000000, true);  //1000 ms
  timerAlarmEnable(timer);
}

void endTimer() {
  timerEnd(timer);
  timer = NULL; 
}

void setup() {
  Serial.begin(115200);
  startTimer();
}

void loop() {}
```

me-no-dev, 2017-02-17:
@arcao you asked me as I was looking at the issue :D

williamhemmingsen, 2017-02-17:
Thanks Guys, That was perfect.  Thanks again for the quick response

me-no-dev, 2017-02-17:
:)

closed 2017-02-17 by me-no-dev

arcao, 2017-02-17:
BTW I added example: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/RepeatTimer/RepeatTimer.ino

williamhemmingsen, 2017-02-23:
Thanks!  I used your timer example successfully to implement my project.
Thanks again

GitHub - williamhemmingsen/WSPR-ESP32: WSPR implementation using the ESP32 and Si5351

  
|  
|   
|   
|   |    |

   |

  |
|  
|    |  
GitHub - williamhemmingsen/WSPR-ESP32: WSPR implementation using the ESP32 ...
 WSPR-ESP32 - WSPR implementation using the ESP32 and Si5351  |   |

  |

  |

tobozo, 2017-02-23:
Your comment is exposing your email address

me-no-dev, 2017-02-23:
not anymore :)

maxpromer, 2017-05-26:
How to restart timer ? I test by
```
hw_timer_t * timer = NULL;

void onTimer(){
    Serial.print("timer ");
    Serial.println(millis());
}

// start/stop the timer
void toggleTimer(){
    if(timer){
        timerEnd(timer);
        timer = NULL;
    } else {
        timer = timerBegin(3, 80, 1);//div 80
        timerAttachInterrupt(timer, &onTimer, 1);
        timerAlarmWrite(timer, 100000, true);//1000ms
        timerAlarmEnable(timer);
    }
}

void setup() {
  Serial.begin(115200);
}

void loop() {
  toggleTimer();
  delay(1000);
}
```
timer not new start.

lrmoreno007, 2017-05-29:
```
hw_timer_t * timer = NULL;

void onTimer(){
	Serial.print("Alarm: make something at ");
	Serial.print(millis()/1000);
	Serial.println("s");
}

void setup() {
	Serial.begin(115200);
	timer = timerBegin(1, 80, 1);//Timer number 1, 80 is the CPU frequency, 1 counter up
	timerAttachInterrupt(timer, &onTimer, 1);
	timerAlarmWrite(timer, 1000000, true); //1 second and autoreload enabled
	timerAlarmEnable(timer);
}

void loop() {

}
```

XingyumLee, 2019-05-27:
> ```c++
> hw_timer_t * timer = NULL;
> 
> void onTimer(){
>     Serial.print("timer ");
>     Serial.println(millis());
> }
> 
> // start/stop the timer
> void toggleTimer(){
>     if(timer){
>         timerEnd(timer);
>         timer = NULL;
>     } else {
>         timer = timerBegin(3, 80, 1);//div 80
>         timerAttachInterrupt(timer, &onTimer, 1);
>         timerAlarmWrite(timer, 1000000, true);//1000ms
>         timerAlarmEnable(timer);
>     }
> }
> ```
where to download the lib

XingyumLee, 2019-05-28:
> **Edit:** @me-no-dev was faster :)
> 
> There is example which call `onTimer` every second:
> 
> ```c++
> hw_timer_t * timer = NULL;
> 
> void onTimer(){
>   static unsigned int counter = 1;
>   
>   Serial.print("onTimer ");
>   Serial.print(counter);
>   Serial.print(" at ");
>   Serial.print(millis());
>   Serial.println(" ms");
> 
>   if (counter == 10)
>     endTimer();
> 
>   counter++;
> }
> 
> void startTimer() {
>   timer = timerBegin(0, 80, true); // timer_id = 0; divider=80; countUp = true;
>   timerAttachInterrupt(timer, &onTimer, true); // edge = true
>   timerAlarmWrite(timer, 1000000, true);  //1000 ms
>   timerAlarmEnable(timer);
> }
> 
> void endTimer() {
>   timerEnd(timer);
>   timer = NULL; 
> }
> 
> void setup() {
>   Serial.begin(115200);
>   startTimer();
> }
> 
> void loop() {}
> ```
so if i end the timer,how can i restart it?

XingyumLee, 2019-05-28:
> How to restart timer ? I test by
>
> ```
> hw_timer_t * timer = NULL;
>
> void onTimer(){
>     Serial.print("timer ");
>     Serial.println(millis());
> }
>
> // start/stop the timer
> void toggleTimer(){ [...]

is seems that it just relay for 1 seconds but dont know whether the timer is stopped or not