espressif/arduino-esp32 #218: Python error when compiling for the ESP32 from the Arduino IDE

dibaggioj, 2017-02-17:
A python error when calling `dlopen` is occurring when attempting to compile any Arduino sketches (e.g., Blink) for the Sparkfun ESP32 The Thing. Please let me know if you have any thoughts for how to fix this, thanks.

The verbose log is below:
```Arduino: 1.8.1 (Mac OS X), Board: "SparkFun ESP32 Thing, 80MHz, 115200"

/Users/johndibaggio/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johndibaggio/ESP32/hardware -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/tools-builder -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Users/johndibaggio/Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johndibaggio/ESP32/libraries -fqbn=espressif:esp32:esp32thing:FlashFreq=80,UploadSpeed=115200 -ide-version=10801 -build-path /var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/johndibaggio/ESP32/sketches/ESP32-Blink/ESP32-Blink.ino
/Users/johndibaggio/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johndibaggio/ESP32/hardware -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/tools-builder -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Users/johndibaggio/Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johndibaggio/ESP32/libraries -fqbn=espressif:esp32:esp32thing:FlashFreq=80,UploadSpeed=115200 -ide-version=10801 -build-path /var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/johndibaggio/ESP32/sketches/ESP32-Blink/ESP32-Blink.ino
Using board 'esp32thing' from platform in folder: /Users/johndibaggio/ESP32/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/johndibaggio/ESP32/hardware/espressif/esp32
Detecting libraries used...
"/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_THING -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/cores/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/variants/esp32thing" "/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649/sketch/ESP32-Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_THING -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/cores/esp32" "-I/Us [... cut ...] vm1cjm0000gr/T/arduino_build_247649/ESP32-Blink.ino.elf"
Error loading Python lib '/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python': dlopen(/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python, 10): no suitable image found.  Did find:
	/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python: code signature invalid for '/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python'

exit status 255
Error compiling for board SparkFun ESP32 Thing.```

me-no-dev, 2017-02-19:
I have a clean Mac here that I will setup tomorrow and check if it will be OK, but you are so far the first OSX user to report such problem. What version is your OS? Have you installed some other python?

dibaggioj, 2017-02-20:
@me-no-dev thanks, and I'm on Mac OS 10.10.5 with Python 2.7.10 installed.

dibaggioj, 2017-02-20:
@me-no-dev I realized that there was already a PR (https://github.com/espressif/arduino-esp32/pull/200) open to fix the same error I'm having. It seems there's an issue with the esptool binary, but the esptool.py script works

me-no-dev, 2017-02-23:
I want to find the best solution. I do not want to require the python version be default because it requires packages not installed by default. Best idea is to filter the OS version maybe in a script and that to call python or binary. Anyway, thanks for confirming it's a 10.10 issue.

dibaggioj, 2017-02-23:
Yes, filtering by OS version sounds like an acceptable solution. And in the setup instructions, probably mention the necessary python packages to manually install on MacOSX 10.10. I at least needed to install pyserial, but I might have already had other packages that are required. Thanks for the work on this, @me-no-dev.

me-no-dev, 2017-02-28:
hey apple fellas :) here is a ZIP with esptool compiled on OSX 10.9. Please test and let me know if it works.
[esptool-mac-10.9.zip](https://github.com/espressif/arduino-esp32/files/809141/esptool-mac-10.9.zip)

JPB1JPB1, 2017-03-23:
For your information, i have recompiled yesterday for the ESP32 this wifibluetoothswitch.ino and other example under Arduino 1.6.13 without any problem, ok under PC with W10, not MAC Os..

The hardware esp32 was set in the hardware directory for esp32-arduino github... 
It looks ok for other wifi example too...  Python is 2.7.12 on my PC..

If this can help you but for me,  ESP32 et Arduino is better now.. from here : https://github.com/espressif/arduino-esp32 and follow under Windows instructions with success..
Sorry i can not try under MAC os..

Best Regards
JP

me-no-dev, 2017-03-23:
@JPB1JPB1 please do not spam the issues with your BT question!

dibaggioj, 2017-03-23:
@me-no-dev I thought I had replied on this thread previously, but the esptool binary in https://github.com/espressif/arduino-esp32/files/809141/esptool-mac-10.9.zip does work on Mac OS X 10.10.5.

lazyzero, 2017-03-29:
I am also on OSX 10.10.5 and had the same issue as @dibaggioj after replacing the esptool with the one for OSX 10.9 my sketch compiled without issue.

me-no-dev, 2017-04-03:
Ok guys, the latest update includes the latest esptool all fixed for older OS X :) Enjoy

closed 2017-04-03 by me-no-dev