espressif/arduino-esp32 #219: Repeated crash at uxListRemove when using WiFiClientSecure and PubSubClient

forthlightning, 2017-02-17:
I have a sketch here: [https://github.com/forthlightning/esp32_tls_test/](url) In which I publish mqtt messages to azure cloud.

Messages are successfully published for some time, on the order of minutes. Eventually the system crashes and has a backtrace like this:

```
0x400854b3: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 212
0x40084990: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4399
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3019
0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400ebe91: sys_mbox_post at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/freertos/sys_arch.c line 562
0x400e39a5: tcpip_callback_with_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 473
0x400e3398: lwip_getsockopt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c line 3235
0x400e3764: lwip_getsockopt_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c line 3235
0x400ff155: mbedtls_net_errno at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/net.c line 409
0x400ff1aa: mbedtls_net_recv at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/net.c line 409
0x400f8bd9: mbedtls_ssl_fetch_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f9913: mbedtls_ssl_read_record at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400fa059: mbedtls_ssl_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400d19f4: data_to_read(sslclient_context*) at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/ssl_client.cpp line 287
0x400d14d4: WiFiClientSecure::available() at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 188
0x400d2164: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*) at /Users/maxfieldmorrison/Documents/Arduino/libraries/PubSubClient/PubSubClient.cpp line 586
0x400d2211: PubSubClient::connect(char const*, char const*, char const*) at /Users/maxfieldmorrison/Documents/Arduino/libraries/PubSubClient/PubSubClient.cpp line 586
0x400d1333: mqtt_reconnect() at /Users/maxfieldmorrison/Desktop/arduino_azure_test/arduino_azure_test.ino line 22
0x400d1418: loop at /Users/maxfieldmorrison/Desktop/arduino_azure_test/arduino_azure_test.ino line 81
0x401298b1: loopTask(void*) at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 11 (discriminator 1)
```

Sometimes there is variation in the stack trace:

```
0x400854b3: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 212
0x40084990: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4399
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3019
0x4008352b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400ebe91: sys_mbox_post at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/freertos/sys_arch.c line 562
0x400e39a5: tcpip_callback_with_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 473
0x400e3398: lwip_getsockopt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c line 3235
0x400e3764: lwip_getsockopt_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c line 3235
0x400ff155: mbedtls_net_errno at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/net.c line 409
0x400ff1aa: mbedtls_net_recv at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/net.c line 409
0x400f8bd9: mbedtls_ssl_fetch_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f9913: mbedtls_ssl_read_record at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f9e29: mbedtls_ssl_parse_change_cipher_spec at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x4010c8cf: mbedtls_ssl_handshake_client_step at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_cli.c line 3357
0x400f7b84: mbedtls_ssl_handshake_step at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f7bc4: mbedtls_ssl_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f9178: mbedtls_ssl_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400d1a30: send_ssl_data(sslclient_context*, unsigned char const*, unsigned short) at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/ssl_client.cpp line 305
0x400d14f0: WiFiClientSecure::write(unsigned char const*, unsigned int) at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 188
0x400d2028: PubSubClient::write(unsigned char, unsigned char*, unsigned short) at /Users/maxfieldmorrison/Documents/Arduino/libraries/PubSubClient/PubSubClient.cpp line 586
0x400d214c: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*) at /Users/maxfieldmorrison/Documents/Arduino/libraries/PubSubClient/PubSubClient.cpp line 586
0x400d2211: PubSubClient::connect(char const*, char const*, char const*) at /Users/maxfieldmorrison/Documents/Arduino/libraries/PubSubClient/PubSubClient.cpp line 586
0x400d1333: mqtt_reconnect() at /Users/maxfieldmorrison/Desktop/arduino_azure_test/arduino_azure_test.ino line 22
0x400d1418: loop at /Users/maxfieldmorrison/Desktop/arduino_azure_test/arduino_azure_test.ino line 81
0x401298b1: loopTask(void*) at /Users/maxfieldmorrison/Documents/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 11 (discriminator 1)
```

But as you can see it always ends with uxListRemove()

me-no-dev, 2017-02-19:
you are hitting a known bug. Check for updated libs tomorrow :) should have your issue sorted.

me-no-dev, 2017-02-22:
try now ;) should be fine

me-no-dev, 2017-03-10:
Closing because of no response

closed 2017-03-10 by me-no-dev