espressif/arduino-esp32 #220: async analogRead

ppisljar, 2017-02-18:
if i understand correctly we can only do a sync analog read at the moment ?
this has a problem of being super slow and wasting many cpu cycles ...

i tried to split analogRead into 3 functions .... analogReadAsync, analogReadAsyncReady and analogReadAsyncGet : 

```
void IRAM_ATTR __analogReadAsync(uint8_t pin)
{
    int8_t channel = digitalPinToAnalogChannel(pin);
    SET_PERI_REG_BITS(SENS_SAR_MEAS_START1_REG, SENS_SAR1_EN_PAD, (1 << channel), SENS_SAR1_EN_PAD_S);
    CLEAR_PERI_REG_MASK(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_START_SAR_M);
    SET_PERI_REG_MASK(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_START_SAR_M);
}

bool IRAM_ATTR __analogReadAsyncReady()
{
    return (GET_PERI_REG_MASK(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_DONE_SAR) != 0);
}

uint16_t IRAM_ATTR __analogReadAsyncGet()
{
    return GET_PERI_REG_BITS2(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_DATA_SAR, SENS_MEAS1_DATA_SAR_S);
}
```

and then use it like:

```
analogAsyncRead(34);
if (analogReadAsyncReady) myvalue = analogReadAsyncGet();
```

however this is far from perfect and could be implemented better i guess.

also is there an Interrupt i could set to be fired when adc reading is done ? or something like free running mode on atmega chips ? can i somehow change clock freq to make it faster (and less precise i guess) ?

me-no-dev, 2017-02-19:
Using Async here is really wrong :) this is not async at all. Interrupts are :)
give me some time to fix some other stuff I'm working on and I'll have another look at the ADC hardware and see what I can do.
Probably would not use "analogRead" in the names at all :)

ppisljar, 2017-02-20:
yeah i agree, its just something i was able to quickly hack together, and it seems its working (instead of waiting for adc finished flag in a loop i just am able to do other things now), but this should be solved with interrupts .... but i don't know enough to do it in the correct way :)

i would appreciate it if you can provide a better way to do it (also timer with interrupt would be nice)

me-no-dev, 2017-03-03:
ADC non-blocking api has been implemented :)

```cpp
void setup(){
  Serial.begin(115200);
  adcAttachPin(35);
  adcAttachPin(25);
}

void loop(){
  //start 1 pin for each ADC bus
  adcStart(35);
  adcStart(25);
  //you can check adcBusy(pin), but adcEnd will wait anyway
  uint16_t a0 = adcEnd(35);
  uint16_t a1 = adcEnd(25);
  Serial.printf("ADC: %u, %u\n", a0, a1);
}
```

closed 2017-03-03 by me-no-dev

chemicstry, 2017-03-03:
Does ESP32 have capabilities to do continuous ADC sampling via DMA? I want to do audio recording and not sure if it's possible.

wrgallo, 2020-10-07:
> ADC non-blocking api has been implemented :)
> 
> ```c++
> void setup(){
>   Serial.begin(115200);
>   adcAttachPin(35);
>   adcAttachPin(25);
> }
> 
> void loop(){
>   //start 1 pin for each ADC bus
>   adcStart(35);
>   adcStart(25);
>   //you can check adcBusy(pin), but adcEnd will wait anyway
>   uint16_t a0 = adcEnd(35);
>   uint16_t a1 = adcEnd(25);
>   Serial.printf("ADC: %u, %u\n", a0, a1);
> }
> ```

This was removed for some reason, should this issue be oppened again?

Removed in https://github.com/espressif/arduino-esp32/commit/d8b209846131119ad6a84af5987e8625f37900df

Tomat7, 2021-03-05:
Hello!

Why this function was removed?
What to use instead?

ILYA.

Triangulix, 2021-05-22:
Hi
I just updated the toolchain from 1.0.4 to 1.0.6. Why are the functions for non-blocking ADC operations gone? I need them!

stg, 2022-11-11:
Not sure if, and for how long this would be useful to anyone - but I needed asynchronous/non-blocking ADC conversions today, and i needed them to be a lot faster than using the regular API:

https://github.com/stg/ESP32-S3-FastAnalogRead

Here's hoping it might be useful to others landing here.