espressif/arduino-esp32 #224: Could not find boards.txt in ....

ikarisan, 2017-02-19:
I tried to install the board definitions on Windows10 but the Arduino IDE misses the boards.txt.
First, after an IDE restart the board is not listed under the available ones and if I open the board manager I am getting the following error messages:

`
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\cores. Is it pre-1.5?
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\doc. Is it pre-1.5?
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\libraries. Is it pre-1.5?
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\package. Is it pre-1.5?
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\tools. Is it pre-1.5?
Could not find boards.txt in X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32\variants. Is it pre-1.5?
WARNING: Error loading hardware folder X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32
  No valid hardware definitions found in folder arduino-esp32.
`

arcao, 2017-02-19:
You cloned this git repository into wrong path. The `board.txt` file has to be in: `X:\Develop\Arduino\IDE-Workspace\hardware\espressif\esp32`. Just move `X:\Develop\Arduino\IDE-Workspace\hardware\arduino-esp32` directory to `X:\Develop\Arduino\IDE-Workspace\hardware\espressif\esp32`. For more, see [Steps to install Arduino ESP32 support on Windows](https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md).

closed 2017-02-22 by me-no-dev

bobybobybob, 2017-08-03:
Totally awkward. Why not as simple as with 8266 with a text line inserted?  So I need to install unnecessary extra programs like this Git program.

amatii, 2017-08-22:
totally annoying

JB47, 2018-05-12:
Hi there!
I use windows 8.1, but get the same error message. So I wonder about the solution mentioned by arcao; did it work??

AnanthaChakravarthi, 2018-09-18:
The link given by arcao is not working

arcao, 2018-09-18:
See installation instructions on homepage: https://github.com/espressif/arduino-esp32#installation-instructions

dungcnttimd, 2021-11-12:
I'm so sorry, I don't really know what's wrong with it? Someone help me?
Arduino: 1.8.16 (Windows Store 1.8.51.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from c:\users\tiendung\onedrive - gm.uit.edu.vn\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\stdlib.h:36,

                 from C:\Users\TienDung\OneDrive - gm.uit.edu.vn\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,

                 from sketch\sketch_nov13a.ino.cpp:1:

c:\users\tiendung\onedrive - gm.uit.edu.vn\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory

   41 | #include <bits/c++config.h>

      |          ^~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Could not find boards.txt in C:\Users\TienDung\OneDrive - gm.uit.edu.vn\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.1. Is it pre-1.5?

java.lang.NullPointerException

java.lang.RuntimeException: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)

	at java.lang.Thread.run(Thread.java:748)

Caused by: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:275)

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)

	... 1 more

java.lang.NullPointerException

java.lang.RuntimeException: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)

	at java.lang.Thread.run(Thread.java:748)

Caused by: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:275)

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)

	... 1 more

java.lang.NullPointerException

java.lang.RuntimeException: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)

	at java.lang.Thread.run(Thread.java:748)

Caused by: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:275)

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)

	... 1 more
[... 221 lines cut ...]
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)

	... 1 more

java.lang.NullPointerException

java.lang.RuntimeException: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:221)

	at java.lang.Thread.run(Thread.java:748)

Caused by: java.lang.NullPointerException

	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:275)

	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:214)

	... 1 more

Could not find boards.txt in C:\Users\TienDung\OneDrive - gm.uit.edu.vn\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.1. Is it pre-1.5?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.