espressif/arduino-esp32 #225: WIN 7 Install Using Arduino 1.8.1     Can Not Find specs-esp32

Alan-Kirk, 2017-02-20:
Using the prescribed Git operations and attempting to compile my first "Blink" program I get the following error plus several :Notes... of natively supported and core architectures   then exits status 1

avr-g++: error: cannot access device-specs for 'esp32' expected at 'c:/users/alan/documents/ardunio compiler/ardunio 1.8.1/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/device-specs/specs-esp32'

avr-g++: note: devices natively supported: ata5272 ata5505 ata5702m322 ata5782 ata5790 ata5790n ata5791  ........................... at90usb1287 at90usb162 at90usb646 at90usb647 at90usb82 at94k m3000

avr-g++: note: supported core architectures: avr2 avr25 avr3 avr31 avr35 avr4 avr5 avr51 avr6 avrxmega2 avrxmega4 avrxmega5 avrxmega6 avrxmega7 avrtiny avr1

avr-g++: note: you can provide your own specs files, see <http://gcc.gnu.org/onlinedocs/gcc/Spec-Files.html> for details

I Can't seem to find any specs-32 file.  
I Can compile HUZZA ESP8266 board.  
What have I missed?  
Is it the WIN 7?

me-no-dev, 2017-02-20:
where did you install the ESP32 core? the log above talks about AVR, which is the regular old arduino.

Alan-Kirk, 2017-02-20:
The ESP32 Core folder is here:

C:\Users\Alan\Documents\ARDUNIO Compiler\Ardunio
1.8.1\Arduino\hardware\espressif\esp32\cores

This appears to be the problem however I don't understand (like don't have
a clue) how Arduino IDE figures out where the appropriate folder is when I
use different boards/MCUs .

I copied the ESP32 Boards file to:

C:\Users\Alan\Documents\ARDUNIO Compiler\Ardunio
1.8.1\Arduino\hardware\arduino\avr

That got the IDE pull down for Tools to show the ESP32 boards but that was
the start of my slippery slide to being really lost.  I had hoped (not a
good part of my plan) the GitUi would put everything in the right place.
That didn't work!

When I used GitUi my destination was:
C:\Users\Alan\Documents\ARDUNIO Compiler\Ardunio
1.8.1\Arduino\hardware\espressif\esp32

with all the \'s replaces with /'s.  This was similar to the example in
this thread, as best as I understood it.

Alan-Kirk, 2017-02-20:
I found a solution to my error.  If I change the name of the file Arduino in the Hardware folder to something else, then the file Espressif is used by the Arduino IDE and all appears well.  
Did a little reading and saw the IDE defaults to the Arduino file in Hardware. 
Thanks for you help...I'm sure I'll be back for more!

me-no-dev, 2017-02-20:
great :) this is still rather strange though

closed 2017-02-20 by me-no-dev