espressif/arduino-esp32 #226: WiFi.h is not working on my esp32

Senthanamuthan, 2017-02-20:
I'm new to ESP32.I did one sample code.

void setup() {
 pinMode(4, OUTPUT);
 Serial.begin(115200);
}
void loop() {
  Serial.println("HELLO");
  digitalWrite(4, HIGH); 
  delay(1000);            
  digitalWrite(4, LOW);  
  delay(1000);            
}
 but the same code is not working with WiFi.h

#include<WiFi.h>
void setup() {
 pinMode(4, OUTPUT);
 Serial.begin(115200);
}
void loop() {
  Serial.println("HELLO");
  digitalWrite(4, HIGH); 
  delay(1000);            
  digitalWrite(4, LOW);  
  delay(1000);            
}
Please help me

me-no-dev, 2017-02-20:
just including WiFi.h would not change a thing. Are you sure you are not missing something else? Also what exactly "same code is not working with WiFi.h" means? Post some serial output ;)
Check your USB cable (starting WiFi uses lots of power). Things like that could help.

Senthanamuthan, 2017-02-20:
Hi @me-no-dev,
                       Thank you very much for responding as soon as possible.

with the code below

void setup() {
 Serial.begin(115200);}
void loop() {
  Serial.println("HELLO ALL");
  delay(1000);
}
I got this output.
[image]

After including WiFi.h i got the below error
#include <WiFi.h>
void setup() {
 Serial.begin(115200);
 }
void loop() {
  Serial.println("HELLO ALL");
  delay(1000);
}
[image]

It shows there is some error in WiFi.h (error: no members matching 'WiFiScanClass::MODE' in 'class WiFiScanClass') How to solve this?
 Please help me.

Thanks and Regards,
 Senthan

andrei-ivanov, 2017-02-20:
Can you try with the latest Arduino IDE? 1.8.1?

Senthanamuthan, 2017-02-20:
I tried with that.While trying with Arduino IDE 1.8.1 it show no error and shows nothing in the serial monitor also

me-no-dev, 2017-02-20:
problem is that you have some library named WiFi in your Arduino libraries folder. Remove that or rename it (folder and headers) :)

Senthanamuthan, 2017-02-20:
Thank you very much @me-no-dev @andrei-ivanov 
I tried that and i got below error. It shows CPU halted 
[image]

me-no-dev, 2017-02-20:
now make sure that you have a good usb cable and power to your board ;) maybe try external power?

Senthanamuthan, 2017-02-21:
I'm really  sorry. I forget to inform that I got this error when i used "WiFi.begin(ssid,pass);" and it shows there is no "WiFi.mode()" in WiFi. h

Senthanamuthan, 2017-02-22:
Thank you very much for your responses.I got "CPU HALTED" when i include "WiFI.begin()". I tried with espressif/esp-idf#242 .It shows like the below image.
[image]
please help me.
Regards,
Senthan

me-no-dev, 2017-02-22:
check you USB cable. Try external power

Alexandravq, 2017-03-02:
Gracias por la ayuda, yo tuve el mismo problema, me arrojaba el error CPU halted, lo que hice fue cambiar la fuente de alimentación por una de mayor corriente y funciono perfectamente. Espero que les funcione.

me-no-dev, 2017-03-02:
@Alexandravq нямам никаква идея какви ми ги пишеш...

me-no-dev, 2017-03-10:
Is this still a problem?

tobozo, 2017-03-10:
@Senthanamuthan if you rename the Documents/Arduino/libraries/Wifi/ folder to /Wifi.old/ and hit "compile" button, do you still get the error ?

JPB1JPB1, 2017-03-23:
From #218 : also around Arduino IDE and ESP32 as my question is #281 ( only is bluetooth running ?)

For your information, i have recompiled yesterday for the ESP32 this wifibluetoothswitch.ino and other example under Arduino 1.6.13 without any problem, ok under PC with W10,

The hardware esp32 was set in the hardware directory for esp32-arduino github...
It looks ok for other wifi example too... Python is 2.7.12 on my PC..

If this can help you but for me, ESP32 et Arduino is better now.. from here : https://github.com/espressif/arduino-esp32 and follow under Windows instructions with success..
After the download, the wifi is running but my question is for Bluetooth ...

Best Regards
JP

me-no-dev, 2017-03-23:
@JPB1JPB1 why are you spamming so many issues with your BT question?

closed 2017-04-03 by me-no-dev

JPB1JPB1, 2018-05-18:
Hello Espressif team

Please, is there a arduino weblink with full support and source code with the 3 interfaces at the same time with power consumption optimization ,  four bits sd card interface too..

Any weblink and information are welcome.
Br
Jp