espressif/arduino-esp32 #228: Multithreading?

AEnimaH, 2017-02-23:
I want my wifi connection to listen in a separate thread? How can i do this?

copercini, 2017-02-23:
Check this out: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino

You can manage your wifi connection without block the main Sketch

me-no-dev, 2017-02-23:
what do you mean by "listen on another thread"?

AEnimaH, 2017-02-24:
I want to have a different thread where a little rest api is running and responds to get requests. This intirely separated from the main loop.

whatnick, 2017-03-17:
You might be better served using the RTOS port for the ESP32 which has a clear concept of threads, rather than baremetal like Arduino. https://github.com/espressif/ESP31_RTOS_SDK

igrr, 2017-03-17:
@whatnick what makes you think that this Arduino core is bare metal? It runs on top of FreeRTOS.

whatnick, 2017-03-17:
I meant as threads explicitly exposed, rather than abstracted away in Arduino land.

me-no-dev, 2017-03-17:
@whatnick nothing is abstracted ;) Arduino runs in one thread and you are free to spin another one or more if you need to. API is exactly the same... as @igrr said, Arduino on ESP32 runs on FreeRTOS

whatnick, 2017-03-17:
What would be the concrete answer to the above ? A threaded webserver sample ?

me-no-dev, 2017-03-17:
@whatnick so you want an example? I'm not quite sure what the issue is here?
Again, Arduino runs on top of RTOS and the specific RTOS is: https://github.com/espressif/esp-idf

whatnick, 2017-03-18:
An example would be great or specific API docs for accessing the
threading.h related functions.

Otherwise I will read the samples above and adapt.

https://github.com/espressif/esp-idf

forthlightning, 2017-03-20:
Not sure about threading.h but freertos has tasks which probably satisfy what you want to do.

I'm on freertos.org all day, here is the task API call that esp-idf uses: http://www.freertos.org/a00125.html

A task has to be of the form

```
void do_thing_task( void* p ) // gotta have a void pointer as parameter even if unused
{
while( true ) // gotta have infinite loop
    {
        // do things
        vTaskDelay( 1000 / portTICK_PERIOD_MS ) // wait / yield time to other tasks
    }
}
```

Then you start it with 
```
xTaskCreate( 
&do_thing_task,        // function
"do_thing_task",       // name
 2048,                 // stack
NULL,                  // void* to input parameter
 10,                   // priority 
NULL                   // task handle 
);

```

baldram, 2017-07-01:
This is a really interesting topic. To clarify my doubts I would like to ask by showing example.

Since arduino-esp32 is built on top of FreeRTOS, does it mean it is possible to call `xTaskCreate` directly from `*.ino` code? Is there a sufficient API for doing that?

It would look somehow like this example then:
https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/Blink_AnalogRead/Blink_AnalogRead.ino

Is it possible with arduino-esp32 ?

hakuren, 2017-07-01:
Check out this example by @copercini 
https://github.com/copercini/esp32-iot-examples/blob/master/multiloop/multiloop.ino

copercini, 2017-08-27:
You can use FreeRTOS APIs without include anything in the arduino sketch, but run directly in IDF is really more optimized

me-no-dev, 2017-11-06:
you can read more here: http://esp-idf.readthedocs.io/en/latest/api-guides/freertos-smp.html
the only way core0 to be less loaded is if you are running without BLE and WiFi, in which case many threads that usually run there will not be started

Humancell, 2017-11-06:
Hello,

Thank you VERY much for this reference ... great information!  Per what I have read, it *appears* that the following are the answers to my questions:

1. "_Note that the vanilla FreeRTOS functions xTaskCreate and xTaskCreateStatic have been macro defined in ESP-IDF FreeRTOS to call xTaskCreatePinnedToCore() and xTaskCreateStaticPinnedToCore() respectively with tskNO_AFFINITY as the xCoreID value._" - This *appears* to mean that the task will be added to the Task List with xCoreID set to tskNO_AFFINITY, and that *appears* to mean that the scheduler run from **either** core (PRO_CORE) or (APP_CORE) would match that task and run it.  I'm thinking of how I can write some code to verify this is the case.

2. Reading further here ( http://www.freertos.org/implementing-a-FreeRTOS-task.html ) it does state:  "_Tasks must not attempt to return from their implementing function or otherwise exit.  In newer FreeRTOS port attempting to do so will result in an configASSERT() being called if it is defined.  If it is necessary for a task to exit then have the task call vTaskDelete( NULL ) to ensure its exit is clean._"  This *appears* to mean it is better to utilize Queues, Semaphores, or Mutexes to control the execution of tasks, rather than attempting to schedule and reschedule them.

3. I did find a few more references to ways to gather task and CPU utilization information.  I'll again have to write some test code and see what I get, but these seemed to have touched on the areas that I'm interested in.  http://www.freertos.org/a00021.html#vTaskGetRunTimeStats  and http://www.freertos.org/rtos-run-time-stats.html

Thanks again!

lonerzzz, 2017-11-06:
@Humancell 

I have found it fairly straightforward to use task control tools as you mentioned to manage the tasks as opposed to playing with scheduling at run time.

As for gathering task information, only some are available based on the FreeRTOS configuration file in IDF so be prepared for some to result in errors. I use those for checking the task heap and whether tasks are blocked primarily.

VojtechBartoska, 2023-04-28:
@PilnyTomas I think this issue can be closed as it was mainly covered by your PR above.

PilnyTomas, 2023-06-15:
> @PilnyTomas I think this issue can be closed as it was mainly covered by your PR above.

I agree :)

closed 2023-06-15 by PilnyTomas