espressif/arduino-esp32 #230: Using as IDF component: fatal error: esp_bt_defs.h: No such file or directory

Curclamas, 2017-02-24:
When trying the "arduino as idf component" option with the latest master of idf and arduino-esp32 I get the following upon ```make```:

```
CC cores/esp32/esp32-hal-bt.o
/home/ubuntu/workspace/esp/arduino-loop-test/components/arduino/cores/esp32/esp32-hal-bt.c:18:25: fatal error: esp_bt_defs.h: No such file or directory
compilation terminated.
make[1]: *** [cores/esp32/esp32-hal-bt.o] Error 1
make: *** [arduino-build] Error 2
```

me-no-dev, 2017-02-24:
maybe BT is not enabled in menuconfig?

Curclamas, 2017-02-24:
Interestingly enough enabling BT fixed the problem. Just out of curiosity: Even if I do not need BT, do I always have to enable it?

me-no-dev, 2017-02-24:
I'm leaning towards considering this as a bug and will look into detecting if BT is enabled and make it not error out. I just have BT enabled all over the place and did not hit it :)

Curclamas, 2017-02-24:
From a user perspective I also think that just disabling a feature (which you do not use after all) should not cause the build process to fail.

chemicstry, 2017-02-24:
I'll add to this issue, since it seems related. After enabling BT in menuconfig I get the following errors:
```CC cores/esp32/esp32-hal-bt.o
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c: In function 'btStarted':
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:22:13: error: implicit declaration of function 'esp_bt_controller_get_status' [-Werror=implicit-function-declaration]
     return (esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED)
             ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:22:47: error: 'ESP_BT_CONTROLLER_STATUS_ENABLED' undeclared (first use in this function)
     return (esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED)
                                               ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:22:47: note: each undeclared identifier is reported only once for each function it appears in
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c: In function 'btStart':
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:26:42: error: 'ESP_BT_CONTROLLER_STATUS_ENABLED' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:29:42: error: 'ESP_BT_CONTROLLER_STATUS_IDLE' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:33:42: error: 'ESP_BT_CONTROLLER_STATUS_INITED' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:34:13: error: implicit declaration of function 'esp_bt_controller_enable' [-Werror=implicit-function-declaration]
         if (esp_bt_controller_enable(ESP_BT_MODE_BTDM)) {
             ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:34:38: error: 'ESP_BT_MODE_BTDM' undeclared (first use in this function)
         if (esp_bt_controller_enable(ESP_BT_MODE_BTDM)) {
                                      ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c: In function 'btStop':
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:47:42: error: 'ESP_BT_CONTROLLER_STATUS_IDLE' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:50:42: error: 'ESP_BT_CONTROLLER_STATUS_ENABLED' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:51:13: error: implicit declaration of function 'esp_bt_controller_disable' [-Werror=implicit-function-declaration]
         if (esp_bt_controller_disable(ESP_BT_MODE_BTDM)) {
             ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:51:39: error: 'ESP_BT_MODE_BTDM' undeclared (first use in this function)
         if (esp_bt_controller_disable(ESP_BT_MODE_BTDM)) {
                                       ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:57:42: error: 'ESP_BT_CONTROLLER_STATUS_INITED' undeclared (first use in this function)
     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
                                          ^
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c: In function 'btStarted':
/home/chemicstry/Dev/ESPProjects/switchwise/components/arduino/cores/esp32/esp32-hal-bt.c:23:1: error: control reaches end of non-void function [-Werror=return-type]
 }
cc1: some warnings being treated as errors
/home/chemicstry/esp32/esp-idf/make/component_wrapper.mk:176: recipe for target 'cores/esp32/esp32-hal-bt.o' failed
make[1]: *** [cores/esp32/esp32-hal-bt.o] Error 1
/home/chemicstry/esp32/esp-idf/make/project.mk:382: recipe for target 'arduino-build' failed
make: *** [arduino-build] Error 2

```

Any ideas?

chemicstry, 2017-02-24:
Fixed. Had to update IDF.

forthlightning, 2017-02-28:
another place where changing a menu setting breaks the build is mbedTLS debug. You have to manually go into the config in the arduino library and change the debug set level. Not sure if this warrants a separate issue.

me-no-dev, 2017-02-28:
What level do you have to set it at. Yes this is a good place to discuss those issues :)

forthlightning, 2017-02-28:
If mbedTLS debug is turned off via make menuconfig. Build fails with
```
.....arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:170: undefined reference to `mbedtls_debug_set_threshold'
```

So I turn debugging back on and go [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/ssl_client.cpp#L31) and change from 4 to 1.

Level 4 spams so hard, and for my purposes at least the debugging provided by the arduino wrapper class is just fine.

me-no-dev, 2017-02-28:
@forthlightning check this out: https://github.com/espressif/arduino-esp32/commit/bcd6dcf5f6f8a4670db53a495ad8c8b556c2a8fa

me-no-dev, 2017-02-28:
@Curclamas your issue should be fixed here: https://github.com/espressif/arduino-esp32/commit/5f2b00228c60995eafbba3d4ad8849ecd82d505b

me-no-dev, 2017-03-10:
Closing because of no response and because all is fine

closed 2017-03-10 by me-no-dev