espressif/arduino-esp32 #231: sntp  NIST time server

kas2, 2017-02-24:
Hi,

I am currently getting date and time using NIST Internet Time Service with my ESP8266.
when porting my code to ESP32, I get following error messages:

error: 'sntp_set_timezone' was not declared in this scope
error: 'sntp_get_current_timestamp' was not declared in this scope
error: 'sntp_get_real_time' was not declared in this scope

I checked the ESP32 "sntp.h", and strangely enough, those 3 functions are not declared in this file

What am I missing ??
Any sntp demo code available ??
TIA

me-no-dev, 2017-02-24:
Not a clue :) maybe ask in ESP-IDF? sntp.h is part of IDF.
Here is what I use to setup/read time
```cpp
#include "time.h"
void printLocalTime()
{
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
        Serial.println("Failed to obtain time");
        return;
    }
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

configTime(-7200, 3600, "pool.ntp.org");
```

You can check the ```configTime``` source for how it works.

kas2, 2017-02-24:
Thanks me-no-dev for your fast response,
and kudoos for your ongoing hard work

closed 2017-02-24 by kas2

me-no-dev, 2017-02-24:
:) glad it helped