espressif/arduino-esp32 #232: Application won't start after flashing

Curclamas, 2017-02-24:
Since I build and flash on separate machines I won't run ```make flash```. Rather I will ```make``` and then copy over ```bootloader.bin```, ```default.bin```, ```boot_app0.bin``` and ```app-template.bin``` to the "flash machine".
I will then flash them with:

```
python esptool.py --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader.bin 0x10000 app-template.bin 0x8000 default.bin
```

This succeedes. On the console the ESP32 (devKit J) will print out:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2488
load:0x40078000,len:6920
load:0x40080000,len:252
entry 0x40080034
I (578) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (579) heap_alloc_caps: At 3FFC52C0 len 0001AD40 (107 KiB): DRAM
I (588) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (609) heap_alloc_caps: At 40097B40 len 000084C0 (33 KiB): IRAM
I (630) cpu_start: Pro cpu up.
I (641) cpu_start: Single core mode
I (654) cpu_start: Pro cpu start user code
I (715) cpu_start: Starting scheduler on PRO CPU.
```

After that nothing else is printed to the console. My App looks like:

```
//file: main.cpp
#include "Arduino.h"

#define BLINK_PIN 5

void setup(){
  Serial.begin(115200);
  Serial.println("hello");
  pinMode(BLINK_PIN, OUTPUT);
}

void loop(){
  Serial.println("loop");
  digitalWrite(BLINK_PIN, LOW);
  delay(1000);
  digitalWrite(BLINK_PIN, HIGH);
  delay(1000);
}
```

I see neither the console logs nor any level changes on ```IO5```. 

What could have gone wrong here? 

Could it be that "make flash" does some other stuff? I also checked the binary ```app-template.bin```, it contains the strings "loop" and "hello" so at least it is the correct file.

Just to be complete here is the arduino part from sdconfig:

```
#
# Arduino Configuration
#
CONFIG_ENABLE_ARDUINO_DEPENDS=y
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_DISABLE_HAL_LOCKS is not set
```

me-no-dev, 2017-02-24:
You have in menuconfig an option that says something like "start scheduler on CPU0 only" and that is selected. Arduino runs on CPU1

Curclamas, 2017-02-24:
That did indeed fix the problem, found the option in the FreeRTOS-Settings. Is there a way to make the "correct" selection a default option? Or maybe this should be noted in the doc since I'm quite sure many people try to use the default template and then run into this problem.

me-no-dev, 2017-02-24:
there are two ways to do it. One is to detect that CORE1 is off and run on core 0, the other is to force CORE1 on. Both exclude each other and can not exists in Arduino at the same time, so I need to pick one, but have not decided which one yet.

me-no-dev, 2017-02-24:
have to think about it more, maybe it's doable through option in Arduino's menuconfig

lonerzzz, 2017-08-25:
@me-no-dev Any update on this issue or suggestions to solve in terms of choosing the correct setting?

copercini, 2017-08-27:
@lonerzzz It was solved here: https://github.com/espressif/arduino-esp32/commit/5f2b00228c60995eafbba3d4ad8849ecd82d505b

closed 2017-08-27 by copercini