espressif/arduino-esp32 #233: How to include external arduino libs while using as IDF component

Curclamas, 2017-02-24:
Use case: I am using arduino-esp32 as an IDF component. I have some Arduino libs which I want to use, for example SHT21 (i2c Thermometer/Hygrometer).

In my code I have ```#include <SHT21.h>``` . This works perfectly well with the classic Arduino IDE approach.

However while using it as an IDF-component I tried to copy the SHT21 lib folder to the components/arduino/libraries folder. Upon ```make``` this yields:

```
CXX main.o
/home/ubuntu/workspace/esp/arduino-iot-hub/main/./main.cpp:3:21: fatal error: SHT21.h: No such file or directory
```

How/where do I have to include my libraries and where should I put the folder?

me-no-dev, 2017-02-24:
your question should be asked in IDF's github. Including libs requires component.mk file to be added, but there might be an easier solution. So better ask in IDF.

Curclamas, 2017-02-24:
I might do, but I think you are right with the component.mk point. The thing is that arduino usualy has some "magic" which automatically resolves these folders (i.e. the structure with the folder name, the "src" folder and then the .cpp/.h with the same name as the folder. So maybe it is possible to add this "magic" to the arduino-component makefile. Otherwise you would have to do this manually (in a more "IDF-fashion" way) for every lib you use.

Curclamas, 2017-02-24:
I managed to get it working by editing the Arduino component.mk to also include my libs, not only the arduino-esp32 default libs. Since this is a little bit hacky in my opinion it would be cool to find a way to add some kind of "wildcard" or something like this to "emulate" the default arduino behavior of just using whatever is in the libraries directory. So from an IDF perspective I have no problem here since I can get it to work manually but it would be cool if we had some "default/automated" way of adding off the shelf arduino libs.

me-no-dev, 2017-02-24:
that location is user/os dependent. And it's a setting in Arduino IDE. not something we can guess. Not something that I can add to arduino.

Curclamas, 2017-02-24:
What also worked for me is editing the component.mk of arduino as follows:

``` ARDUINO_LIB_DIRS := $(wildcard libraries/*/src) ```

That way it will include everything thats in the arduino libraries folder and follows the arduino lib/src convention. Would that be acceptable for you?

me-no-dev, 2017-02-24:
yes. that can work for local libs (mostly). source could be in subfolders and other stuff. I have some code I wrote before that follow Arduino conventions. Maybe I can integrate something like that to autodetect the required libs and include them (if the user have specified libraries folder)

me-no-dev, 2017-03-01:
wildcard for core libs is in main ;) thanks @Curclamas for the idea. Will have to dig up that old makefile stuff and see how and if I'll be able to adapt it to IDF

me-no-dev, 2017-03-01:
can't find it :( found where it was supposed to be but no file there.

kyjak, 2017-06-24:
@Curclamas hello, did you manage to get the sht21 work on esp32?

ljozsa, 2017-12-29:
@Curclamas I tried to add  

ARDUINO_LIB_DIRS := $(wildcard libraries/*/)

to component.mk and copied over all the libs from ~/Arduino/libraries/ to the libraries folder that resides at the same level as the component.mk. All the libs have their *.h and *.cpp files in the root folder of each library (no src subdir as in your example).

However running make still results in an error. What am I doing wrong?

Many thanks.

tejaswigowda, 2019-02-17:
any progress?

tejaswigowda, 2019-02-18:
Any help is appreciated. Using Arduino as component is near-useless if we cannot include Arduino libraries

me-no-dev, 2019-02-18:
This will become even bigger mess once we move to cmake :) What is exactly that you want? The build to get the libs from your Arduino libraries folder or from a libraries folder inside the project?

tejaswigowda, 2019-02-18:
From Arduino libraries folder. I put my libs here: `components/arduino-esp32/libraries/`

tejaswigowda, 2019-02-18:
Looks like it is getting it, just not including it right

``
X build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEBeacon.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLECharacteristic.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLECharacteristicMap.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEClient.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEDescriptor.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEDescriptorMap.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEDevice.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEEddystoneTLM.o
CXX build/arduino-esp32/libraries/ESP32BLEArduino/src//BLEEddystoneURL.o
``

Does the double slash (//) caus a problem you think? Please help!! This is a great feature

robert-alfaro, 2019-04-17:
@tejaswigowda did you ever figure this out...hopefully with CMake?

closed 2019-08-15 by stale[bot]

roysG, 2022-09-02:
/Users/roysabah/esp/esp32-arduino-lib-builder/main/sketch.cpp:15:10: fatal error: SSLClient.h: No such file or directory
 #include <SSLClient.h>

Same issue for me but with the library "SSLClient.h".

Someone can helps?

roysG, 2022-09-03:
> I managed to get it working by editing the Arduino component.mk to also include my libs, not only the arduino-esp32 default libs. Since this is a little bit hacky in my opinion it would be cool to [...]

Can you please write what lines you added?

I am trying to add the library of SSLClient.h

But i get errors:

eFiles/__idf_main.dir/sketch.cpp.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/sketch.cpp.obj -c /Users/roysabah/esp/esp32-arduino-lib-builder/main/sketch.cpp
/Users/roysabah/esp/esp32-arduino-lib-builder/main/sketch.cpp:15:10: fatal error: SSLClient.h: No such file or directory
 #include <SSLClient.h>
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.