espressif/arduino-esp32 #234: Router connection: 50% failure (50% success)

kas2, 2017-02-24:
Hi,
I use this standard sketch to connect to Internet

```
#include <WiFi.h>

char* ssid = "xxxxxxxxxx";
char* password = "xxxxxxxx";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.print("Connecting to "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.print("\nWiFi connected   IP: "); Serial.println(WiFi.localIP());
}

void loop() {}

```
ESP8266 connects after each reboot
ESP32 only connects exactly half the case:
 - connect OK
 - reboot
 - connection failed
 - reboot
 - connect OK
 - reboot
 - connection failed
 - reboot  ...

Any clue ??
Thanks

me-no-dev, 2017-02-24:
I have no such problem here. What you describe is rather strange. The only time I have had issues like this is when I turn wifi on/off repeatedly while booted. Maybe your router has some sort of protection?

kas2, 2017-02-24:
@me-no-dev ,

I did the test again, both with ESP8266 and ESP32
Same code (just change <ESP8266WiFi.h> to <WiFi.h>)

ESP8266 connects  _every_  time, not a single glitch:

```
..................
Connecting to Livebox-27C3
66666663
WiFi connected   IP: 192.168.1.14

..........
Connecting to Livebox-27C3
66663
WiFi connected   IP: 192.168.1.14

..................
Connecting to Livebox-27C3
66663
WiFi connected   IP: 192.168.1.14

..........
Connecting to Livebox-27C3
66663
WiFi connected   IP: 192.168.1.14

```

ESP32 connects only half the case
If I get a connection, at next reboot connection is guaranted to fail,
no random phenomenon here

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Connecting to Livebox-27C3
66663
WiFi connected   IP: 192.168.1.15
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Connecting to Livebox-27C3
666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Connecting to Livebox-27C3
66663
WiFi connected   IP: 192.168.1.15
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Connecting to Livebox-27C3
66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Connecting to Livebox-27C3
6663
WiFi connected   IP: 192.168.1.15
```

The board is only connected to the serial cable
Should I pullUp / pullDown some IO's ??

> Maybe your router has some sort of protection?

My router is not supposed to know which board I am using  :)
Any other suggestion ??

Thanks for your help

plerup, 2017-02-26:
Maybe you should try calling esp_wifi_stop() before rebooting

plerup, 2017-02-26:
Hrm, or rather not. It works as long as you do call ESP.restart() to reboot but when doing a hard reset, it still fails every second time.

kas2, 2017-02-26:
Thanks plerup,
I will perform additional test and report here
I feel now less lonely in this wild world  :)

kas2, 2017-02-27:
More testing

using this sketch
```
#include <WiFi.h>

char* ssid = "Livebox-27C2";
char* password = "23456789";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.print("\nConnecting to "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.print("\nWiFi connected   IP: "); Serial.println(WiFi.localIP());
  
  delay(5000);
  WiFi.disconnect(true);
}

void loop() {}

```
just added ...
```
  delay(5000);
  WiFi.disconnect(true);

```
... to initial sketch.

If I wait 5 seconds before rebooting (allow WiFi.disconnect), ESP32 will connect each time
When I reboot immediately after "WiFi connected   IP: xxxx" display (no WiFi.disconnect), connection will fail every second time.

I think I remember ESP8266 had this beheaviour in the early development time

Any thought ??

kas2, 2017-02-28:
More testing using the WiFiClientEvents.ino ESP32 example
the board still connects every other time:

```
Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.11

<Hard reset>

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 5  - WiFi lost connection

<Hard reset>

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.11

<Hard reset>

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 5  - WiFi lost connection

```
This was tried with two different routers
 - Orange LiveBox router
 - Motorola MotoG WiFi acces point
 
------------------------------------------------------------------- 

Now, if I only request a soft reset after connection
```
  case SYSTEM_EVENT_STA_GOT_IP:
      Serial.println("WiFi connected");
      Serial.print("IP address: "); 
      Serial.println(WiFi.localIP());
      system_restart();   // <<<<<<<<<<<
      break;
```
Board will connect **each** time
```
Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.11

<soft reset>

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.11

<soft reset>

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.11

```
Any clue ??
Thanks

me-no-dev, 2017-03-01:
here is my log with the above sketch without wifi off and doing 3 resets
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034

Connecting to nbis-test
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
6666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
3
WiFi connected   IP: 192.168.254.182
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034
�
Connecting to nbis-test
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
6666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
66666666666666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
3
WiFi connected   IP: 192.168.254.182
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034

Connecting to nbis-test
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
6666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
3
WiFi connected   IP: 192.168.254.182
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034

Connecting to nbis-test
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
6666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
3
WiFi connected   IP: 192.168.254.182

```

plerup, 2017-03-02:
For me it also always works with my ASUS router but when using an Android phone as access point I get the same result as @kas2

kas2, 2017-03-02:
@ me-no-dev 
@ plerup
Thanks for your feedback

Let me sum up my findings sofar:
```
 - connect every other time with HARD reset
 - connect each time with SOFT reset
 - connect each time if disconnected (WiFi.disconnect) before HARD reset
 - tested with two diferent routers
 - same sketches are 100% OK with ESP8266 (same routers)
 - seems I am not the only one in this situation
```
There is definitely something different in my setup
This is my board: [https://s11.postimg.org/vw9fjbtkj/P1020827_copy.jpg](url)

Do you have any added pullup/pulldown resistors ??
Any work around, suggestions, additional test I can perform ??

Thanks again for your help

me-no-dev, 2017-03-02:
@kas2 are you saying that the "connect every other time" problem occurs regardless of the WiFi access point? I have not tried Android (I do not have one), but have no problems using my computer or iPhone as AP.
BTW Those boards suck! Whoever thought that it's a good idea to put board under the antenna is not in his right mind... I already see reports of bad reception

kas2, 2017-03-02:
@me-no-dev 
Yes I get this "connect every other time" with both my Orange router and my MotoG android smartphone
Also, if Wifi is disconnected using WiFi.disconnect() prior to hard reboot, the board will connect every time
Please suggest work around or additional test I can perform ??

> BTW Those boards suck! Whoever thought that it's a good idea to put board under the antenna is not in his right mind...

I saw your comment in an other thread
I am not an expert and may be wrong; the PCB is a dielectric material and shouldn't block radio emission by any means

me-no-dev, 2017-03-02:
if you wrap a bit of tape, even one that is somewhat made for antenna shielding, you are changing te properties of it. Did they at least remove the shielding of the board around the antenna? If you look at how Espressif boards are made, the antenna is outside of the board.

kas2, 2017-03-03:
OK me-no-dev , I will consider my board as defective
Where can I purchase an Espressif board ??

Thanks again for your help

me-no-dev, 2017-03-05:
Please do not consider it defective so soon :) if it connects every other time, then the module on the board actually works. I wonder if you should maybe fill an issue over at espressif/esp-idf repo and see if core engineers will be able to decipher what is going wrong. You can forward to this conversation so they know what we have tried so far.

kas2, 2017-03-06:
OK,  [link to issue #401](https://github.com/espressif/esp-idf/issues/401)

kas2, 2017-03-06:
@me-no-dev 
Following your comments, I cut the board under the antenna
https://s9.postimg.org/75qk3n8nz/P1020828_copy.jpg

Same behaviour, no change :(

kas2, 2017-03-08:
@me-no-dev 

More testing using the WiFiClientEvents.ino ESP32 example
[https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino](url)

if I add "WiFi.begin()" in the "SYSTEM_EVENT_STA_DISCONNECTED" event:
```
void WiFiEvent(WiFiEvent_t event) {
  Serial.printf("[WiFi-event] event: %d  - ", event);
  switch(event) {
  case SYSTEM_EVENT_STA_GOT_IP:
      Serial.println("WiFi connected");
      Serial.print("IP address: "); Serial.println(WiFi.localIP());
      break;
  case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("WiFi lost connection");
      WiFi.begin();   // <<<<<<<<<<<  added  <<<<<<<<<<<<<<<
      break;
  case SYSTEM_EVENT_STA_START:
      Serial.println("ESP32 station start");
      break;
  case SYSTEM_EVENT_STA_CONNECTED:
      Serial.println("ESP32 station connected to AP");
      break;
  }
}
```

The board will connect **every** time:
```
Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 5  - WiFi lost connection
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.25
ets Jun  8 2016 00:22:57

(reboot)

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 5  - WiFi lost connection
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.25
ets Jun  8 2016 00:22:57

(reboot)

Wait for WiFi... 
[WiFi-event] event: 2  - ESP32 station start
[WiFi-event] event: 5  - WiFi lost connection
[WiFi-event] event: 4  - ESP32 station connected to AP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.25
ets Jun  8 2016 00:22:57
```

Here is ESP-IDF code for the event handler
[https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_request/main/http_request_main.c](url)
```
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
```
I strongly suspect that the Arduino IDE  issue is related to
         - esp_wifi_connect();
or possibly
         - xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
as the controller remembers its last state when rebooting

Please let me have your comments
TIA

kas2, 2017-03-16:
@me-no-dev 

Up  ;-)

Please confirm that this simple sketch connects every time with your own board:

```
#include <WiFi.h>

const char* ssid     = "xxxxxxxxxx";
const char* password = "xxxxxxxxxxxx";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.print("\nConnecting to "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.print("\nWiFi connected   IP: "); Serial.println(WiFi.localIP());
}

void loop() { }
```

I tested with
 - two different boards
 - two different routers
Always connects every other time...

Thanks

me-no-dev, 2017-03-20:
@kas2 really sorry for not answering sooner. I think I got a clue on how to fix this now :)

me-no-dev, 2017-03-20:
@kas2 please select the ESP32 dev board in ArduinoIDE and select debug level to "Verbose" then post the log from failed disconnect again. I tried all APs here at home and none gave me this issue. New connect should not be started in all cases, so let's figure out which one you end up in.

kas2, 2017-03-20:
@me-no-dev 
Thanks, I will let you have the log ASAP

My current work around is to add WiFi.begin() in the event handler

  case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("WiFi lost connection");
      WiFi.begin();              //   <<<
   break;

me-no-dev, 2017-03-20:
Yes, but that would attempt reconnect at any disconnect reason (like if you want to disconnect it will connect again)

kas2, 2017-03-20:
@me-no-dev 

This is the verbose log for two resets (official Espessif board)
 - can't connect
 - can connect

Thanks for your help

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034

Connecting to Livebox-27C3
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
66666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:180] _eventCallback(): Reason: 2 - AUTH_EXPIRE
666666666666666666666666666666666666ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034

Connecting to Livebox-27C3
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
666[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
3
WiFi connected   IP: 192.168.1.25
```

me-no-dev, 2017-03-25:
Try the latest master please :)

kas2, 2017-03-26:
@me-no-dev 

Thanks me-no-dev, it works :)

using
```
else if(reason == WIFI_REASON_AUTH_EXPIRE) { 
     WiFi.begin(); 
```
versus
```
  case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("WiFi lost connection");
      WiFi.begin();              //   <<<
   break;
```
is definitly smarter, but...  it's still looks like a work around ;)
as replacing an electric fuse with a copper wire
Any reason why it always connects using ESP 8266 board and only every-other-time with ESP32 ???

This is the new debug Log with fix applied:
```
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2  - ESP32 station start
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:180] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[WiFi-event] event: 5  - WiFi lost connection
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4  - ESP32 station connected to AP
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
[WiFi-event] event: 7  - WiFi connected
IP address: 192.168.1.25
```
Strangely enough, the board now apparently disconnects and reconnects **each time**

Anycase, you solved the problem, thanks again me-no-dev

me-no-dev, 2017-03-26:
I don't have access to that part of the source so I have no idea why it does not automatically reconnect on auth expiration. What I patched though will reconnect only in that case and not in cases where you want to actually disconnect ;)

closed 2017-03-26 by me-no-dev

lrmoreno007, 2017-05-11:
Hello, I have repeated problems connecting to my WIFI and this is my configuration:
Board: ESP32-ST (HIMALAYA ESP32)
Selected board: ESP32 Dev Board
Arduino version: 1.8.2
ESP32 version: Git version
***********************************
Scenario 1: DHCP
Sketch:
```
#include <WiFi.h>

const char* ssid     = "XXXXXX";
const char* password = "XXXXXXXXX";

void enciendeWIFI() {
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WIFI");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println(" OK");
}

void setup() {
  Serial.begin(115200);
  enciendeWIFI();
}

void loop() {
  const uint16_t port = 80;
  const char * host = "10.0.2.1";
  delay(5000);
  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("Can't connect to server");
  }
  else {
    Serial.println("Connected");
  }
  client.stop();
  ESP.restart();  // To debug with multiple connections in a row
}

```
---> Result: sometimes connect OK, sometimes "Connecting to WIFI ..................[ infinite ]"
************************************
Scenario 2: Fixed IP
Sketch:
```
#include <WiFi.h>

IPAddress ip(10, 0, 1, 20);
IPAddress gateway(10, 0, 2, 200);
IPAddress subnet(255, 0, 0, 0);
const char* ssid     = "XXXXX";
const char* password = "XXXXXXXXX";

void enciendeWIFI() {
  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WIFI");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println(" OK");
}

void setup() {
  Serial.begin(115200);
  enciendeWIFI();
}

void loop() {
  const uint16_t port = 80;
  const char * host = "10.0.2.1";
  delay(5000);
  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("Can't connect to server");
  }
  else {
    Serial.println("Connected");
  }
  client.stop();
  ESP.restart();
}
```
---> Result:
Sometimes:
Connecting to WIFI .. OK
Connected

Other times:
Connecting to WIFI .. OK
Can not connect to server
*************************************
After fighting with the router and the code for several days, I found this issue and after reading it I decided to activate the "Core Debug Level --- Debug" and curiously after activating it the connection is always OK.

I do not have enough knowledge to solve it, but I think the problem is that if the "Debug" level is not activated, the SYSTEM_EVENT_STA_DISCONNECTED event can never reach the WIFI_REASON_AUTH_EXPIRE reason in "WiFiGeneric.cpp"

I would greatly appreciate a solution without having to activate Debug.
*************************************
On the other hand, why when using fixed IP this condition WL_CONNECTED is "true" if it is not actually connected?

Greetings and thank you.

me-no-dev, 2017-05-11:
please set the debug level in the board settings to verbose and run again. It should tell you what happened

lrmoreno007, 2017-05-17:
As I said: if I active debug everything goes OK. Only very occasionally fails in Debug mode and in those cases is because it crashes.

I will check that I do not have any hardware problems. If I do not solve it I will ask again and try to explain myself better.

Regards.

lrmoreno007, 2017-05-17:
Wow, with last git version updated now, **only connect first time**. I get the following:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 2 - STA_START
Connecting to WIFI
..[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 7 - STA_GOT_IP
 OK
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:184] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 7 - STA_GOT_IP
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 4 - STA_CONNECTED
```

Is this normal after ESP.restart ()?
```
...
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:184] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 7 - STA_GOT_IP
...
```

Before, It just return:

```
...
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:184] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][WiFiGeneric.cpp:175] _eventCallback(): Event: 3 - STA_STOP...
```

lrmoreno007, 2017-05-18:
I have already solved the problem, now it work like a charm.

The module had several pins with poor welding like the one shown in the image. Although I do not think it influenced for the connectivity, but I have improved the stability:
[image]

I have also changed some things in the sketch that have improved connectivity. This is the current test sketch (commented).

```
#include <WiFi.h>

IPAddress ip_local(10, 0, 2, 89);      //Added post fixed "_local"
IPAddress gateway_local(10, 0, 2, 200);      //Added post fixed "_local"
IPAddress subnet_local(255, 0, 0, 0);      //Added post fixed "_local"
const char* ssid_local     = "XXXXXX";      //Added post fixed "_local"
const char* password_local = "YYYYYYY";      //Added post fixed "_local"

void enciendeWIFI() {
	WiFi.enableSTA(true);   //Without this sometimes blocked when arriving at WiFi.begin ()
	WiFi.enableAP(false);
	delay(2000);    // Without this timeout connectivity was still very bad
	if (WiFi.status() == WL_CONNECTED) {
		Serial.println("Already connected");
	}
	else{
		WiFi.config(ip_local, gateway_local, subnet_local);
		Serial.print("Connecting ");
		WiFi.begin(ssid_local, password_local);

		while (WiFi.status() != WL_CONNECTED) {
			Serial.print(".");
			delay(500);
		}
		Serial.println(" OK");
	}
	Serial.print("Local IP: ");
	Serial.println(WiFi.localIP());
	Serial.print("MAC: ");
	Serial.println(WiFi.macAddress());
}

void setup() {
	Serial.begin(115200);
	enciendeWIFI();
}

void loop() {
//  delay(10000);
  const uint16_t port = 80;
  const char * host = "10.0.2.1";
  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("Can't connect to server");
  }
  else {
    Serial.println("Connected");
  }
  client.stop();
  ESP.restart();
}
```

Thank you.

chevdor, 2017-05-22:
I am using an ESP32 and **espruino** and see the exact same issue:
- ESP32.reboot() results in the Wifi connection to be established 100% of the time in a few seconds
- hard reboot / power cycle results in the Wifi connection to established 50% of the time. This is not random. Each hard reboot does work / fail / work / fail.

jurriaan, 2017-08-05:
I'm having the same issue, currently working around it by timing out the initial connect and doing an ESP.restart() if it doesn't connect within a certain time.

Scott--R, 2017-09-27:
I'm having the same issue as well. I'm using the ESP32-WROOM module from Espressif. I also currently have to work around it by timing out the initial connect and doing an ESP.restart() if it doesn't connect within a certain time.

BluCodeGH, 2017-12-26:
@me-no-dev I suggest re-opening this issue, as I am still getting this problem. I am using the MH-ET LIVE MiniKit for ESP32, which uses the ESP-WROOM-32 as its base. I have the latest master of both the esp-idf and arduino-esp32 and I am using the arduino-esp32 as an idf component. My code:
```cpp
#include <Arduino.h>

#include <WiFi.h>

#include <HTTPClient.h>

#define USE_SERIAL Serial

void setup() {

    USE_SERIAL.begin(115200);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFi.begin("ssid", "pwrd");
    Serial.println(WiFi.getAutoReconnect());

    wl_status_t status = WiFi.status();
    while (status != WL_CONNECTED) {
        Serial.print("Status: ");
        Serial.println(status);
        delay(500);
        status = WiFi.status();
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}
```

This works exactly every other time. Here is the verbose output when it fails:
```ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5568
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13704
entry 0x40078fb0
I (30) boot: ESP-IDF v3.1-dev-107-g5401a75 2nd stage bootloader
I (31) boot: compile time 21:13:07
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 80MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (67) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (74) boot:  2 app0             OTA app          00 10 00010000 00140000
I (82) boot:  3 app1             OTA app          00 11 00150000 00140000
I (89) boot:  4 eeprom           Unknown data     01 99 00290000 00001000
I (97) boot:  5 spiffs           Unknown data     01 82 00291000 0016f000
I (104) boot: End of partition table
[... 141 lines cut ...]
I (8339) wifi: mode : sta (24:0a:c4:81:8c:cc)
1
Status: 255
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
Status: 6
Status: 6
Status: 6
Status: 6
I (10749) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
Status: 6
Status: 6
I (11729) wifi: state: init -> auth (b0)
I (11749) wifi: state: auth -> assoc (0)
I (11779) wifi: state: assoc -> run (10)
I (11789) wifi: connected with LinksysAC, channel 6
Status: 6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
Status: 6
I (12659) event: sta ip: 192.168.1.163, mask: 255.255.255.0, gw: 192.168.1.1
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP

WiFi connected
IP address:
192.168.1.163
```

gdombiak, 2018-01-21:
I had same problem. It was connecting to Wifi once and failed the next time and then worked and then failed. I was unplugging usb cable or hard resetting with button.

I then enabled 'core debug level' to verbose and it started working each time. I then went back to 'core debug level' to none and now it works each time. This is clearly not a fix but may provide some information to find root cause. 

hardware: ADAFRUIT HUZZAH32 – ESP32 FEATHER BOARD
framework: Arduino (had esp-idf before)
app: was using example ESP32 -> Time --> SimpleTest

Happy to run more tests if that helps.

Scott--R, 2018-01-21:
We just gave up and accepted it as a fact of life for now. Just built-in a restart when it fails to connect every other time. It works but it’s not right.

BluCodeGH, 2018-01-23:
@gdombiak My core debug level is verbose and I still have the issue, are you sure your didn't change anything else? It doesn't make sense that the debug level fixes it, and especially that it stays fixed.

tablatronix, 2018-02-20:
FYI this hack
` } else if(reason == WIFI_REASON_AUTH_EXPIRE) {`

causes some auth failures to return 
`WL_DISCONNECTED`
instead of
`WL_CONNECT_FAILED`

It also runs forever until mode is changed