espressif/arduino-esp32 #235: RTC_DATA_ATT / RTC user memory?

torntrousers, 2017-02-26:
Is it possible to have RTC user memory persist over deep sleep in an Arduino sketch?

I found and example in the [IDF SNTP example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/main/sntp_main.c) which has the RTC_DATA_ATT attribute on a variable, but trying that in a Arduino sketch compiles ok but the value isn't maintained over sleeps.

Any ideas on how to do this?

krzychb, 2017-02-26:
I have not been using Arduino IDE for a while to provide a definite answer.

Basing on my experience with esp-idf, RTC values are not retained in memory if ESP32 board is "Falling back to built-in command interpreter" due to "flash read err, 1000".

Do you see a message like below after module wakes up from deep sleep and reboots?

```
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57
```

torntrousers, 2017-02-26:
Ok experimenting more, the RTC_DATA_ATT does seem to work ok if the Flash Frequency is set to 80MHz instead of the default for my board (ESP32 Thing) of 40 MHz.

me-no-dev, 2017-02-26:
that is rather strange, I also have esp32 thing here and it has Winbond flash which is more than capable at running at 80MHz (should be one of the best out there)

torntrousers, 2017-02-26:
Hmm well I can't recreate this anymore. The idea of using the 40 MHz flash speed came from [this thread](https://www.esp32.com/viewtopic.php?t=644) which suggested the slower speed worked. Before trying that it always didn't work but now after trying at 40MHz and then back to 80 MHz it works fine at both speeds. I don't think I changed anything else. perhaps something somewhere got reset by the speed change?

closed 2017-02-26 by torntrousers