espressif/arduino-esp32 #236: Python TypeError when compiling sketch

xtaco, 2017-02-26:
Hi All, 

I am not able to compile any sketches due to this error. Anyone come across this? 

```
Traceback (most recent call last):
  File "/home/user/Arduino/hardware/espressif/esp32/tools/gen_esp32part.py", line 356, in <module>
    main()
  File "/home/user/Arduino/hardware/espressif/esp32/tools/gen_esp32part.py", line 345, in main
    table.verify()
  File "/home/user/Arduino/hardware/espressif/esp32/tools/gen_esp32part.py", line 83, in verify
    for p in sorted(self):
TypeError: '<' not supported between instances of 'PartitionDefinition' and 'PartitionDefinition'
exit status 1
Error compiling for board ESP32 Dev Module.

```

me-no-dev, 2017-02-26:
check if there are no errors on your side in [this](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv) file.

carlgonz, 2017-02-27:
Seems to be a problem with de default python version. The script gen_esp32part.py seems to work only with python2

bessl, 2017-03-01:
I had the same problem. I was using the Python 3 interpreter. You can fix it temporary by doing this:

mkdir /tmp/bin
PATH=/tmp/bin:$PATH
ln -s /usr/bin/python2 /tmp/bin/python

when you start arduino ide from your terminal window it should work!

me-no-dev, 2017-03-01:
I'll turn it into binary next time around and will not be a problem.

mcgiver0510, 2017-03-09:
change this line in platform.txt 
tools.gen_esp32part.cmd=python "{runtime.platform.path}/tools/gen_esp32part.py"
to
tools.gen_esp32part.cmd=python2 "{runtime.platform.path}/tools/gen_esp32part.py" 
lg

dmopalmer, 2017-05-06:
Is it worthwhile to make gen_esp32part.py compatible with python3, or will that just kick the problem to the next incompatible step?

This requires changing line 83 of gen_esp32part.py something like (untested code):
```
+        for p in sorted(self, key=lambda x:x.offset):
-        for p in sorted(self):
```

me-no-dev, 2017-05-06:
TBH I'm no python coder :) I get around with it and can write some stuff, but I am in no way intimate with it to the point where I know the differences between 2.7 and 3 and how to make things work.
maybe try/catch could do the trick for both versions?

copercini, 2017-09-29:
Solved here: https://github.com/espressif/esp-idf/pull/577 =)

closed 2017-09-29 by copercini