espressif/arduino-esp32 #237: ESP32 ETH PHY support in arduino-esp32

ESP32DE, 2017-02-26:
Hi

can we work together to import this well to the arduino ide?
FYI:  the LAN8720 module are cheap ( 2.50 USD + ) 
on market as ready to go with 50 MHz , pretty good for RMII on esp32.

ping things are done.
https://esp32.com/viewtopic.php?f=12&t=1178&start=10#p5733

robin makes pull request (esp-idf) on the changes to ethernet example 
https://github.com/espressif/esp-idf/tree/master/examples/ethernet
to make the 8720 work

i am honest, have never ported things to arduino ( c, c++.. guy ) 
but would now start with this, ( need start help support ) 
so i ask, is someone here who can help to import this things to arduino IDE?

best wishes
rudi ;-)

me-no-dev, 2017-02-26:
Rudi, I need to have ETH board before I can do anything ETH related. Can not fly blind :) Will request one to be sent to me and will implement ETH to be the same way as WiFi is. Clients will be the same for all obviously (WiFiClient etc). I do have a few things lined up for completion also. It's not productive to not finish the projects you are already ahead on :)
Overall I need to get the board and it should be easy from then on.

ESP32DE, 2017-02-26:
Hi Hristo,

no problem, have ordered few tlk110, lan8720 and other "ready to go eth phy boards" but need few weeks for delivery, hope olimex get the crazy things with "Bundesnetzagentur" clean, they can't sell just in time the very good ESP32 EVB with LAN8710A on Board.

if you can send me an address i will forward to you few things free.
perhabs you can send me a PM over esp32.com if you like this.

btw, if you want to look up preview,
the "pullreq" for esp-idf/demos is here:
https://github.com/espressif/esp-idf/pull/383

let me know how i can do anything

best wishes
rudi ;-)

WALLTECH, 2017-02-26:
Doesn't the ESP32 have built in hardware ethernet? How would that connection work with arduino?

ESP32DE, 2017-02-26:
> Doesn't Arduino Due (SAM3X8E) integrates an EMAC module to implement a 10/100 MAC compatible with the IEEE 802.3 standard?

> The approach of Atmel evolves around to the idea of connect an external Physical layer transceiver (or Ethernet PHY) by Media Independent Interface (MII) or Reduced Media Independent Interface (RMII). Then, they made a comparison between both considering frequency, cost and manufacturers. Finally, they present the implementation using a 10/100 Mbps fast ethernet physical layer TX/FX single chip transceiver (DM9161) set in RMII mode with a 50MHZ crystal connected to the development kit SAM3X-EK using a HTTP demo under BeRTOS

ESP32 have RMII, can ( just in time not ready ) generate 50 MHz clk on Pin.
So ...

ESP32DE, 2017-02-26:
[image]

ESP32DE, 2017-02-26:
any question let me know 

;-)

ESP32DE, 2017-02-26:
MII Version 
https://github.com/elechouse/EMAC-DEMO

arcao, 2017-04-28:
Any news with LAN8720 module? I have one piece and I am thinking about creating library by IDF sample.

me-no-dev, 2017-04-28:
They are trying to get it to work and I know that there has been some success, though I have no idea how much and how close they are to getting it to fully work. I kinda need ethernet right now also

igrr, 2017-04-28:
https://github.com/espressif/esp-idf/commit/90c8bd93e04c768e1c4619c964935ba311a39efa

The waveshare breakouts of LAN8720 do work with IDF now, aside from the issue with clock output going into GPIO0. There is a workaround to break out the clock disable pin though.

ESP32DE, 2017-04-28:
@arcao 
ETH Phy and CAN Lib for Arduino-esp32 is allready done and comes next time as PR to @me-no-dev after testing on ESP-Wrover, ESP32 Rev1 and other we get today.

[image]

best wishes
rudi ;-)

gtortone, 2017-06-24:
Hi, any news about ETH PHY support on ESP32 ?

me-no-dev, 2017-06-25:
I'm running Ethernet currently. Support to come soon ;) Here is some code for you (using TLK110):
```cpp
#include "esp_eth.h"
#include "eth_phy/phy.h"
#include "eth_phy/phy_tlk110.h"
#include "eth_phy/phy_lan8720.h"

#define ETH_PHY_CONF  phy_tlk110_default_ethernet_config
#define ETH_PHY_ADDR  PHY31
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

static eth_config_t eth_config = ETH_PHY_CONF;

static void ethernet_config_gpio(void){
    // RMII data pins are fixed:
    // CRS_DRV = GPIO27
    // TXD0 = GPIO19
    // TXD1 = GPIO22
    // TX_EN = GPIO21
    // RXD0 = GPIO25
    // RXD1 = GPIO26
    // CLK == GPIO0
    phy_rmii_configure_data_interface_pins();
    // MDC is GPIO 23, MDIO is GPIO 18
    phy_rmii_smi_configure_pins(PIN_SMI_MDC, PIN_SMI_MDIO);
}

static void ethernet_power_enable(bool enable){
    pinMode(PIN_PHY_POWER, OUTPUT);
    digitalWrite(PIN_PHY_POWER, enable);
    delay(1);// Allow the power up/down to take effect, min 300us
}

extern void tcpipInit();

void ETH_begin(){
    eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = ethernet_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    eth_config.phy_power_enable = ethernet_power_enable;
    tcpipInit();
    esp_err_t err = esp_eth_init(&eth_config);
    log_i("ETH Init: %d", err);
    if(!err){
        err = esp_eth_enable();
        log_i("ETH Enable: %d", err);
    }
}

IPAddress ETH_localIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.ip.addr);
}

IPAddress ETH_subnetMask()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.netmask.addr);
}

IPAddress ETH_gatewayIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.gw.addr);
}

bool ETH_setHostname(const char * hostname)
{
    return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_ETH, hostname) == 0;
}

bool ETH_fullDuplex()
{
    return eth_config.phy_get_duplex_mode();
}

uint8_t ETH_linkSpeed()
{
    return eth_config.phy_get_speed_mode()?100:10;
}

String ETH_macAddress(void)
{
    uint8_t mac[6];
    char macStr[18] = { 0 };
    esp_eth_get_mac(mac);
    sprintf(macStr, "%02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    return String(macStr);
}

```

gtortone, 2017-06-25:
Thanks a lot! do you have a fork of this repository or you are using an additional library?  I don't find esp_eth.h header file here...

me-no-dev, 2017-06-25:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/ethernet/esp_eth.h

gtortone, 2017-06-26:
Thanks again !
but is it possible to use SDK header file in Arduino IDE ?
any tutorial/howto about this ?

me-no-dev, 2017-06-26:
Have you at least tried the code I pasted above? What makes you think that I would post a code that will not work in Arduino IDE? :D You can use all IDF APIs through Arduino

me-no-dev, 2017-06-26:
well code is missing a couple of things (attaching the event callback and the loop)

gtortone, 2017-06-26:
Thanks again !

me-no-dev, 2017-06-26:
add ```WiFi.onEvent(WiFiEvent);``` in setup() before starting ETH (example above)

tectiv3, 2017-06-26:
Even if it's not wifi?

me-no-dev, 2017-06-26:
yes

me-no-dev, 2017-06-26:
you might need to include WiFi.h ;)

tusharvb19, 2017-07-06:
Sample code for using LAN8720?
TIA.

fesch, 2017-07-10:
@tusharvb19 : Here are my files...

The code for putting a static network configuration **does not work**. Any help would be appreciated ;-)

Ethernet.h
`#ifndef ethernet_h
#define ethernet_h

//#include <inttypes.h>
#include "IPAddress.h"
#include "WiFi.h"
#include "esp_eth.h"
#include "eth_phy/phy.h"
#include "eth_phy/phy_lan8720.h"

#define ETH_PHY_CONF  phy_lan8720_default_ethernet_config
#define ETH_PHY_ADDR  PHY0
//#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

class EthernetClass {
private:
  
public:
  // Initialise the ethernet controller and gain every configuration through DHCP.
  // Returns 0 if the DHCP configuration failed, and 1 if it succeeded
  int begin();
  int begin(IPAddress local_ip, IPAddress subnet, IPAddress gateway);
 
  IPAddress localIP();
  IPAddress subnetMask();
  IPAddress gatewayIP();
  
  bool setHostname(const char * hostname);
  bool isFullDuplex();
  uint8_t getLinkSpeed();
  String getMacAddress();
  
  friend class EthernetClient;
  friend class EthernetServer;
};

class ESP32Ethernet : public EthernetClass {
};

#endif
`

Ethernet.ccp
`#include "Ethernet.h"

extern "C" {
	#include "WiFiGeneric.h"
	//void tcpipInit();
};

static eth_config_t eth_config = ETH_PHY_CONF;

static void ethernet_config_gpio(void){
    // RMII data pins are fixed:
    // CRS_DRV = GPIO27
    // TXD0 = GPIO19
    // TXD1 = GPIO22
    // TX_EN = GPIO21
    // RXD0 = GPIO25
    // RXD1 = GPIO26
    // CLK == GPIO0
    phy_rmii_configure_data_interface_pins();
    // MDC is GPIO 23, MDIO is GPIO 18
    phy_rmii_smi_configure_pins(PIN_SMI_MDC, PIN_SMI_MDIO);
}

void tcpipInit();

int EthernetClass::begin()
{
	eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = ethernet_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    //eth_config.phy_power_enable = ethernet_power_enable;
    tcpipInit();
    esp_err_t err = esp_eth_init(&eth_config);
    if(!err){
        err = esp_eth_enable();
		return 1;
    }
	return 0;
}

int EthernetClass::begin(IPAddress local_ip, IPAddress subnet, IPAddress gateway)
{
	tcpip_adapter_init();
	
	tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
	
	tcpip_adapter_ip_info_t info;
    info.ip.addr = static_cast<uint32_t>(local_ip);
    info.gw.addr = static_cast<uint32_t>(gateway);
    info.netmask.addr = static_cast<uint32_t>(subnet);
	
	tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &info);
	
	esp_event_loop_init(&WiFiGenericClass::_eventCallback, NULL);
	
	if(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &info) == ESP_OK) {
        // continue
    } else {
        return 0;
    }
	
	eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = ethernet_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    //eth_config.phy_power_enable = ethernet_power_enable;
    esp_err_t err = esp_eth_init(&eth_config);
    if(!err){
        err = esp_eth_enable();
		return 1;
    }
	
	return 0;
}

IPAddress EthernetClass::localIP()
{
	tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.ip.addr);
}

IPAddress EthernetClass::subnetMask()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.netmask.addr);

}

IPAddress EthernetClass::gatewayIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.gw.addr);
}

bool EthernetClass::setHostname(const char * hostname)
{
    return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_ETH, hostname) == 0;
}

bool EthernetClass::isFullDuplex()
{
    return eth_config.phy_get_duplex_mode();
}

uint8_t EthernetClass::getLinkSpeed()
{
    return eth_config.phy_get_speed_mode()?100:10;
}

String EthernetClass::getMacAddress()
{
    uint8_t mac[6];
    char macStr[18] = { 0 };
    esp_eth_get_mac(mac);
    sprintf(macStr, "%02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    return String(macStr);
}

#if !defined(NO_GLOBAL_INSTANCES) && !defined(NO_GLOBAL_ETHERNET)
EthernetClass Ethernet;
#endif
`

tusharvb19, 2017-07-11:
It will be great if you give sample code like webserver ,etc to use the above files.
I am a newbie.
TIA.

fesch, 2017-07-11:
Take this library: [WebServer](http://www.fisch.lu/junk/WebServer.zip)

To be copied into: Arduino\hardware\expressif\esp32\libraries

tusharvb19, 2017-07-11:
Thank you and all the best...
How about using Wiznet5100 with esp32 and arduino environment?
Have you tried it?

tusharvb19, 2017-07-11:
Got this error for after editing the sample code which is-
#include <ESP32WebServer.h>

const char* ssid = "........";
const char* password = "........";

ESP32WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
 // WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
 // while (WiFi.status() != WL_CONNECTED) {
   // delay(500);
    //Serial.print(".");
 // }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
 // Serial.println(WiFi.localIP());

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}

Got this when compiled-
In member function 'void ESP32WebServer::close():
error: 'class WiFiServer' has no member named 'close'

   _server.close();

In member function 'void ESP32WebServer::sendContent_P(const char*, size_t):
error: 'class WiFiClient' has no member named 'write_P'

   _currentClient.write_P(content, size);

fesch, 2017-07-11:
No, my ESP32-EVB board from Olimex has no Wiznet5100 ...

Do you have ESP32-dev things installed?

tusharvb19, 2017-07-12:
yes I am using ESP32 as Dev module

clivepengelly, 2017-08-13:
Hi guys. Please could you give a step by step process of adding this code into the Arduino IDE?
Many thanks.

fesch, 2017-08-13:
Everything is in my fork ...

me-no-dev, 2017-08-17:
Ethernet is implemented ;) closing this now

closed 2017-08-17 by me-no-dev

copercini, 2017-11-14:
@fesch Is the static IP example of your fork working?

fesch, 2017-11-14:
@copercini Yes, it is working. The problem was not in Arduino-ESP but in the precompiled ESP libraries (as far as I remember ;))

jatayu86, 2017-11-16:
Hi,I am trying to find samples for MQTT over TLS using LAN8720 on esp32. Is it supported yet ,and if yes pointers to a sample code would be really helpful :)

fesch, 2017-11-16:
I'm not shure, I need to re-test, but I think I got it running. But I'm not shure anymore ... checking later ...

dvvrao, 2018-08-08:
Can any one give me an example code for 'MQTT client' that uses 'ethernet client' for Olimex ESP32 - EVB.

xorbit, 2018-08-28:
Why is this not pulled into the main repo?  As long as it is in someone else's fork, you can't claim it's "implemented"!

aerodolphin, 2021-01-09:
I would like to use the waveshare lan8720 in ESP32-WROOM-32D to communicate with Home Assistant in MQTT.
Can someone confirm if this will work?

fesch, 2021-01-10:
If there is a connection, I don't see why it should not work...

fesch, 2021-01-10:
I have this board up and running pushing data to my MQTT server. I uses the 7810 chip, but I suppose it's quite similar to the 8720.
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

aerodolphin, 2021-01-10:
@fesch, Could work, but just to make sure, I will change the hardware to be equal the ESP32-POE since I can have an example that really works. Can you please inform what is the version of your OLIMEX ESP32-POE board? I can base me on that circuit to make my custom board.

Are you using arduino IDE MQTT on that board? If yes, can you show the code?

fesch, 2021-01-10:
I just checked: It's the version with galvanic separation that I was using, but the other one works in the exact same way:
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

I'm using Arduino IDE with PubSub client fir MQTT.

[MQTT_ESP32_ETH_example.txt](https://github.com/espressif/arduino-esp32/files/5792470/MQTT_ESP32_ETH_example.txt)

aerodolphin, 2021-01-10:
On your example, "foo.some" is the topic name?

fesch, 2021-01-10:
yes ;)

aerodolphin, 2021-01-10:
@fesch, thank you. I think I will change my schematic from the lan8720 to the Olimex ESP32-POE, because that way I have arduino code that is working.

Maybe this code will work with the waveshare lan8720, but only testing to really know. I don't have any available.