espressif/arduino-esp32 #241: Wifi Init crashes with latest build

palomarcoder, 2017-02-28:
I am running the Watson IoT Foundation esp32 client at https://developer.ibm.com/recipes/tutorials/connect-an-esp32-to-the-watson-iot-platform/

For the past couple of months (since Mid-December) it has been working fine.  It is a pretty straightforward initialization of wifi and use of the pubsub client.  

Today I updated to the latest build with all the new IDF and wifi fixes, and recompiled, re-flashed.  Now I get an error and a CPU halt during the wifi init.  Everything else is the same - same hardware, same software.  I tried re-flashing a couple of times.  Has anyone else see this?

E (22608) wifi: wifi_init 1198 ret=4363
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x400839c7
Guru Meditation Error: Core  1 panic'ed (abort)

Backtrace: 0x40008155:0x3ffcc640 0x40007d16:0x3ffcc660 0x40093b88:0x3ffcc6a0 0x400f5cb4:0x3ffcc6f0 0x400ef531:0x3ffcc710 0x400d1c44:0x3ffcc740 0x400d1c52:0x3ffcc770 0x400d1cbe:0x3ffcc7a0 0x400d1d97:0x3ffcc7c0 0x400d0f4a:0x3ffcc8c0 0x400d1785:0x3ffcc8f0 0x4010bdca:0x3ffcc910

CPU halted.

copercini, 2017-02-28:
Please, see: https://github.com/espressif/arduino-esp32/issues/207

palomarcoder, 2017-03-01:
@copercini That worked, thanks!

esptool --port COM3 erase_flash

closed 2017-03-01 by palomarcoder