espressif/arduino-esp32 #242: WifiScan:: Weak signal reception compared to ESP8266

tobozo, 2017-02-28:
Last month I received [this ESP32S dev board](https://www.aliexpress.com/store/product/NodeMCU-32S-Lua-WiFi-LOT-Development-Board-Dual-Core-ESP-32S-Wireless-WIFI-Module-Serial-Port/1829717_32758824128.html) from aliexpress and ran the WifiScan.ino example but the scan results are much worse than what I get when I scan with an ESP8266.

Instead of getting the usual amount of SSIDs (between 50 and 80 in my area), I only get two or three, and sometimes nothing. And the ESP32S is heating a bit more than the ESP8266.

There are no interferences, the ino code is similar, both ESP8266 and ESP32S had their code compiled and tested under the same circumstances.

After posting on reddit, someone [suggested](https://www.reddit.com/r/esp32/comments/5mea5j/esp32s_nodemcu_weak_reception/) it could be a design error, so I ordered another model from aliexpress.

Today I've Just received [this other board](https://aliexpress.com/item/ESP-32-ESP-32S-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP32-Board/32791060025.html) and it's exactly the same problem, displaying only 3 AP where the ESP8266 shows 50.

Now I'm wondering if there's some kind of setting that sets the Wifi to low power mode, or if I'm just unlucky in my model choices.

me-no-dev, 2017-02-28:
What you describe is rather strange. I have a few 8266 and 32 boards here and will have a look. Though my area is not that noisy.

chenheli, 2017-03-01:
ESP32 wifi scan only outputs APs using channel 1.

tobozo, 2017-03-01:
More observations: when the ESP8266 sees an AP with -40db, both ESP32 models see the same AP with an RSSI between -60 and -70db. 
There's an average 20db difference. 
I took a photo of both models, maybe I've just bought a shitty clone or something.
I had some pins removed off the first model (the ones too close to the antenna) with no effect whatsoever.

[image]

me-no-dev, 2017-03-01:
yeah,... what a horrible design... see how there is a board under the antenna? that should not be there at all.
[image]

tobozo, 2017-03-02:
Yep free space is recommended, but 20db difference is a lot, it is the equivalent of having metal shielding. i.e. laptop casings can attenuate the signal more than 10db , but a piece of PCB shouldn't attenuate that much, otherwise Wemos D1 mini and other dev kits have a problem ?

Also the second model is [ESP32 DEVKIT V1](https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP32), aka "DOIT/SmartArduino ESP32 DevKit" is a recommended model from the sp32.net page, can I assume this one has a placement reasonable enough not to attenuate the signal by 20db ?

I've ordered a third model, same as the one you posted here.

tobozo, 2017-03-03:
I just tried yet another ESP32 module, the [ESPea32](http://wiki.aprbrother.com/wiki/ESPea32), and it behaves the same way.

@me-no-dev are you positive dev kits for ESP8266 and ESP32 should have the same reception quality for Wifi? 
I'm starting to believe they don't, or that something hasn't been tuned-up yet somewhere in the toolchain.

me-no-dev, 2017-03-05:
@tobozo Will do my best to test this next week ;) I will be on a trip to help customers, but I'll have ESP8266 and ESP32 boards with me. Please ping back here around tuesday/wednesday so I do not forget :)

tobozo, 2017-03-05:
(sorry for the misclose)

I just followed the tutorial to install [micropython for esp32](https://github.com/micropython/micropython-esp32), then flashed the ESP32 and ran the wifi scan micropython code just to compare how many AP are returned.

I've run 10 scans per situation, here are the average values:

With the ESP32:
  - Arduino IDE finds 15 AP
  - Micropython finds 30 AP 

With the ESP8266:
  - Arduino IDE finds 60 AP
  - Micropython finds 60 AP

me-no-dev, 2017-03-05:
interesting enough, there isn't any limitation imposed by the WiFi library in Arduino to how many APs are returned. Code is taken from the ESP8266 lib and only method calls are adapted. Maybe something default?

tobozo, 2017-03-05:
I just found out that the gap between micropython and arduino is only observable in micropython after I `make erase` and `make deploy`.

Flashing with Arduino seems to cancel that effect.

If I `make deploy` micropython after a successful Arduino flash, there's no observable gap: the micropython library scan results give the same average as with Arduino.

Does it make sense the `make erase` somehow resets something that would allow me to enumerate more access points when using micropython ?

me-no-dev, 2017-03-05:
There might be a setting that get's stored in NVS based on some config value maybe... and then does not get overridden unless you erase the flash

tobozo, 2017-03-13:
Hey @me-no-dev I just received the same model as on your picture, and the quality/behaviour is the same as with the other models.

What settings should I be looking for? Is that part of Arduino, ESP-IDF or something else I should recompile?

tobozo, 2017-03-14:
Could this be related 207#issuecomment-286378285

kriswiner, 2017-03-20:
I'm curious how my ESP32 development board would fare in your tests. I only have one or two other wifi nodes close to my house (neighbors) but in my tests the ESP8285 and ESP32 behave about the same. Of course I don't use the ESP modules, I prefer to design my own circuit and I wonder if such a hardware design difference could be responsible for the difference in reception behavior you are seeing?

https://www.tindie.com/products/onehorse/esp32-development-board/?pt=full_prod_search

me-no-dev, 2017-03-20:
Ok so I got around to do some tests today and can say that my ESB32 boards actually perform better than my ESP8266 boards. Signal quality is about the same (in terms of dB) but the ESP32 board finds much more networks. Now I do not have an env that has more than 10-15 APs in range, but the ESP32 get's most while the ESP8266 only finds like 4-5 most of the time.

tobozo, 2017-03-20:
I haven't updated my sdk for a while so I'll git pull & test again later today. I've also ordered [this model](https://www.tindie.com/products/lspoplove/d-duino-32/) and probably will try @kriswiner 's suggestion too.

tobozo, 2017-03-20:
I wrote a [test sketch](https://github.com/tobozo/ESPWifiBench/blob/master/ESPWifiBench.ino) and collected [some data]((http://pastebin.com/dLUUZZME)) from 4 different models o ESP32 and 2 different models of ESP8266.

- ESPEA32 (snap thing from tindie)
- ESP-WROOM32 devkit (shitty design from aliexpress)
- ESP32-S Dev kit (another shitty design, worse)
- ESP32-WRoom Core Board v2 (best design so far)
- ESP8266 NodeMCU V3.0
- ESP8266 Wemos Mini

I'm not sure what this all means, is that test even valid ?

kriswiner, 2017-03-20:
I ran the test on my board and was able tod etect just one AP (my local router); I got 93/100 hits with an average RSI of -29. Not sure what this means either. I often can detec the neighbors AP also, not this time.

tobozo, 2017-03-20:
Would this test be more significant in a busy area?
But looking again at the collected data, RSI is the same on ESP32 and ESP8266.
The only notable difference is the amount of acces points returned by the scan.

me-no-dev, 2017-04-03:
Latest commits include an updated api for scanning where you can define max time in ms to search each channel. Number returned results change when the default 300ms is increased, but scan takes longer too. Give it a go ;) I would like to close this one.

tobozo, 2017-04-04:
Just updated the SDK (git pull + tools/get.exe) and ran the [test sketch](https://github.com/tobozo/ESPWifiBench/blob/master/ESPWifiBench.ino)

Active Scan
------------
- `WiFi.scanNetworks(false, true, false, 1500);`  gives slightly better results than 'WiFi.scanNetworks(false, true);` but still far behind what an ESP8266 detects. I had to go up to 1500 (increasing by 100 on every test) to measure a noticeable change.

Passive Scan
-------------
- `WiFi.scanNetworks(false, true, true, xxxx);`  : whatever the value I give to xxxx it returns zilch.

Am I using this right ?

tobozo, 2017-07-18:
bump

This is still a problem: in an area where any ESP8266 model easily detects a 100 unique SSID in one single WiFiScan(), all ESP32 models I have only return an average 15 SSID (even the ESP32 Wrover Kit V3).

This time I've also tried [this script](https://github.com/VALERE91/ESP32_WifiScan) and can conclude it's not Arduino related since this one does not use the IDE and features the same symptoms.

@me-no-dev @copercini how can I help resolve this issue? I have a bunch of ESP32/ESP8266 dev boards I can test with..

copercini, 2017-07-21:
@tobozo Try connect to a network before scan, like replace [this line of example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino#L13) by `WiFi.begin("router", "password");`

tobozo, 2017-07-21:
@copercini interesting, now it returns a bigger amount of SSIDs, but after a few scans:

`
192.168.1.12
Setup done
round #0 .. 64 results
round #1 .. 52 results
round #2 .. 53 results
round #3 .. 54 results
round #4 .. 61 results
round #5 .. no networks found
round #6 .. 64 results
round #7 .. 59 results
round #8 .. 54 results
round #9 .. E (54393) wifi: pm.c 956
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE

`

tobozo, 2017-07-21:
@copercini I'll close this issue since it's kinda sorted out, the watchdog thing only happens on the wrover kit anyway.

What I'm wondering is why do I have to `WiFi.begin()` (no arguments required) on ESP32 while it's not required on ESP8266, and also why does it matter?

copercini, 2017-07-21:
Wrover v3 have a flash bug which does not work fine at 80Mhz.

--------------
For me it works:
``` cpp
  WiFi.disconnect();
  WiFi.begin("lolllool","loooooooool"); //Can be a inexistent network
```
But this doesn't works:
``` cpp
  WiFi.disconnect();
  WiFi.begin();
```
So my guess it's something inside this function (RF calibration maybe): https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi.h#L659 which I can't see the source code. 

That's all I know.

tobozo, 2017-07-21:
@copercini  That's good to know, thanks for solving that long time mystery :+1:

copercini, 2017-07-21:
For me it's still a bug!

The Espressif guy should take a look on the root cause
or at least edit the example with the WiFi.begin workaround

tobozo, 2017-07-21:
this doesn't work for me
```
  WiFi.disconnect();
  WiFi.begin();
```
however without WiFi.disconnect() this works without arguments
```
  WiFi.begin();
```

> For me it's still a bug!

should I reopen this ticket then ?

copercini, 2017-08-18:
Will reopen due it's still not fixed

Testato, 2017-11-08:
I tried many core version and this Bug is present on some but not on other.

I can reproduce it, for example:
- on this commit it is not present
https://github.com/espressif/arduino-esp32/commit/8dc0c86457d913de9147054638a2b449b6398d1b

- on this commit it is present (so i must use WiFi.begin("bug","bug") before a scan)
https://github.com/espressif/arduino-esp32/commit/3261348e8132b46006874455be9a8cf69dbf14de

nevercast, 2017-11-22:
Do you mind doing a git bisect to see if you can find the offending commit? @Testato

Testato, 2017-11-23:
I'm not a Git expert so I never used bisect.

This bug is very important, it is present in the actual master version, i hope someone can solve it

me-no-dev, 2017-11-23:
The reality is that those functions are in the closed wifi lib (no source to look at). BTW have you tried ```WiFi.mode(WIFI_STA);``` before going through scan?

tablatronix, 2018-06-30:
What is the status of this issue?

tobozo, 2018-06-30:
@tablatronix I can't observe the gap anymore, whether I use:
```
  WiFi.disconnect();
  WiFi.begin();
```
... or ... 
```
  WiFi.begin("blahblah", "blahblah");
```
... the wrover kit I'm using for testing finds the same amount of AP's.

Closing this as resolved \o/

closed 2018-06-30 by tobozo

cgmckeever, 2020-08-17:
@tobozo Sorry for dragging up an old issue .. But this seems to be the most conversed thread on this topic. May I ask what you found? Im finding that my ESP32's wifi performance are extremely poor compared to my ESP8266. Was that the consensus you had? And it seems that the Wrover ESP32 was more inline with the ESP8266 performance? Really appreciate it

tablatronix, 2020-08-17:
@cgmckeever Ill try to test a comparison tomorrow, I have a network with over 30 aps, and both an esp8266 and esp32 variant of my own pcb I use.

Just to clarify you are strictly referring to wifiscan when you say "performance"

cgmckeever, 2020-08-17:
@tablatronix Thanks for the ping! Actually, I mean just crappy wifi range in general. I cant get an ESP32 to connect to a AP 10 feet away, let alone through any obstacle

i0x71, 2020-09-14:
I just got nodemcu with esp32-wroom-32 and tested with an older nodemcu esp8266mod side by side
Wifi signal
ESP32 19%
ESP8266 59%
Both are located same distance away from router and using same power supply.
This is the esp32 board
https://www.amazon.ca/WayinTop-Development-ESP-WROOM-32-Bluetooth-Microcontroller/dp/B086ZMDB7H/ref=sr_1_2_sspa?dchild=1&keywords=esp32&qid=1600105665&sr=8-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFZRVlMTzcwNUtNNFYmZW5jcnlwdGVkSWQ9QTAxMTYyMDEySzk5QzhZWFBXNkRQJmVuY3J5cHRlZEFkSWQ9QTA2Nzc1ODEyRjRDQUVOM1NFUDZTJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

andi07x, 2020-10-04:
I had the same problem with my ESP32 boards and Arduino IDE. After some research I found a hint that the WiFi is in power save mode. After adding

`
#include <esp_pm.h>
#include <esp_wifi.h>
#include <esp_wifi_types.h>

...
void setup() {
   WiFi.mode(WIFI_STA);
   esp_wifi_set_ps( WIFI_PS_NONE );

...
`
to my code the performance was comparable to the ESP8266. Hope this helps somebody.

leo-hofman, 2023-03-16:
I changed the following in the YAML file and it appears to help:
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.178.62
    gateway: 192.168.178.1
    subnet: 255.255.255.0
  output_power: 20dB
  power_save_mode: LIGHT
```