espressif/arduino-esp32 #243: Integrte BLEPeripheral, instead of custom API library

vshymanskyy, 2017-02-28:
This is just an idea/recommendation, to implement hw layer for excellent BLEPeripheral library, instead of inventing own API: https://github.com/sandeepmistry/arduino-BLEPeripheral
I already asked if it can be a feasible solution, and got a positive response: https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/126

Thx guys!

vshymanskyy, 2017-02-28:
As I mentioned, ```This is just an idea/recommendation```. ESP32 BLE has no Arduino wrapper currently, and implementing it with BLEPeripheral can be much easier and quicker.
I just wanted to inform you. Thanks.

me-no-dev, 2017-02-28:
I'm sure there will be tons of libraries. The reason I have not looked at Arduino implementations yet, is because BLE is not finished and is the most worked on aspect on the ESP32. Instead I implemented an API for developers to start using BT and give people options. The example included shows how to start, stop and attach stuff to BT.

vshymanskyy, 2017-03-02:
Great, I have no issue with that (at all) ;)