espressif/arduino-esp32 #244: WiFi.onEvent(); causes ESP32 to constantly reboot

hicotton02, 2017-02-28:
I have tried various sketches, including [WiFiClientEvents](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino) and it seems that this line:

`WiFi.onEvent(WiFiEvent);`

causes my ESP32 to constantly reboot. here is what the serial output shows:

`Doneets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1848
load:0x40078000,len:6800
load:0x40080000,len:252
entry 0x40080034
Display Power Mode: 0x9C
MADCTL Mode: 0x48
Pixel Format: 0x5
Image Format: 0x0
Self Diagnostic: 0xC0`

it just keeps repeating that

me-no-dev, 2017-02-28:
you probably do not have enough power going to the ESP. WiFi sucks juice when powering.
Closing this as it is not a library issue, but you can keep commenting here and I'll reply :)

closed 2017-02-28 by me-no-dev

hicotton02, 2017-02-28:
I appreciate that!

I have a 5A power supply that I will add to it, since the USB power from my computer seems to not be enough.

me-no-dev, 2017-03-01:
could be the usb power, could be the cable or combination. Shitty cables and usb power bring issues and are the main cause of those. 5A is overkill though :) 500-600mAh stable will do. You could also add larger cap on the 5V in rail.

Krezalis, 2020-04-21:
The sane problem:
`D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
`

atanisoft, 2020-04-21:
@Krezalis please file a new issue and follow the issue template