espressif/arduino-esp32 #246: More than 32 bytes cannot be written via I2C

arcao, 2017-03-03:
When I try to write more than 32 bytes (2B for address + >30B data) via I2C to [AT24C256](http://www.atmel.com/images/doc5121.pdf) EEPROM device, next written byte is NAKed. I guess it's because a new transaction is started and EEPROM device thinks that is a new **(PAGE) WRITE** command. AT24C256 supports up to 64 bytes to be transferred with that command.

[image]

_Note: Between **PAGE WRITE** and any other command has to be a delay (max 5ms by [datasheet](http://www.atmel.com/images/doc5121.pdf)) to be data physically written, so this is the reason of NAK._

**Test code:**
```cpp
#include <Arduino.h>
#include <Wire.h>

#define PIN_SDA  21  // I2C SDA
#define PIN_SCL  22  // I2C SCL

void setup() {
  int i2cdevice = 0x50;  // 24C256 EEPROM address with A0=0, A1=0, A2=0
  int addr;
  int res;

  Serial.begin(115200);
  Serial.println("Initializing I2C");
  Wire.begin(PIN_SDA, PIN_SCL);

  addr = 0;

  Wire.beginTransmission(i2cdevice);
  Wire.write(addr >> 8);
  Wire.write(addr & 0xff);
  
  for (uint8_t i = 1; i <= 64; i++) {
    Wire.write(i);
  }
  res = Wire.endTransmission();
  if (res != 0) {
    Serial.println("Write error!");
  }

}

void loop() {}
```

me-no-dev, 2017-03-05:
@arcao if you have time, you can try to play with the i2c code [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L167-L175) and see if not sending the address again on the second iteration of the loop will help you.

arcao, 2017-03-05:
I will try it later today. I'm not at home just now.

kriswiner, 2017-03-20:
The 32 byte limit is a common hard coded limit used in the Wire library. You can change this in the Wire.h library somewhere but I usually just accept this and buffer data writes in 32 byte chunks to keep the code platform agnostic.

lonerzzz, 2017-08-25:
@me-no-dev @kriswiner Does it make sense to add an API to set the Wire library to change the byte limit from a set default? Is there any known implementation where this has been done before?

kriswiner, 2017-08-25:
Just change it yourself. 32 bytes is standard for avr arduinos. I have
bumped mine up to 64, which is enough for most tasks. You can also just
read/write twice!

copercini, 2018-09-21:
Fixed with RC1

closed 2018-09-21 by copercini