espressif/arduino-esp32 #249: Websocket support

williamesp2015, 2017-03-04:
It would be very nice if ESP32 provide websocket support like ESP8266 (https://github.com/Links2004/arduinoWebSockets/tree/master/examples)

me-no-dev, 2017-03-05:
not an issue though :) maybe ping @Links2004 ?

closed 2017-03-05 by me-no-dev

Links2004, 2017-03-05:
the port isnt that hard since the WiFiClient and WiFiServer interface is likely the same,
but I currently not on a ESP32 for testing.
In theory its only some ```ifdefs``` in 
https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSockets.h

me-no-dev, 2017-03-05:
@Links2004 nice to see you again mate ;) it's been a while :)

Links2004, 2017-03-05:
Linking issue:
https://github.com/Links2004/arduinoWebSockets/issues/161

@me-no-dev yes, loots todo right now, and unfortunately less to no time for all the open source projects where  I working on.

vindolin, 2017-05-05:
@Links2004 I gladly send you one :)