espressif/arduino-esp32 #250: get.py bug on Linux-3.16.0-4-686-pae-i686-with-debian-8.7, python 2.7.9

melehin, 2017-03-04:
```bash
~/Arduino/hardware/espressif/esp32/tools$ python get.py 
System: Linux, Info: Linux-3.16.0-4-686-pae-i686-with-debian-8.7
Platform: i686-pc-linux-gnu
Downloading xtensa-esp32-elf-linux32-1.22.0-59.tar.gz
Done
Extracting xtensa-esp32-elf-linux32-1.22.0-59.tar.gz
Downloading esptool-fe69994-linux32.tar.gz
Done
Extracting esptool-fe69994-linux32.tar.gz
Traceback (most recent call last):
  File "get.py", line 138, in <module>
    get_tool(tool)
  File "get.py", line 103, in get_tool
    unpack(local_path, '.')
  File "get.py", line 66, in unpack
    rename_to = re.match(r'^([a-z][^\-]*\-*)+', dirname).group(0).strip('-')
AttributeError: 'NoneType' object has no attribute 'group'
```

me-no-dev, 2017-03-05:
hmm... are you positive this is python 2.7? rather strange error

melehin, 2017-03-05:
yes, 2.7
```bash
~/Arduino/hardware/espressif/esp32/tools$ python --version
Python 2.7.9
~/Arduino/hardware/espressif/esp32/tools$ python get.py 
System: Linux, Info: Linux-3.16.0-4-686-pae-i686-with-debian-8.7
Platform: i686-pc-linux-gnu
Tool xtensa-esp32-elf-linux32-1.22.0-59.tar.gz already downloaded
Extracting xtensa-esp32-elf-linux32-1.22.0-59.tar.gz
Tool esptool-fe69994-linux32.tar.gz already downloaded
Extracting esptool-fe69994-linux32.tar.gz
Traceback (most recent call last):
  File "get.py", line 138, in <module>
    get_tool(tool)
  File "get.py", line 103, in get_tool
    unpack(local_path, '.')
  File "get.py", line 66, in unpack
    rename_to = re.match(r'^([a-z][^\-]*\-*)+', dirname).group(0).strip('-')
AttributeError: 'NoneType' object has no attribute 'group'
```

melehin, 2017-03-05:
1. I change dir to dist folder
2. untar esptool-fe69994-linux32.tar.gz
3. ls command show me
```bash
-rwxr-xr-x 1 login login 3823588 Jan 19 03:47 esptool
-rwxr-xr-x 1 login login      379 Jan 19 03:47 ._esptool
```
**._esptool** !!!!
file ._esptool is "AppleDouble encoded Macintosh file" tar'ed to archive by developer

4. I set print(dirname) in get.py before re.match and saw **._esptool** that not satisfy regex

me-no-dev, 2017-03-05:
Good find! Will fix in next version that is coming up

me-no-dev, 2017-04-03:
back to python on linux because of issues with python to binary conversion and different distros.

closed 2017-04-03 by me-no-dev