espressif/arduino-esp32 #252: missing esptool.exe

ouki-wang, 2017-03-06:
the lastest script using esptool.exe。but i can not find it

me-no-dev, 2017-03-10:
run get.exe. It's explained in the guide

closed 2017-03-10 by me-no-dev

eric-schleicher, 2018-01-07:
This should be re-opened since running get.exe doesn't work for many since the SSL cert make it not work on system with good ssl hygene.

Is there a workaround?

shirish47, 2018-05-02:
hey, how do I get esptool.exe? it only has esptool.py and I get esptool.exe does not exist in windows

beegee-tokyo, 2018-05-03:
In the [Wiki](https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/) I listed all download links including for esptool.exe

flywire, 2019-07-07:
Very poor process, [wiki](https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/) links to outdated software. Also see https://github.com/espressif/esptool/issues/181#issuecomment-508980616