espressif/arduino-esp32 #254: Successful TLS connection Event?

forthlightning, 2017-03-09:
Hi, publishing to azure requires successful tls connection, so I need to know that one has been successfully created. 

I added a flag in the ssl_client.cpp in WiFiClientSecure. Seems like a hacky fix, so I am thinking of pushing a wifi event with the esp_send_event() method and have my event handler listen. 

I think others will also like a programmatic way to check that TLS connection is successul. Or maybe there already is one!

WALLTECH, 2017-03-09:
I agree, this would be useful

copercini, 2017-03-10:
Why don't use `client.connected()`  ?

me-no-dev, 2017-03-11:
@copercini because not being able to connect to the server and handshake fail are two different things.

VojtechBartoska, 2023-12-08:
Closing this ticket as expired.