espressif/arduino-esp32 #259: Maple ESP32 SD/MMC Tests failed

LordJakson, 2017-03-11:
I got recently the Maple ESP32 board from AnalogLamb. 
http://www.analoglamb.com/product/maple-esp32/

It use the ESP-WROOM32 module, has an micro SD slot and a plug for a Battery.

So far as I read it the SD is connected to the Pins 2/4/12-15 Like described in the SDMMC_Test.
https://hackaday.io/project/19362-maple-esp32-esp32-wifi-bt-board-with-micro-sd
https://cdn.hackaday.io/files/19362828383808/Schematic__Maple%20ESP32%20R1.pdf

If I run the SDMMC_Test I got this response:

```
E (601) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109
Card Mount Failed
```

Is this some bug in the SD/MMC implementation or is there something other wrong

igrr, 2017-03-11:
I have checked the schematics and the board has no pull-up resistors on IOs 2, 4, 12, 13 which are connected to SD card pins. All these IOs need to be pulled up for reliable operation (even though pulling up IO2 can affect entry into download mode, and pulling up IO12 will set the internal VDD_SDIO regulator of the chip to 1.8V).
SCR is the first initialization command which uses data pin (SD card D0, GPIO2), so you may get a bit further by adding a 10k pullup resistor on that pin.

Staubgeborener, 2017-06-28:
Do i need 10k pull-up or 1k pull-up? Here stands 10k https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#hardware
But in the SDMMC_Test (Arduino IDE) stands 1k. And only for IO 2.

melvinpmathew, 2017-07-07:
Were you able to get this working? I am facing the same issues.

Staubgeborener, 2017-07-07:
Well, no. Not really. I integrated Arduino as a component in the esp-idf, so i could use the SPI.h and SD.h librarys (oherwise the SPI driver is currently not working). I can use my SD Card now (for example with the sd card example in the arduino IDE), but with the limitation that only D0 works for the data transfer. That's really bad for the data transfer rate. It's a compromise. But a sick one. I'm also interested how this example can work with all data pins (D0, D1, D2, D3). Maybe the following threads in the esp32.com forum can help: 
https://esp32.com/viewtopic.php?f=13&t=2366
https://esp32.com/viewtopic.php?f=2&t=1845&p=9159#p9155

I will try this in the next days. If it's working on your side, please write me @melvinpmathew!

williamesp2015, 2017-07-08:
@LordJakson Could you get data from SD card using Maple ESp32 R1? The pinouts are here but no pull-up resistors and only GPIO15 is connected to pinheaders and can not solder resistor. 
Maple ESP32 R1 pin outs
DAT2(X) 12
CD/DAT3 (CS) 13
CMD(DI) 15
CLK(SCLK) 14
DAT0(DO) 2
DAT1(X) 4
DET () 16
I tried to do 
gpio_pullup_en(GPIO_NUM_13); //Internal Pull up 13
gpio_pullup_en(GPIO_NUM_15); //Internal Pull up 15
gpio_pullup_en(GPIO_NUM_14); //Internal Pull up 14
gpio_pullup_en(GPIO_NUM_2); //Internal Pull up 2
but there was no success.

melvinpmathew, 2017-07-10:
@williamesp2015 @Staubgeborener I am using ESP32 Wroom device and was able to get the SD test work fine on SPI. It's the SD_MMC test which is not working correctly for me. The above question was related to SD_MMC.
(SD runs on SPI, and SD_MMC uses the **SDMMC hardware bus on the ESP32**)

I am still getting the error  (563) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109. So sure that its some electrical issue, so tried pullup resistors(10k,1k) on the suggested pins and still no progress. Only change I notice is that the error changes from 0x107 and 0x109.

Any more suggestions anyone?

thezenox, 2017-07-28:
I'm trying to get it to work as well, After a few tests with different resistors i get the example it working but its not very stable. It only works 40% of the time. Sometimes only parts are working and the other commands throw 107/109.
I use 15k pullup resistors on all pins exept CLK. DIO2 and DIO12 are connecteted via pullup to IO27 what is switched hight after bootup followed by 1.5 seconds delay. so flashing is possilbe on the fly and Flash Voltage is at 3,3V.
Related: https://www.esp32.com/viewtopic.php?f=2&t=849

Looks good:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
SD_MMC Card Type: SDHC
SD_MMC Card Size: 7476MB
Listing directory: /
  FILE: /test.txt  SIZE: 1048576
  FILE: /foo.txt  SIZE: 13
  DIR : /System Volume Information
Creating Dir: /mydir
Dir created
Listing directory: /
  FILE: /test.txt  SIZE: 1048576
  FILE: /foo.txt  SIZE: 13
  DIR : /System Volume Information
  DIR : /mydir
Removing Dir: /mydir
Dir removed
Listing directory: /
  FILE: /test.txt  SIZE: 1048576
  FILE: /foo.txt  SIZE: 13
  DIR : /System Volume Information
Listing directory: /System Volume Information
  FILE: /System Volume Information/IndexerVolumeGuid  SIZE: 76
  FILE: /System Volume Information/WPSettings.dat  SIZE: 12
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 1440 ms
1048576 bytes written for 4299 ms`

Looks not that good:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
Starting
SD_MMC Card Type: SDHC
SD_MMC Card Size: 7476MB
Listing directory: /
  FILE: /test.txt  SIZE: 0
  DIR : /System Volume Information
  FILE: /foo.txt  SIZE: 13
E (17226) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (17227) diskio_sdmmc: sdmmc_read_blocks failed (265)
Creating Dir: /mydir
E (18231) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (18231) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (18231) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (18316) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (18317) diskio_sdmmc: sdmmc_read_blocks failed (265)
mkdir failed
Listing directory: /
E (19320) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (19320) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (19320) diskio_sdmmc: sdmmc_read_blocks failed (263)
Removing Dir: /mydir
E (19330) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (19333) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (20339) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (20339) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (20339) diskio_sdmmc: sdmmc_read_blocks failed (263)
rmdir failed
Listing directory: /
  FILE: /test.txt  SIZE: 0
  DIR : /System Volume Information
Listing directory: /System Volume Information
  FILE: /System Volume Information/IndexerVolumeGuid  SIZE: 76
  FILE: /System Volume Information/WPSettings.dat  SIZE: 12
  FILE: /foo.txt  SIZE: 13
E (20371) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (20376) diskio_sdmmc: sdmmc_read_blocks failed (265)
Writing file: /hello.txt
E (21384) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (21384) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (21384) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (21400) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (21401) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (22403) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (22403) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (22403) diskio_sdmmc: sdmmc_read_blocks failed (263)
Write failed
Appending to file: /hello.txt
E (22487) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (22487) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (23487) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (23487) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (23487) diskio_sdmmc: sdmmc_read_blocks failed (263)
Append failed
Reading file: /hello.txt
E (23499) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (23502) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (24507) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (24507) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (24507) diskio_sdmmc: sdmmc_read_blocks failed (263)
Failed to open file for reading
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
E (24541) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (24541) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (25542) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (25542) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (25542) diskio_sdmmc: sdmmc_read_blocks failed (263)
Rename failed
Reading file: /foo.txt
E (25555) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x109
E (25557) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (26562) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (26562) sdmmc_cmd: sdmmc_read_sectors: sdmmc_send_cmd returned 0x107
E (26562) diskio_sdmmc: sdmmc_read_blocks failed (263)
Failed to open file for reading
0 bytes read for 0 ms
1048576 bytes written for 5012 ms
`

Staubgeborener, 2017-07-31:
Thanks. But still no success here, even with 15k pull up. No 1-bit, no 4-bit. Just in case, coul you post a picture of your circuit @thezenox ?

Staubgeborener, 2017-07-31:
Thanks. But still no success here, even with 15k pull ups. No 4-bit, even no 1-bit. Just in case, could you post a picture of your circuit @thezenox ?

me-no-dev, 2017-08-01:
Looking at the schematic myself, maybe @williamesp2015 has the best workaround (SPI+SD)

thezenox, 2017-08-07:
This is how  I connected it. I use the ESP32_core_board_V2
[image]
The wiring is really crappy, 7cm of wire, a microSD socket wirebonded on a prototyping pcb and resitors across there everywhere (didnt want to cut the legs unitl it works)
If you put your finger on the left side of the ESP32 board (IO13,GND,IO12,IO14...) sometimes it fails, so still not very stable.

LongmontRobotics, 2017-09-10:
I am new to the SDMMC interface.
I too got the earliest errors as here.
The schematic from thezenox above seems not to work for me.
Has any one else got this version working reliably ?

simmunity, 2017-11-12:
The solution provided by wiliameps2015 works for me with Analoglamb Maple 32 board using Sandisk 512mb microSD card (that is many years old) and Sandisk 16gb Ultra microSD card that is brand new (late 2017).
"
@LordJakson
I have Maple ESP32 R1 and I could use SD Card after initializing customized SPI Pins
void setup() {

  SPI.begin(14,2,15,13);//SCK,MISO,MOSI,ss
  delay(10);
  if (!SD.begin(13, SPI, 40000000, "/sd")) {
    Serial.println("Card Mount Failed");
  return;
  }

// the rest of the setup code goes here
"

PilnyTomas, 2023-05-17:
> @PilnyTomas This should be candidate for FAQ section but it's quite old and a bit HW specific.
> 
> What do you think @P-R-O-C-H-Y? Is this something what's still good to explain?

It is a 3rd party HW design issue, and the board is no longer available, so I don't think this should go into our documentation.

closed 2023-08-10 by PilnyTomas