espressif/arduino-esp32 #260: OTAbasic cpu halted

DiegoMendoza1, 2017-03-11:
hi, 

i was trying the ota example, and then ocurrs this:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2016
load:0x40078000,len:7780
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
Booting
E (98) wifi: wifi_init 1253 ret=4363
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40083723
Guru Meditation Error: Core  1 panic'ed (abort)

Backtrace: 0x40008155:0x3ffcda40 0x40007d16:0x3ffcda60 0x40093ab8:0x3ffcdaa0 0x400fa1d6:0x3ffcdaf0 0x400f376d:0x3ffcdb10 0x400d143c:0x3ffcdb40 0x400d144a:0x3ffcdb80 0x400d14b6:0x3ffcdbb0 0x400d152f:0x3ffcdbd0 0x400d0f51:0x3ffcdcd0 0x401107d6:0x3ffcdd00

CPU halted.
```
the code is the example of arduinoOTA:

```
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "...................";
const char* password = ".....................";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

```

erhardd, 2017-03-11:
[image]
That's my..

rodgergr, 2017-03-11:
@DiegoMendoza1 This looks very like the error mentioned in #207 and may be unrelated to your OTA Code. Have you tried erasing the flash?

me-no-dev, 2017-03-11:
@rodgergr was about to ask the same question.

me-no-dev, 2017-03-11:
it could be bad USB cable or not enough power too. Since it throws at the moment the WiFi is booting.

me-no-dev, 2017-03-11:
Anyway, closing this as it is surely not a code/library issue (you do not even get to that point)

closed 2017-03-11 by me-no-dev

PeWos, 2017-03-11:
I have erase flash and now it works fine

DiegoMendoza1, 2017-03-11:
I have erase the flash and now it works fine

mrohner, 2017-10-15:
@DiegoMendoza1
How did you erase the flash?

sansillusion, 2017-10-16:
I had to use IDF for that dont know if possible in Arduino but in IDF I used:

     python ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM6 --baud 115200 --before default_reset --after hard_reset erase_flash