espressif/arduino-esp32 #263: SimpleWiFiServer with high data loss rate

copercini, 2017-03-14:
Using SimpleWiFiServer with CSS and html some lines are just not printed by client. 

For test I adapted the SimpleWiFiServer example for print a list from 0 to 999, and it have a considerable amount of missing data, here is the test code:
``` c
            client.println("0 to 999 list<br>");
            for (int i = 0; i < 1000; ++i){
              client.print(i);
              client.println("<br>");         
            }
```

Some errors:
[image]

screenshot2:
[image]

copercini, 2017-03-15:
Increasing ` #define WIFI_CLIENT_SELECT_TIMEOUT_US (100000)` to 1000000 in libraries/WiFi/src/WiFiClient.cpp fix this issue. 

Maybe change this constant just for WiFi Server or allow user set it on arduino IDE code solve it.

me-no-dev, 2017-03-20:
Done and thanks :)

closed 2017-03-20 by me-no-dev