espressif/arduino-esp32 #265: WiFiClientSecure self signed cert support?

forthlightning, 2017-03-15:
WiFiClientSecure supported self-signed certs (perhaps accidentally) up until [this](https://github.com/espressif/arduino-esp32/commit/e30447449feac1588945dd544d39405a4ba0fa20) commit. Is there a way to enable self signed operation?

rodgergr, 2017-03-20:
@forthlightning @copercini 
 I may have been experiencing a similar problem when I could no longer connect to AzureIoTHub after commit #255 on the WiFiSecure library.

So, I did some digging and found some changes in ssl_client.cpp that seemed to be the cause of the issue.

In commit #255 the while() loop from line 241 onward was changed to return an error code, rather than printing out a handshake code and then breaking from the while loop.

I was not sure of intent of the change (i.e. whether it was an intended feature) , but tried changing this back to the original pre-#255  code as follows and everything began to work again:-)

`    DEBUG_PRINT( "Performing the SSL/TLS handshake...\n");

    while ((ret = mbedtls_ssl_handshake(&ssl_client->ssl_ctx)) != 0) {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE && ret != -76) {
            //return handle_error(ret);
			printf("mbedtls_ssl_handshake returned -0x%x\n", -ret);
			break;	 
					  
						 
        }
        delay(10);
        vPortYield();`

I realise that the printf statement may need to be upgraded to a DEBUG_PRINT, to match the new style, but did not want to play too much with the code.

As I'm only a newbie, I was not sure of the appropriate way to request a code change, but hope it is of some use.

me-no-dev, 2017-03-20:
Error should be exposed better. @copercini the client should act as any other client. All that is needed as error info for the handshake is to be exposed as some method that you can query on failed connect.
like: 
```cpp
if(!client.conect(...)) Serial.printf("SSL error was: %u\n", client.errorCode());
```

copercini, 2017-03-20:
@me-no-dev Yes, and change the internal debug print to be compatible with ESP debug levels.

@forthlightning @rodgergr
This commit just break the handshake if an error occur (some errors crash the entire board!)

About this issue:
When no CA is used, MbedTLS is returning:
> No CA Chain is set, but required to operate.
> MbedTLS message code: -30336

Which looks like a mbedtls bug: https://github.com/ARMmbed/mbedtls/issues/506

I am not sure what is the best workaround for WiFiClientSecure avoid this.

me-no-dev, 2017-03-21:
@copercini I would rather you use ESP32 Arduino's debug procedures (```log_v```/```log_d```/```log_i```/```log_w```/```log_e```)

forthlightning, 2017-03-23:
So is the above hack from @rodgergr the recommended fix so far? Currently I just froze my arduino component at e30447449feac1588945dd544d39405a4ba0fa20 pending real fix

copercini, 2017-04-03:
@forthlightning @rodgergr 
Please check if it is working with last core version...

rodgergr, 2017-04-05:
@copercini I downloaded the latest core version and sslclient now works fine with a self-signed certificate😀Thanks for all of your hard work👍

forthlightning, 2017-04-05:
same here, this is great!

me-no-dev, 2017-04-05:
lets close this then :)

closed 2017-04-05 by me-no-dev