espressif/arduino-esp32 #266: Analog interrupts?

Nikg92, 2017-03-16:
In the datasheet it says "The ULP-coprocessor in ESP32 is also designed to measure the voltages while operating in the sleep mode, to enable low power consumption; the CPU can be woken up by a threshold setting and/or via other triggers."
Are there any plans to implement this?

Nikg92, 2017-03-16:
First off, thank you for all your work and your quick response.
While manually managing memory does seem outside the scope of Arduino, wouldn't it be possible to implement an analog interrupt that is triggered above or below a set voltage? The memory management could be handled in the background and users could use an analogAttachInterrupt function that is similar to touchAttachInterrupt.

me-no-dev, 2017-03-17:
@Nikg92 I do not see an interrupt vector for the ADC to attach to ;) I've looked into this before.

closed 2017-03-20 by me-no-dev