espressif/arduino-esp32 #268: ICACHE_RAM_ATTR question

ccfoo242, 2017-03-16:
Hello,
I'm trying to get EspSoftwareSerial working (https://github.com/plerup/espsoftwareserial) and it doesn't look like ICACHE_RAM_ATTR is defined anywhere. 

I changed it from including "gpio.h" to <driver/gpio.h> after getting a file not found for gpio.h. Next the compiler is complaining about expected initializer for:

void ICACHE_RAM_ATTR sws_isr_0() { ObjList[0]->rxRead(); };

Any suggestions? Also, is this the proper place to ask this type of question, is there a forum that would be more appropriate?

Thanks,
Steve

me-no-dev, 2017-03-16:
That library would not work on ESP32. Why would you need it though? Arduino supports 3 serials on any pin you like.

closed 2017-03-16 by me-no-dev

ccfoo242, 2017-03-16:
Thanks. Shows how ignorant I am :-)