espressif/arduino-esp32 #269: ESP8266AVRISP library porting on ESP32

VashTheProgrammer, 2017-03-18:
Hi,
I'm trying porting of this library: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP
on arduino-esp32 repository but I don't have successful.

I have solved the compilation problems, but I can not load the firmware on my atmega microprocessor for some reason. Have any of you tried to import this library?

I published the library with my changes on this github repo so everyone can work on it.
link: https://github.com/VashTheProgrammer/ESP32AVRISP

Thanks so much.

me-no-dev, 2017-03-20:
I intend to write better programmer actually  (I already have some uploaders and bootloaders written for AVR)

VashTheProgrammer, 2017-03-21:
Perfect, I'm waiting you, if you want close the issue you can.
Thank you

If I can help you, tell me.

cybergogo, 2018-10-12:
Hi All, any news on this?

tab4moji, 2019-07-12:
I ported this library to micropython on ESPs.
https://github.com/tab4moji/ESPAVRISP

ferchinas, 2020-08-19:
Hi All, any news? I´m very interested in this feature.
Thanks

TimotF, 2021-01-02:
Looks like it has been done here => https://github.com/lbernstone/ESP_AVRISP

igrr, 2022-04-05:
Let's consider this feature implemented in https://github.com/lbernstone/ESP_AVRISP. We won't add AVRISP functionality to arduino-esp32 core at this time.

closed 2022-04-05 by igrr