espressif/arduino-esp32 #271: Make SPI variant info work

ladyada, 2017-03-18:
hiya - testing that TFT code on the Feather! :) right now the SPI pins defined in the variant files aren't ever used - this tweak will fix so the defaults are used (the Feather ESP32 uses other pins for routing ease)

https://github.com/adafruit/arduino-esp32/commit/d602169ebcd126c309009e87c1e455f2c892d08f

me-no-dev, 2017-03-18:
Ha! I kinda wonder why I have not caught this :) You want to make a PR or? do you have more changes in your fork or is it otherwise synced with us?

ladyada, 2017-03-18:
do you mind hand-adding? its only two lines...(my fork got desynced pretty badly and i didn't manage the conflict right so i'll start fresh. whee git!)

me-no-dev, 2017-03-19:
Will add manually :) No problem, but one good thing of github is  to track contributions :)

me-no-dev, 2017-03-19:
Done :)

closed 2017-03-19 by me-no-dev