espressif/arduino-esp32 #272: SD lib w/ card 'flakey'?

ladyada, 2017-03-18:
heya im using known good hardware, with pullups on the SD CS line, no level shifters, PCB traces not jumpers - but getting 'flakiness' on init/mount

```
[E][sd_diskio.cpp:99] sdSelectCard(): timeout
[E][sd_diskio.cpp:708] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed
```

oddly, it works maybe 10% of the time. once it starts working, it 'keeps' working until a full power reset. if it doesn't work, sometimes hitting software reset helps, but usually not. this is with the SPI library - tried changing SPI speed in SD.begin() without any noticable changes. still investigating ... but making a note here in case others can reproduce

me-no-dev, 2017-03-19:
Hm that is rather strange. I had gazillion of issues with SD_MMC and pullups but none with SD. I just connected the 4 data lines directly to SPI and that was it (I use SD->MicroSD with some wires soldered directly on the pads).
Which pins are you using?

ladyada, 2017-03-19:
static const uint8_t MOSI  = 18;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 5;

and cs on any pin really, but im using
  #define SD_CS      33

me-no-dev, 2017-03-19:
Done :)

closed 2017-03-19 by me-no-dev

ladyada, 2017-03-19:
looks good - would you like me to re-test?

me-no-dev, 2017-03-20:
Sure :) Confirmation is always good, but I was able to reproduce the issue and after the fix, was not able to anymore, regardless of what I did :) I wonder how it was working while I was developing the library?
BTW I tested with your pin configuration
```cpp
void mountSdCard(){
    SPI.begin(5,19,18);
    if(!SD.begin(33)){
        Serial.println("Card Mount Failed");
        return;
    }
}
```

ladyada, 2017-03-21:
hiya - ok so for one of my feather setups, with cardcs on #27, it now works every time. on another feather, with cs on #14 it still doesnt work

> Initializing SD card...[E][sd_diskio.cpp:99] sdSelectCard(): timeout
> [W][sd_diskio.cpp:452] ff_sd_initialize(): GO_IDLE_STATE failed
> [E][sd_diskio.cpp:724] sdcard_mount(): f_mount failed 0x(3)
> failed!

me-no-dev, 2017-03-21:
maybe on io14 there is another issue? I mean, if SPI is properly configured, and you use pins below 34, the lib itself does not care which pin you have selected. It calls digitalWrite to control SS

ladyada, 2017-03-21:
yeah unknown. ill investigate more later... :)

NickWaterton, 2017-06-03:
Interesting, I am having the same problem. Using the TFT feather wing (CS 14), the SD card works about 50% of the time, using the music maker featherwing, I can't get the same SD card to work at all (also CS 14).

SPI works fine with graphics and music etc, it's just the SD card that is unreliable. If I wasn't having a similar problem (but not as bad) with the TFT featherwing, I'd suspect a bad reader on the music maker.

When I have a chance I'll try it with an M0 or a WICED, see if they work with the music maker.

Frosticles, 2017-06-28:
I'm using the same pins as ladyada, coming from the adalogger featherwing, and I can't get the card to mount at all using: 

`#include <SD.h>
#include <SPI.h>

void setup() {
  Serial.begin(115200);
  SPI.begin(5, 19, 18);
  if (!SD.begin(33)) {
    Serial.println("Card Mount Failed");
  }
}`

SD card works in an M0 datalogger, but no luck with the ESP32 :/

ladyada, 2017-06-28:
@me-no-dev hmm do you mind if we send you some hardware that you can test with? the SD card works if we use our ported 'arduino-SD-library' but not with the SD library ya'll included. so I think its some init thing?

me-no-dev, 2017-06-28:
@ladyada yes please! let's get this fixed :) How are we going to arrange this?

ladyada, 2017-06-28:
sure! please email pt@adafruit.com and say "Hiya, I need to pick up some ESP32 Feathers and parts" and we'll reply to you by email. thx :)

me-no-dev, 2017-06-29:
done :)

ladyada, 2017-06-29:
k email addr deleted, we'll reply

dstoiko, 2017-07-04:
`[E][sd_diskio.cpp:724] sdcard_mount(): f_mount failed 0x(3)`
@me-no-dev thanks for this lib! Having the same issue here... Thought it was an issue with pull-up resistors at first but it doesn't seem to be. Any updates on this?

me-no-dev, 2017-07-04:
Things just showed up :) will solder the boards tonight and prep everything

ladyada, 2017-07-04:
yay - we have a full kit of cousre, and can try replicating any branches/fixes

dstoiko, 2017-07-05:
So do we know where this issue comes from? What can I do to further test?

me-no-dev, 2017-07-06:
OK I need more info on the failing cards because here is my first result at this (Feather selected as board) Maybe there are particular steps to follow?:
[image]
[image]
[image]

ladyada, 2017-07-07:
can you try a few times? also try plugging the esp32 into the tft feather and drawing a BMP from SD? that's the only thing i can say we tried and had difficulty with :)

NickWaterton, 2017-07-07:
I found with the tft feather, it works about 50% of the time.

I was using the music maker feather also, and it works 0% of the time with that (same pins).

I haven tried the data logger feather at all.

Nick Waterton P.Eng

Unsubscribe from GE's commercial electronic messages: http://sc.ge.com/*casl-unsubscribe 
Désabonner des messages électroniques commerciaux de GE: http://sc.ge.com/*lcap-desabonnement

bbx10, 2017-07-07:
@NickWaterton Are you referring to the SD card slot on the music maker wing? The VS1053 part works fine using the [ESP32-Radio](https://github.com/Edzelf/ESP32-Radio) project.

me-no-dev, 2017-07-07:
Hit a wall with the TFT Wing :) looking into it :)
@bbx10 yes, SD is the issue

me-no-dev, 2017-07-07:
@ladyada I seem to get intermittent issues wit the touch on the TFT as well. Like it will not start every once in a while. Have you notice this?

me-no-dev, 2017-07-07:
please try the latest commit :)

me-no-dev, 2017-07-07:
And here is a JPEG read from the SD card :)
[image]

bbx10, 2017-07-07:
@me-no-dev Works fine with SD card slot on Feather Adalogger (CS=33) and Music Maker (CS=14).

dstoiko, 2017-07-07:
@me-no-dev @ladyada Our config is custom, nothing to do with Feather. We have an ESP32-DevKitC with a WROOM-32, mounted on our PCB, and we've tried several different µSD/SD modules, all with the same intermittent mounting error:
- Adafruit µSD module (https://www.adafruit.com/product/254)
- "normal" SD module with adapter for µSD
- Bare µSD module with 10k resistors as shown per [this documentation](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card)
Again no luck with any of those options. We have only this on the SPI bus.
We have tried on our PCB and off of it on a breadboard with just the ESP32 and the SD module.

As a side note, when using an SPI-based LED matrix module (MAX7219) on the bus, trying to use the SD module crashes the ESP32. We had to remove the SD from our configuration which is pretty debilitating at the moment.

Any ideas where the solution might be?

@ladyada you said your port of the Adafruit SD library for Arduino boards worked 100%. I haven't seen that port anywhere, just the Arduino version. Do you have it on one of your repos?

me-no-dev, 2017-07-07:
@dstoiko did you try the latest commit? It's not a fix just for this board.

me-no-dev, 2017-07-07:
@dstoiko also I do not see you mention which pins you are using for SPI

NickWaterton, 2017-07-07:
Yes, the VS1053 works just fine. The SD card slot does not. I didn’t build the ESP32-Radio project, I’m streaming Bluetooth audio and playing it via the music maker. I noticed that the SD card slot doesn’t work though, so I tried it on a tft feather I have, and it works about 50% of the time.

I’m not using the sd card slot in my project, but I tried to get it to work (for testing) and couldn’t.

Nick Waterton P.Eng.
National Support Leader - Nuclear Medicine, PET and RP
GE Healthcare

M+1 416 859 8545
F +1 905 248 3089
E nick.waterton@med.ge.com<mailto:nick.waterton@med.ge.com>

2300 Meadowvale Boulevard
Mississauga, ON  L5N 5P9  Canada

GE imagination at work

dstoiko, 2017-07-07:
@me-no-dev will try asap. We are using these pins:
MOSI -> 23
MISO -> 19
SCK -> 18
CS -> 5

me-no-dev, 2017-07-07:
@dstoiko pins look good! You and @NickWaterton should have success if using the latest code.

ladyada, 2017-07-07:
hiya!
@dstoiko yah the default SD library worked for us, see it in https://github.com/adafruit/arduino-esp32/tree/d602169ebcd126c309009e87c1e455f2c892d08f
@me-no-dev not sure about the touch screen, it does share SPI pins and also the touch chip is annoying in that it changes modes based on SPI floating voltages on bootup. ill take a look again soon!

me-no-dev, 2017-07-07:
@ladyada it's possible that it was caused by the sd card stuff ;) have not seen if happen since SD is working.

ladyada, 2017-07-07:
awesome :) you also have some fun hardware to try other stuff!

me-no-dev, 2017-07-07:
@ladyada yeah! thanks for that! So far I have tried all but the servo shield and is working as expected :)

In our WROVER_KIT fork of the ILI driver, I have implemented direct drawing of bmp and jpeg files and buffers. Maybe you will like to add this for the ESP32 platform to your ILI driver as well?

ladyada, 2017-07-07:
oh yah that would be awesome. if you can submit a PR and just make sure it still compiles for arduino. jpg decoding would be great and would probably work on other M4/high speed platforms!

me-no-dev, 2017-07-07:
There are two ways to do this. ESP32 has the decoder in ROM, but the code footprint is also small.
So either compile it only for ESP32 and use the functions in ROM, or also include the source for other compatible platforms. And I agree ;) definitely not an AVR territory :)

Edzelf, 2017-07-08:
SD is working now correctly.

NickWaterton, 2017-07-11:
Just wanted to say the SD card is working correctly on the Musicmaker feather now also.

dstoiko, 2017-07-13:
Working more reliably now, mounting consistently all the time. I still get an error when running the SDCard test code : `[W][sd_diskio.cpp:137] sdCommand(): no token received`... Any ideas?

Also, in this example the functions that open files dont `close()` them. Is that normal behavior? I thought you needed to close the file after writing to it at least.

Full serial monitor output:
`SD Card Type: SDHC
SD Card Size: 15193MB
Listing directory: /
  FILE: /mode.txt  SIZE: 2
  FILE: /profile.txt  SIZE: 2
  FILE: /._profile.txt  SIZE: 4096
  FILE: /165166M.CSV  SIZE: 33
  FILE: /165166T.CSV  SIZE: 43
  DIR : /.Spotlight-V100
  DIR : /.fseventsd
  FILE: /165166C.CSV  SIZE: 16
  FILE: /165166A.CSV  SIZE: 13
  DIR : /.Trashes
  DIR : /.TemporaryItems
  DIR : /mydir
  FILE: /test.txt  SIZE: 0
  FILE: /foo.txt  SIZE: 13
Creating Dir: /mydir
Dir created
Listing directory: /
  FILE: /mode.txt  SIZE: 2
  FILE: /profile.txt  SIZE: 2
  FILE: /._profile.txt  SIZE: 4096
  FILE: /165166M.CSV  SIZE: 33
  FILE: /165166T.CSV  SIZE: 43
  DIR : /.Spotlight-V100
  DIR : /.fseventsd
  FILE: /165166C.CSV  SIZE: 16
  FILE: /165166A.CSV  SIZE: 13
  DIR : /.Trashes
  DIR : /.TemporaryItems
  DIR : /mydir
  FILE: /test.txt  SIZE: 0
  FILE: /foo.txt  SIZE: 13
Removing Dir: /mydir
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
rmdir failed
Listing directory: /
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
Writing file: /hello.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sd/hello.txt) failed
Write failed
Appending to file: /hello.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sd/hello.txt) failed
Append failed
Reading file: /hello.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:64] open(): /sd/hello.txt does not exist
Failed to open file for reading
Deleting file: /foo.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:135] remove(): /foo.txt does not exists or is directory
Delete failed
Renaming file /hello.txt to /foo.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:96] rename(): /hello.txt does not exists
Rename failed
Reading file: /foo.txt
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:64] open(): /sd/foo.txt does not exist
Failed to open file for reading
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:64] open(): /sd/test.txt does not exist
Failed to open file for reading
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sd/test.txt) failed
1048576 bytes written for 0 ms`

me-no-dev, 2017-07-15:
Yes you need to close the file if you open it

dstoiko, 2017-07-16:
@me-no-dev I was saying that because in the arduino-esp32 [SD test example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino), `file.close()` is not used in the file manipulation functions `readFile`, `writeFile` and `appendFile`.

me-no-dev, 2017-07-16:
@dstoiko must be error on my end :(

dstoiko, 2017-07-16:
@me-no-dev made a tiny PR for that

me-no-dev, 2017-07-17:
thank you @dstoiko merged :)

capedra, 2017-08-02:
Wow, five months and I can't mount my SD card on MH-ET Live ESP32 Minikit.

me-no-dev, 2017-08-02:
@capedra wow... that could be so many things... what is "MH-ET Live ESP32 Minikit" anyway? And how come I can mount all of my cards on all of my dev boards?

capedra, 2017-08-02:
@me-no-dev 
this board:
http://mh.nodebb.com/assets/uploads/files/1497417590890-1111.jpg?v=hfe2ofo3e00

apparently, it's not only me:
https://github.com/espressif/arduino-esp32/issues/495

me-no-dev, 2017-08-02:
@capedra well totally different issues ;) and I do not see a schematic where the card is connected or how are you trying to start it.

me-no-dev, 2017-08-02:
@capedra 
```cpp
#include <SD.h>
#include <SPI.h>

void setup() {
  Serial.begin(115200);
  SPI.begin(18, 19, 23);
  if (!SD.begin(5)) {
    Serial.println("Card Mount Failed");
  }
}
```

capedra, 2017-08-02:
@me-no-dev 
Hello Hristo, I changed the SD.cpp file and I found out that the following if condition is returning false:

if(!sdcard_mount(_pdrv, mountpoint))
>>>>>>>>>>>>>>>>>> !sdcard_mount(_pdrv, mountpoint)
Card Mount Failed

capedra, 2017-08-02:
@me-no-dev Apparently, this issue does not happen only with me: https://www.esp32.com/viewtopic.php?t=2592

me-no-dev, 2017-08-02:
@capedra how is the card formatted? how big is it? how do you init/mount it? show some code please!

capedra, 2017-08-02:
@me-no-dev it's formatted as FAT32. It's a 16 GB card (SDHC).

I did it the exact way that you told me, but I tried another ways too:
```
#include <SD.h>
#include <SPI.h>
void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  SPI.begin(18, 19, 23);
  if (!SD.begin(5)) {
    Serial.println("Card Mount Failed");
  }
}

```

this same error persists using the codes below:
```
void setup() {
  Serial.begin(115200);
  SPI.begin(SCK, MISO, MOSI, SS); //SCK,MISO,MOSI,ss
  delay(10);
  if (!SD.begin(SS)) {
    Serial.println("Card Mount Failed");
    return;
  }
}
```
Even with this library below, I can't mount:
```
#include <SPI.h>
#include <mySD.h>

// set up variables using the SD utility library functions:
Sd2Card card;
SdVolume volume;
SdFile root;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
const int chipSelect = 5;

void setup()
{

  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  Serial.print("\nInitializing SD card...");
  // On the Ethernet Shield, CS is pin 4. It's set as an output by default.
  // Note that even if it's not used as the CS pin, the hardware SS pin
  // (10 on most Arduino boards, 53 on the Mega) must be left as an output
  // or the SD library functions will not work.
  pinMode(SS, OUTPUT);
  //pinMode(5, OUTPUT);

  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  // uint8_t init(uint8_t sckRateID, uint8_t chipSelectPin, int8_t mosiPin = -1, int8_t misoPin = -1, int8_t clockPin = -1);
  while (!card.init(SPI_HALF_SPEED, 5, 23, 19, 18)) { // (5, 23, 19, 18) <=> CS, MOSI, MISO, SCK
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card is inserted?");
    Serial.println("* Is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
  }

```

capedra, 2017-08-02:
I have to go now but I'll come back later. Thanks!

capedra, 2017-08-03:
@me-no-dev I formatted it using SD Memory Card Formatter 5.0 for SD/SDHC/SDXC.
It's a 16 GB SDHC card which works pretty well with ESP-12 and SdFat library.

capedra, 2017-08-03:
@me-no-dev @MHEtLive @ladyada it was working with your libraries (the espressif's library and ladyada's library: https://github.com/adafruit/arduino-esp32/tree/d602169ebcd126c309009e87c1e455f2c892d08f) after I formatted the card using "[SD Memory Card Formatter 5.0 for SD/SDHC/SDXC](https://www.sdcard.org/downloads/formatter_4/index.html)". It does never work on pin 5 but it was working on pin 4, but as soon as I pressed the reset button, the card doesn't mount anymore, not even powering the ESP32 off/on. Did anyone solve that problem? I can't find the "f_mount" implementation, just the header file where this method/function is interfaced. Please help!

[42 more replies]