espressif/arduino-esp32 #273: Crash after running a while

torntrousers, 2017-03-19:
I've a sketch running on an ESP32 and it works for a while but always eventually crashes with the following:

E (8906392) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x400855b8  PS      : 0x00060234  A0      : 0x80084782  A1      : 0x3ffd41c0  
A2      : 0x3ffd3d2c  A3      : 0x00060223  A4      : 0x00060220  A5      : 0x3ffeebe0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x40083108  A9      : 0xb33fffff  
A10     : 0x00060223  A11     : 0x00000000  A12     : 0x3f417a20  A13     : 0x0000001f  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400855b8:0x3ffd41c0 0x40084782:0x3ffd41e0 0x40083e3b:0x3ffd4200 0x400856a9:0x3ffd4240 0x400856c2:0x3ffd4280

CPU halted.

From that is there some way I can debug what the problem might be?

me-no-dev, 2017-03-20:
See [this tool](https://github.com/me-no-dev/EspExceptionDecoder) to help you decode the issue.
Some explanation on what you are running might help us also ;)

torntrousers, 2017-03-20:
Thanks. The sketch is scanning for BLE advertisements and for each sending them off over Wifi with an MQTT publish. 

Using the tool for the above exception shows:

0x400855b8: xTimerGenericCommand at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40083108: update_written_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 291
:  (inlined by) spi_flash_mark_modified_region at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 266
0x400855b8: xTimerGenericCommand at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40084782: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3045
0x40083e3b: xQueueGenericSendFromISR at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400856a9: prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400856c2: prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

And for another one of the crashes:
.E (1905257) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40085724  PS      : 0x00060034  A0      : 0x800848ee  A1      : 0x3ffd41c0 
A2      : 0x3ffd3d34  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x3ffeebe0 
A6      : 0x00000003  A7      : 0x00060e23  A8      : 0x00000066  A9      : 0x00000066 
A10     : 0x00060023  A11     : 0x00000000  A12     : 0x3f417a34  A13     : 0x0000001f 
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005 
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb 

Backtrace: 0x40085724:0x3ffd41c0 0x400848ee:0x3ffd41e0 0x40083fa7:0x3ffd4200 0x40085815:0x3ffd4240 0x4008582e:0x3ffd4280

CPU halted.

shows:
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400848ee: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
0x40083fa7: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x40085815: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x4008582e: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

Doesn't mean much to me, any ideas?

me-no-dev, 2017-03-20:
Yes, BLE and WiFi do not play nice together yet ;) You can post the issue over at espressif/esp-idf, so the proper people get to see it :)

closed 2017-03-20 by me-no-dev