espressif/arduino-esp32 #274: support Scheduler library

Jorgen-VikingGod, 2017-03-20:
Is there a way to use the [Scheduler](https://www.arduino.cc/en/Reference/Scheduler) library?
Would be great to have multiple tasks/loops running.
(it would be also great to have priorities too)

**Example use:**
```
#include <Scheduler.h>

int counter = 0;
int counter1 = 0;

void setup()
{
  Scheduler.startLoop(loop1);
}

void loop () {
 analogWrite(9, counter);
 counter++;
 if (counter > 255){
  counter = 0;
 }
 delay(33);
}

void loop1 () {
 analogWrite(10, counter1);
 counter1=counter1+5;
 if (counter1 > 255){
  counter1 = 0;
 }
 delay(10);
 yield();
}
```

forthlightning, 2017-03-20:
Do tasks do what you need? You can access them from the arduino environment.

http://www.freertos.org/a00125.html

Jorgen-VikingGod, 2017-03-20:
wow great - did not know, that it is accessible within arduino environment.
So Maybe I build my own helper class and using **xTaskCreate** and **vTaskDelay**

hreintke, 2017-03-21:
@Jorgen-VikingGod : 
Did not look into it yet so this is not a comment on usability and/or quality.
https://github.com/michaelbecker/freertos-addons

copercini, 2017-06-22:
I made a little example of multiples loops , check this out: https://github.com/copercini/esp32-iot-examples/blob/master/multiloop/multiloop.ino

copercini, 2017-08-27:
Since the arduino core run over FreeRTOS and you can call the APIs directly in the Arduino sketch, so it's not necessary an additional library

closed 2017-08-27 by copercini

eadmaster, 2024-10-06:
the Scheduler.h API is simpler to use and does not require to specify a stacksize, maybe this can be reconsidered?