espressif/arduino-esp32 #275: smartconfig bug

chenkaiyao, 2017-03-21:
I use the smartconfig example, but had a problem. 
when I config a device to connect to Router1， it  return:
``
Waiting for SmartConfig.
....................
SmartConfig received.
Waiting for WiFi
.....WiFi Connected.
IP Address: 192.168.3.207
ets Jun  8 2016 00:22:57 
``

I change the Router2, it return:

``
Waiting for SmartConfig.
............................................................................................................................
SmartConfig received.
Waiting for WiFi
.....WiFi Connected.
IP Address: 192.168.3.207
``

why IP Address is the same?

chenkaiyao, 2017-03-21:
the Router2 IP Address should be 192.168.1.***

me-no-dev, 2017-03-21:
hmm.. can you confirm that it connects actually to the other access point? If you use ArduinoIDE, please enable debug level "Verbose" or "Debug" so we see what events come from WiFi. When printing the IP, print SSID as well.

chenkaiyao, 2017-03-22:
**ok! it connects to one access point, it printing:**

[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 12 - AP_START
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 13 - AP_STOP
Waiting for SmartConfig.
..............................................................................[D][WiFiSTA.cpp:564] _smartConfigCallback(): ssid     = MOLMC_HUAWEI
[D][WiFiSTA.cpp:565] _smartConfigCallback(): password = 12345678
.
SmartConfig received.
Waiting for WiFi
....[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
.WiFi Connected.
IP Address: 192.168.3.239
SSID: MOLMC_HUAWEI

**connets the other access point, it pringting:** 

[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 12 - AP_START
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 13 - AP_STOP
Waiting for SmartConfig.
................

iCh lock
.

.

...[D][WiFiSTA.cpp:564] _smartConfigCallback(): ssid     = MOLMC-NETGRAR
[D][WiFiSTA.cpp:565] _smartConfigCallback(): password = 12345678
.
SmartConfig received.
Waiting for WiFi
....[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): Event: 7 - STA_GOT_IP
.WiFi Connected.
IP Address: 192.168.3.239
SSID: MOLMC_HUAWEI

**I am printing the Received ssid and password, but connecting seems no work!**

copercini, 2017-03-22:
Both are on the same network? and have DHCP enabled?

chenkaiyao, 2017-03-24:
@me-no-dev smartconfig example is ok for you ?

me-no-dev, 2017-03-24:
Sorry guys, just forgot to reply, but I will advise you to file the issue at espressif/esp-idf, because the problem is somewhere in the system wifi libs. It looks like once you connect, the configuration is not overwritten and it connects to the last station configured.
Now we can hack this to clear the configuration when smart config is enabled, but that is a hack and not the proper way to do things :)

me-no-dev, 2017-03-24:
OK, let me try a few things and see if I can fix this without hackery.

me-no-dev, 2017-03-24:
Oh this is embarrassing :D check which interface is the new setting set on... https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L564

closed 2017-03-24 by me-no-dev