espressif/arduino-esp32 #276: ESP32 + SX1272 Lora

PhilColbert, 2017-03-21:
Hi,

Just trying to port esp8266 code to esp32....all going well until an issue within an irq receiver.

Within this function which is run many times the esp calls an abort ( even if the variable address is trying to be printed out to serial - it aborts ), the only differences is that the code is called from in an IRQ........

https://github.com/3s1d/fanet/blob/master/fanet_stack/sx1272.cpp - line 100

void SX1272::writeRegister(byte address, byte data)
{
	select();
	/* bit 7 set to read from registers */
	SPI.transfer(address | 0x80);    /
	SPI.transfer(data);
	unselect();

#if (SX1272_debug_mode > 1)
	Serial.print(F("## SX1272 write reg("));
	Serial.print(address, HEX);
	Serial.print(F(")="));
	Serial.print(data, HEX);
	Serial.println();
#endif
}

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1445 (xQueueGenericReceive)- assert failed! 
abort() was called at PC 0x4008375d 
Guru Meditation Error: Core  1 panic'ed (abort) 
 
Backtrace: 0x40009203:0x3ffc1050 0x400d5596:0x3ffc1070 0x400d5d5c:0x3ffc10b0 0x400d4c5c:0x3ffc10d0 0x400d4d75:0x3ffc10f0 0x400d4db0:0x3ffc1110 0x400d4e92:0x3ffc1130 0x400d41fb:0x3ffc1150 0x400d43e6:0x3ffc1170 0x400d4aae:0x3ffc11a0 0x400d4b2d:0x3ffc11c0 0x400809e0:0x3ffc11e0 0x40081861:0x3ffc1200 
 
CPU halted.

This is well beyond me :) Any ideas ? 

Thanks

me-no-dev, 2017-03-21:
I have some, yes. Peripherals are not optimized for calls from IRQs directly. I'm going to do what I can to change that, but I can not promise full success. ESP32 is quite different than other MCUs and requires different approach towards tasks :) Best case to get it running is the author to optimize for ESP32 and not use IRQs, but rather use Queue and a thread to manage the events on ESP32

PhilColbert, 2017-03-21:
Thanks for the information, will look into what you mentioned - looks like quite a bit rewrite ... ahh well, nothings ever easy :)

Do IRQ's always fail then ?

Thanks

me-no-dev, 2017-03-21:
It's not that they fail, it's because we have two cores and need to protect the peripherals from accessing them at the same time, or while in interrupt. Imagine you have a code that writes periodically to I2C OLED some info and that you have IRQ setup to read some I2C sensor (that has IRQ). If the ISR is called while you are talking to the OLED, things will go bad :)

PhilColbert, 2017-03-21:
Understood - just had a quick play, just doing an interrupt just to set a variable works fine, its further into the code that it decides to stop abort, so I can hopefully sort in the main thread - we shall see !

Thank you for the info - do you have a contribution paypal address or something to send some funds for this project , or is this a paid job now ;)

Thanks

me-no-dev, 2017-03-21:
I do work for Espressif now, so no contribution is necessary :)

closed 2017-03-21 by me-no-dev

sabas1080, 2017-03-21:
Maybe this can help

https://github.com/matthijskooijman/arduino-lmic/pull/77

PhilColbert, 2017-03-22:
Thanks, will look into it