espressif/arduino-esp32 #277: I installed $ pip install pyserial

Hungluiz, 2017-03-21:


Hungluiz, 2017-03-21:
First of all, I can speak english ja little. I installed command 

> $ pip install pyserial

. and that's show message success " Installing collected packages: `pyserial Successfully installed pyserial `".  
But after i run command " 

> `~/esptool$ python esptool.py -p /dev/ttyUSB0 `--baud` 460800 write_flash --flash_size=detect 0 /home/nmhung/Downloads/0x00000_blank1m.bin `

" .
 and then show error " File "esptool.py", line 2294, in <module>
    main()
  File "esptool.py", line 2051, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before)
  File "esptool.py", line 186, in detect_chip
    detect_port = ESPLoader(port, baud) "
How to to fix it, can help me....

copercini, 2017-03-22:
Are you using ESP32 or ESP8266?

5chufti, 2017-03-28:
no, he most likely is using XP and recent serial.py is not compatible.
Get new esptool.py from here
http://www.esp8266.com/viewtopic.php?f=26&t=14222&p=64201#p64201

closed 2017-05-19 by me-no-dev