espressif/arduino-esp32 #278: WifiClientSecure has no verify method

salqadri, 2017-03-23:
I'm porting some esp8266 code and discovered that the ESP32 version of the WiFiClientSecure class does not seem to have any methods to help verify any certs or fingerprints. Is this something that is on the roadmap?

copercini, 2017-03-23:
SHA1 fingerprint is broken now: https://shattered.io/ 

You can use the entire CA certificate to verify. How to do that?

## Open your target site (I will use https://www.howsmyssl.com/ for example...)
## See the certificates
[image]

## Download the CA certificate
[image]

## Open in your favorite text editor, format and put in your arduino sketch
[image]

terryjmyers, 2017-04-18:
Should we incorporate a SHA-3 512 library as a standard: https://github.com/DuSTman31/SHA-3

zenmanenergy, 2017-10-20:
It appears to be in the esp8266 version. https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L517

Is there a reason the same code cannot be used?

copercini, 2017-10-20:
@zenmanenergy All the SSL core was changed from ESP8266 (AxTLS) to ESP32 (MbedTLS), there is a lot of new features, and SHA1 fingerprinting was deprecated due security reasons (explanation [here](https://github.com/espressif/arduino-esp32/issues/657#issuecomment-331709453) ) for now there is no fingerprinting support, but something with SHA256 or similar can be done in the future... 
For now it supports certificates verification (like browsers do)

vupqtran, 2018-02-21:
Hi @copercini 
Can i use 1 CA Certificate for more esp32? Or can i get CA certificate with esp not using web browser?
Thanks a lot.

copercini, 2018-02-21:
@vupqtran yes, since CA is public, you can use the same in a lot of browsers and ESPs (not confuse CA with client certificates, which are unique for each device)

The idea of having a CA is to verify the server identity, if you are talking with your real server and not a fake one. So get it "on the fly" is not the best idea since you can download a fake certificate that verifies a fake server...

closed 2018-05-14 by me-no-dev

teenkevo, 2022-05-17:
> @zenmanenergy All the SSL core was changed from ESP8266 (AxTLS) to ESP32 (MbedTLS), there is a lot of new features, and SHA1 fingerprinting was deprecated due security reasons (explanation [...]

@[kedzjyw](https://github.com/kedzjyw)

PR #1397 (2018-05-14): Port SSL fingerprint checking from ESP8266 WiFiClientSecure to ESP32
This PR ports `verify` function from ESP8266 WiFiClientSecure which verifies remote certificate fingerprint.

Since sha1 is no longer secure, I switched to sha256.

Fixes #278