espressif/arduino-esp32 #281: WifiBluetoothSwitch.ino  bluetooth not ok ?

JPB1JPB1, 2017-03-23:
Hello,

thanks for this good example and i success to recompile all of it under Arduino 1.6.13, and using AVRISP mkII programmer mode then upload speed at 115200 for the USB-UART on SIlabs chipset in ESP-32-Wroom-devkitC ..
Just to download the code, i need to press the BOOT button to enter in flashing mode... and wait a little ...

but reset with the EN button, and the serial monitor at 115200 is running nicely...

TEST :
Wifi AP is ok.
WIFI sta may be, i have not yet tried..
Bluetooth and BLE : 
But i can not see the device in any bluetooth under PCs or Smartphone...  
May be i miss something ... Any idea why and how...

Great job to get both mode at the same time... Keep going... 
JP

JPB1JPB1, 2017-03-23:
To flash or download the code in it , press and hold the BOOT button  during press one time the EN button, then on the serial port in the monitor under Arduino, it is displayed waiting for download
Just to inform everybody...
Best Regards
JP

copercini, 2017-03-23:
In this case, btStart() just "get Bluetooth ready for use" what will not necessarily make it visible to other devices.

You can hack it using https://github.com/espressif/arduino-esp32/blob/master/libraries/SimpleBLE/examples/SimpleBleDevice/SimpleBleDevice.ino too, and making visible to your computer/smartphone and other cool things  :smile:

JPB1JPB1, 2017-03-23:
Thanks for your reply, and suggestion, Copercini.

Yes, i have seen this example too. I'm trying it, and the serial link looks ok..

But with Pc and Bluetooth, it is not seen it under Pcs.... I'm testing ... also..

looking for more Bluetooth example under this directory https://github.com/espressif/arduino-esp32/tree/master/libraries ......

Best regards
Jp

copercini, 2017-03-23:
Arduino IDE still has few examples for BLE. 

First you have to check if your PC support Bluetooth low energy (BLE), because it's a bit different of classic Bluetooth.

JPB1JPB1, 2017-03-23:
thanks for your reply, but the ESP32 is not yet supporting the dual mode bluetooth ( Bluetooth 2.0 (3.0 ?)  and only Bluetooth smart-low-energy BTLE or BLE now ) ?

Question :  any timeframe when the "old" bluetooth mode may be in the good on going code ?

Thanks Copercini for your answer.
Jp
The simpleble example compile and download ok with ARDUINO IDE .... Thanks..

me-no-dev, 2017-03-24:
No timeframe for Classic BlueTooth but it will come. For you to use the current code, you need BLE

closed 2017-03-24 by me-no-dev

Serpent999, 2017-03-24:
The SimpleBLE sketch won't let other Bluetooth devices discover the ESP32. It could be a bug.

nicoverduin, 2017-05-07:
I tried the simpleBle example on my ESP32 dev board. Compiles fine, flashes fine but not visible on Iphone 6 and Samsung Galaxy S6. Do I need to update some core firmware?

Serpent999, 2017-05-10:
nicoverduin try using nRF toolbox app to discover the ble ESP32. In the app, use UART mode .

me-no-dev, 2017-05-12:
BLE does not show on the device the same way that classic BT devices do. You need to use some sort of BLE explorer

nicoverduin, 2017-05-12:
Ok, I will try that. I used BLE scanner on my Iphone 6+ but there it did not show up. Then I installed the ESP-IDF environment and took one of the examples. There it did show up on the BLE scanner, but not on the bluetoot scanner through settings. So ther might still be some discrepancy

me-no-dev, 2017-05-12:
Aaahhh sorry. Ble switch just shows how to switch on/off both stacks. It does not actually setup ble to advertise

electronicsguy, 2017-06-27:
Just soldered my ESP32 and was trying out the SimpleBLEDevice sketch. Was surprised to see nothing on my iphone :D. Didn't know that BLE don't show up directly. Finally got around to seeing the device. For those using i-devices, this app is great at discovery and info from BLE devices: "Bluetooth Smart Scanner" by Ace Sensor. (not advertising for them, just tried out a few and saw that this is the best)