espressif/arduino-esp32 #283: I2C do not read correct values

Jorgen-VikingGod, 2017-03-23:
I currently working on a project with ESP32 devboard as I2C master and some other board (XMEGA) as I2C slave.

Working with Atom and platformio (arduino-esp32 master) and also same with ArduinoIDE 1.8.0 with arduino-esp32 master.

As I can so with LogicAnalyzer the data is correctly written and read on the I2C bus - but my ESP32 gets wrong values back from the Wire.read() method.

first I speak with device 0x10 and read 15 Bytes (device Info) of data from register offset 0x30
Than I set new address 0x20 on offset 0x40.
After that read 12 Bytes (=device ID) from offset 0x33 and compare it with first fetched data...

On LogicAnalyzer the data from first read and after adress change the second read is the same (see dump below)

Following dump I got from the LogicAnalyzer:
```
Time [s], Analyzer Name, Decoded Protocol Result
2.199277083333333,I2C,Setup Write to [0x00] + NAK
3.236384750000000,I2C,Setup Write to [0x10] + ACK
3.236475750000000,I2C,0x30 + ACK
3.236600583333333,I2C,Setup Read to [0x11] + ACK
3.236691583333333,I2C,0xAF + ACK
3.236782666666667,I2C,0x01 + ACK
3.236873666666667,I2C,0x01 + ACK
3.236964666666667,I2C,0x00 + ACK
3.237055666666667,I2C,0x0D + ACK
3.237146666666667,I2C,0x0D + ACK
3.237237750000000,I2C,0x0D + ACK
3.237328750000000,I2C,0x0D + ACK
3.237419750000000,I2C,0x0D + ACK
3.237510750000000,I2C,0x0D + ACK
3.237601750000000,I2C,0x13 + ACK
3.237692833333333,I2C,0x0D + ACK
3.237783833333333,I2C,0x0D + ACK
3.237874833333333,I2C,0x13 + ACK
3.237965833333333,I2C,0x0D + NAK
3.291141250000000,I2C,Setup Write to [0x10] + ACK
3.291232250000000,I2C,0x40 + ACK
3.291323250000000,I2C,0x20 + ACK
3.291664666666667,I2C,Setup Write to [0x20] + ACK
3.291755666666667,I2C,0x33 + ACK
3.291868500000000,I2C,Setup Read to [0x21] + ACK
3.291959500000000,I2C,0x00 + ACK
3.292050583333333,I2C,0x0D + ACK
3.292141583333333,I2C,0x0D + ACK
3.292232583333333,I2C,0x0D + ACK
3.292323583333333,I2C,0x0D + ACK
3.292414583333333,I2C,0x0D + ACK
3.292505666666667,I2C,0x0D + ACK
3.292596666666667,I2C,0x13 + ACK
3.292687666666667,I2C,0x0D + ACK
3.292778666666667,I2C,0x0D + ACK
3.292869666666667,I2C,0x13 + ACK
3.292960750000000,I2C,0x0D + NAK
```

But on ESP32 the data read back with Wire.read() is different.

Here are all DebugPrints:
```
Main: Core(123)
Main: Hostname: Core-123
Main: timestamp: Thu Mar 23 20:17:45 2017
Main: Join I2C bus as master
Main: Initial enable and interrupt pins
Main: loop > state MLS_INITIAL_DEVICES:
DeviceList: setSuccessInitialCallback:
DeviceList: initialDevices:
DeviceList: initialDevices > loop and initial all divices:
I2C_Device: I2C_Device (10):
I2C_Device: fetchDeviceInfo:
I2C_Device: 15 = readDataByAddressAndOffset(10, 30, AF 0D E0 FC C8 FC 10 00 9C FC 0F 00 01 00 0C , 15)
I2C_Device: fetchDeviceInfo > manufactorID: AF
I2C_Device: fetchDeviceInfo > softwareID: 01
I2C_Device: fetchDeviceInfo > deviceType: 01
I2C_Device: fetchDeviceInfo > deviceID: 00 0D 0D 0D 0D 0D 0D 13 0D 75 FC 3F
I2C_Device: changeDeviceAddress (20):
I2C_Device: 1 = writeDataToAddressAndOffset(10, 40, 20 , 1)
I2C_Device: 12 = readDataByAddressAndOffset(20, 33, 00 FC 2F 00 23 00 00 00 20 00 20 FC, 12)
Helper: false = compareBuffer(00 0D 0D 0D 0D 0D 0D 13 0D 75 FC 3F, 00 0D 0D 0D 0D 0D 0D 13 0D 0D 13 0D, 12)
DeviceList: initialDevices > device address change failed! abort loop
DeviceList: initialDevices > return DL_INIT_ERROR
```

This is my method to read in data by offset:
```
/**
   * request data from slave by address at register address and read back the data.
   * deviceAddress: slave address
   * addr:          slave register address
   * buf:           return data buffer
   * length:        data length to read back
   */
  static uint8_t readDataByAddressAndOffset(uint8_t deviceAddress, uint8_t addr, uint8_t* buf, uint8_t length) {
    Wire.beginTransmission(deviceAddress);
    Wire.write(addr);
    Wire.endTransmission(false);  // Don't issue a stop because we need to read the value.
    uint8_t n = Wire.requestFrom(deviceAddress, static_cast<size_t>(length), true);
    DEBUG_PRINT("I2C_Device: ");
    DEBUG_PRINT(n);
    DEBUG_PRINT(" = readDataByAddressAndOffset(");
    DEBUG_PRINT_HEX(deviceAddress);
    DEBUG_PRINT(", ");
    DEBUG_PRINT_HEX(addr);
    DEBUG_PRINT(", ");
    for (uint8_t i = 0; i<n; i++) {
      if (Wire.available()) {
        *buf = Wire.read();
        DEBUG_PRINT_HEX(buf[i]);
        DEBUG_PRINT(" ");
        buf++;
      }
    }
    DEBUG_PRINT(", ");
    DEBUG_PRINT(length);
    DEBUG_PRINTLN(")");
    return n;
  } // readDataByAddressAndOffset
```

I wonder how to debug the values coming in from:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L330

me-no-dev, 2017-03-24:
for debugging you can use the logging facility of arduino (esp32-hal-log.h) and adjust the output level with an option in the ArduinoIDE or IDF menuconfig (if board selected is the dev module)

From the dum above it seems like the wrong bytes come after 12? the last 3 are the ones that look weird right?

Jorgen-VikingGod, 2017-03-24:
@me-no-dev thank you for the hint with esp32-hal-log.h
I just try it out and add some log_d for i2c->dev->fifo_data.val & 0xFF;
And the data logged is the same the data on the LogicAnalyzer.
But as I read it later in with the Wire.read() only the first 11 Bytes are correct.
I also changed my helper method to read in data from
```
for (uint8_t i = 0; i<n; i++) {
  if (Wire.available()) {
    *buf = Wire.read();
    DEBUG_PRINT_HEX(buf[i]);
    DEBUG_PRINT(" ");
    buf++;
  }
}
```

to
```
for (uint8_t i = 0; i<n; i++) {
  if (Wire.available()) {
    buf[i] = Wire.read();
    DEBUG_PRINT_HEX(buf[i]);
    DEBUG_PRINT(" ");
  }
}
```

Same result and I'm stuck now :/ 
Any other hints for me?

Jorgen-VikingGod, 2017-03-24:
Sorry for generating extra work. I found the issue now on my helper method to copy buffer to local buffer. There was a small bug by using old pointer offset.

>> this issue can now be closed. Sorry again guys
And many thanks for the help

closed 2017-03-24 by Jorgen-VikingGod