espressif/arduino-esp32 #286: I2C returns garbage when you have multiple devices and start hardware interrupts

marcmerlin, 2017-03-25:
I have fully working reliable code that talks to I2c IO expander + accelerometer + temp/humidity sensor.
Then, I added a hardware timer to read from an IO pin for the IR receiver, and I2C started returning random garbage.

@MartyMacGyver nicely pointed me out to https://www.esp32.com/viewtopic.php?f=14&t=1350&p=6988#p6988 and in turn I patched my hal like so:

`saruman [mc]$ diff -u esp32-hal-i2c.c.orig esp32-hal-i2c.c
--- esp32-hal-i2c.c.orig	2017-03-25 14:52:38.774550583 -0700
+++ esp32-hal-i2c.c	2017-03-25 14:55:06.907534396 -0700
@@ -22,6 +22,12 @@
 #include "soc/i2c_struct.h"
 #include "soc/dport_reg.h"
 
+// #define DR_REG_I2C_EXT_BASE                     0x3ff53000
+#define DR_REG_I2C_EXT_BASE_FIXED               0x60013000
+// #define DR_REG_I2C1_EXT_BASE                    0x3ff67000
+#define DR_REG_I2C1_EXT_BASE_FIXED              0x60027000
+
+
 //#define I2C_DEV(i)   (volatile i2c_dev_t *)((i)?DR_REG_I2C1_EXT_BASE:DR_REG_I2C_EXT_BASE)
 //#define I2C_DEV(i)   ((i2c_dev_t *)(REG_I2C_BASE(i)))
 #define I2C_SCL_IDX(p)  ((p==0)?I2CEXT0_SCL_OUT_IDX:((p==1)?I2CEXT1_SCL_OUT_IDX:0))
@@ -49,16 +55,16 @@
 #define I2C_MUTEX_UNLOCK()
 
 static i2c_t _i2c_bus_array[2] = {
-    {(volatile i2c_dev_t *)(DR_REG_I2C_EXT_BASE), 0},
-    {(volatile i2c_dev_t *)(DR_REG_I2C1_EXT_BASE), 1}
+    {(volatile i2c_dev_t *)(DR_REG_I2C_EXT_BASE_FIXED), 0},
+    {(volatile i2c_dev_t *)(DR_REG_I2C1_EXT_BASE_FIXED), 1}
 };
 #else
 #define I2C_MUTEX_LOCK()    do {} while (xSemaphoreTake(i2c->lock, portMAX_DELAY) != pdPASS)
 #define I2C_MUTEX_UNLOCK()  xSemaphoreGive(i2c->lock)
 
 static i2c_t _i2c_bus_array[2] = {
-    {(volatile i2c_dev_t *)(DR_REG_I2C_EXT_BASE), NULL, 0},
-    {(volatile i2c_dev_t *)(DR_REG_I2C1_EXT_BASE), NULL, 1}
+    {(volatile i2c_dev_t *)(DR_REG_I2C_EXT_BASE_FIXED), NULL, 0},
+    {(volatile i2c_dev_t *)(DR_REG_I2C1_EXT_BASE_FIXED), NULL, 1}
 };
 #endif
`
After this patch, my problem went away, and I2C is reliable with the hardware interrupt running.

me-no-dev, 2017-03-25:
cool. will do!

me-no-dev, 2017-03-25:
Done :)

closed 2017-03-25 by me-no-dev

MartyMacGyver, 2017-03-25:
I'm thinking this should be fixed in the ESP IDF as well, if it hasn't been already.