espressif/arduino-esp32 #290: Enabling RMT/semaphores in complex code can cause analogread to return values 2x too big

marcmerlin, 2017-03-28:
As first reported here MartyMacGyver/ESP32-Digital-RGB-LED-Drivers#7
using the RMT neopixel driver causes analogread to return 4096 for 50%-100% and proper values from 0 to 50%.
Sadly if you are sampling data that needs to go in the 50 to 100% range (joystick), it's impossible to get usable data.
If i comment out just these 3 lines of code, then analogread works properly again:
//RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
//RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
//xSemaphoreTake(ws2812_sem, portMAX_DELAY);

projectgus, 2017-03-28:
This feels like it could be power glitch problems. Do you still see the bug if you reconfigure RMT to use a different pin which isn't connected to RGB LEDs?

marcmerlin, 2017-03-28:
example code showing the failure: https://github.com/marcmerlin/IoTuz/commit/305b9f7673ae88621e29f911e4ef6134ae9fc230
@projectgus sure, I just tried this:
--- a/IoTuz.cpp
+++ b/IoTuz.cpp
@@ -378,7 +378,7 @@ void IoTuz::begin()
     //pixels.setPixelColor(0, 10, 10, 10);
     //pixels.show();
 #else
-    ws2812_init(RGB_LED_PIN, LED_WS2812B);
+    ws2812_init(0, LED_WS2812B);

I still get:
Analogread test before LEDs enabled, returns 1780 or so): 1767
Turning on LEDs
Before colors
After colors
Analogread test after LEDs enabled, returns 3 if RMT is turned on): 3

marcmerlin, 2017-04-04:
Hi @projectgus, since this is likely a platform problem (can't tell if it's arduino-esp32 or the chip underneath), is there someone this could get assigned to?

projectgus, 2017-04-04:
Are you able to reproduce this on non-IoTuZ hardware?

marcmerlin, 2017-04-04:
I will give that a shot since I now have some, and get back to you.

marcmerlin, 2017-04-04:
I was able to reproduce on WROVER, but I then updated arduino-esp32 again, and as of
commit 2b075f320477278be0eec55a34de9eca19426019
Author: Me No Dev <me-no-dev@users.noreply.github.com>
Date:   Tue Apr 4 02:50:27 2017 +0300
the problem is fixed.
I can now use the RMT neopixel library MartyMacGyver/ESP32-Digital-RGB-LED-Drivers
and analogread now returns proper enough looking values
I'm going to mark this closed, thanks @me-no-dev for the newer arduino-esp32

closed 2017-04-04 by marcmerlin