espressif/arduino-esp32 #292: How to read analog pin?

szerwi, 2017-03-29:
Hi,
How to read from analog pins in ESP32? What should I type in analogRead()? What pin?

me-no-dev, 2017-03-29:
the pin that you want to read. pins that support analog are shown in the image in this repository's README

closed 2017-03-29 by me-no-dev

szerwi, 2017-03-29:
I was trying but when I connected some resistors I had only value 0 or 4056.

me-no-dev, 2017-03-29:
what did you try? which pin? show some code :)

szerwi, 2017-03-30:
float odczyt;

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);

}

void loop() 
{
  odczyt = analogRead(A4);
  Serial.println(odczyt);
  delay(1000);
}

If I give "4" instead of "A4" it's writing 0 on all pins.

copercini, 2017-03-30:
Your wire is connected in GPIO32 or GPIO04 for use "A4"?

szerwi, 2017-03-30:
32 and it's giving only 0 or 4095 value.

copercini, 2017-03-30:
What is your circuit?

szerwi, 2017-03-30:
2 pins from potentiometer are connected first to Vcc second to GND and middle pin to analog pin. I'm checking the same with Arduino and value is only 0 or 1023 so I connected something wrong :/

szerwi, 2017-03-30:
Oh, it's working mow! :D Before I had only 1 pin to Vcc and another to analog pin. When I connected one to Vcc, second to GND and third to analog it's working :D

copercini, 2017-03-30:
=)