espressif/arduino-esp32 #293: ESP32 OTA

szerwi, 2017-03-30:
Hi,

I don't know how to program ESP32 over the air. There is example in ESP32 library, but how to upload code to ESP32 using Arduino IDE?

me-no-dev, 2017-03-30:
How have you tried? You select the device in the ports drop down (Arduino board menu) and hit upload. https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#application-example
Points 4-6

closed 2017-03-30 by me-no-dev

szerwi, 2017-03-30:
It's so complicated :D I've uploaded code to ESP via cable and I will try now by WiFi :D Thanks for help :D

me-no-dev, 2017-03-30:
complicated? really?

szerwi, 2017-03-30:
Oh, There are 3 ways to do that. But option with Arduino IDE is for ESP8266. Will it work with ESP32?

me-no-dev, 2017-03-30:
@szerwi it was to show you where to select the board. yes it's the same on esp8266. Have you hit upload on the OTA example yet? If not then do so. Then restart the IDE (to find the board faster) and select it in the Ports menu (as shown in the esp8266 page I gave you). Done

szerwi, 2017-03-30:
It's working. Thank you very much ;)