espressif/arduino-esp32 #294: Blocking UART RX

agcaahmet, 2017-04-03:
I have problems reading pulse widths of receiver PWM signals (with simple interrupt functions) if "Serial" has begun and data is coming into Rx port. It does not matter you are reading or not, uart rx interrupt blocks the processor if there is data coming.
Since reading pwm signal has to be accurate up to microseconds, it may not be possible to measure correctly during serial rx data existance on line. Is there a proper way to use both serial communication and my pwm interrupt functions to get accurate pulse widths.

igrr, 2017-04-03:
Take a look at the RMT peripheral. It captures pulse lengths and stores them into a block of RAM. Everything happens in hardware so interrupts are not an issue.

agcaahmet, 2017-04-03:
@igrr Thats good news for me. I wasn't aware of RMT peripheral, and haven't encountered such example. Is there any reference guide or example sketch on how to use related functions?

closed 2017-05-19 by me-no-dev