espressif/arduino-esp32 #295: WiFiUDP crash during destructor

martinayotte, 2017-04-03:
/home/martin/git-work/esp-idf/components/freertos/./heap_regions.c:368 (vPortFreeTagged)- assert failed!
abort() was called at PC 0x4008389b
0x4008389b: vPortFreeTagged at /home/martin/git-work/esp-idf/components/freertos/./heap_regions.c:410
Guru Meditation Error: Core  1 panic'ed (abort)
Backtrace: 0x40009203:0x3ffcdfe0 0x40085108:0x3ffce000 0x4008673c:0x3ffce020 0x40081c28:0x3ffce040 0x4000beca:0x3ffce060 0x4010bf6c:0x3ffce080 0x4010bfac:0x3ffce0a0 0x40111085:0x3ffce0c0 0x401110e3:0x3ffce0f0 0x4010e56c:0x3ffce110 0x4010e6da:0x3ffce190 0x4010ea15:0x3ffce230 0x4010bfcf:0x3ffce270
0x40085108: prvIdleTask at /home/martin/git-work/esp-idf/components/freertos/./tasks.c:4421
0x4008673c: vPortFree at /home/martin/git-work/esp-idf/components/esp32/./heap_alloc_caps.c:414
0x40081c28: _free_r at /home/martin/git-work/esp-idf/components/newlib/./syscalls.c:32
0x4010bf6c: _ZdlPv at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
0x4010bfac: _ZdaPv at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_opv.cc:32
0x40111085: _ZN7WiFiUDP4stopEv at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/components/arduino/libraries/WiFi/src/WiFiUdp.cpp:128
0x401110e3: _ZN9PrintableD4Ev at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/components/arduino/libraries/WiFi/src/WiFiUdp.cpp:128
 (inlined by) _ZN9IPAddressD4Ev at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/components/arduino/cores/esp32/IPAddress.h:52
 (inlined by) _ZN7WiFiUDPD2Ev at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/components/arduino/libraries/WiFi/src/WiFiUdp.cpp:36
0x4010e56c: _Z10NTPRequestR5Print at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/main/./main.cpp:238
0x4010e6da: _Z12parseCommandR5PrintPKc at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/main/./main.cpp:391
0x4010ea15: _Z4loopv at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/main/./main.cpp:472
0x4010bfcf: _Z8loopTaskPv at /home/martin/git-work/esp-idf/examples/arduino/ntp_ota_minimal/components/arduino/cores/esp32/main.cpp:18 (discriminator 1)

The above is been seen with a sketch compiled with Arduino as a ESP-IDF component using a minimal sketch doing an NTP request.

martinayotte, 2017-04-03:
Here is the sketch, a zipped TGZ since github refuse TGZ attachement.

[ntp_ota_minimal.zip](https://github.com/espressif/arduino-esp32/files/890234/ntp_ota_minimal.zip)

martinayotte, 2017-04-03:
Oh ! 
Simply adding "tx_buffer = NULL;", after the line [https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiUdp.cpp#L101](url) is dixing the crash. Still, I don't understand why it wasn't crashing with plain Arduino, since it is the same code

me-no-dev, 2017-04-03:
fixed in https://github.com/espressif/arduino-esp32/commit/7e45310771a43b9a47ec0c4556ad945176d4556b :) thanks

closed 2017-04-03 by me-no-dev