espressif/arduino-esp32 #296: Fatal error: SoftwareSerial.h: No such file or directory

tobozo, 2017-04-04:
Just found out this and it seems to compile OK https://github.com/jdollar/espsoftwareserial/

me-no-dev, 2017-04-04:
Why do you file this as an issue? And are 3 hardware serials not enough? No SoftSerial that uses delays could be any good on ESP32 ;)

closed 2017-04-04 by me-no-dev

tobozo, 2017-04-04:
Because backwards compatibility with ESP8266 is sometimes useful whee reusing old code (lazy me :p).

Googling a bit made me find out [I'm not alone](https://www.reddit.com/r/hackerboxes/comments/5xba6g/softwareserial_working_for_esp32arduino_ide/) and also that it's [somehow official](http://esp8266.github.io/Arduino/versions/2.2.0/doc/libraries.html#softwareserial).

I know it's not ideal to use it but it seems good enough to compile and do simple things like talk to a GPS module that used to be plugged to an Arduino Pro Mini, then to an ESP8266, then to an ESP32, without changing any line of code.

me-no-dev, 2017-04-04:
SoftwareSerial exists because some systems do not have enough serial ports. That is not the case with ESP32. People should detect ESP32 and ifdef one line. All those SoftwareSerial libs are no good on RTOS. There is other hardware that can be used to implement SoftwareSerial and that is something I would consider including.

marcmerlin, 2017-04-17:
@tobozo to echo what @me-no-dev is saying softserial is so timing critical and brittle that I had it break just by upgrading my compiler to another one that made more efficient (and faster) code. Bit banging of time critical stuff on ESP32 just isn't a good idea. I think it could be made to work for Softserial given that the bit rate is low, but the incentive to do this the wrong way when there are multiple UARTs with perfect timing that don't tie up the CPU, is kind of low :)

tobozo, 2017-04-17:
Thanks for the insight, I finally ended up using HardwarsSerial although it does not share the same signature as the other versions.

hinjo, 2021-11-25:
> Thanks for the insight, I finally ended up using HardwarsSerial although it does not share the same signature as the other versions.

Hi, I realize this is an old post but maybe I'get lucky :). I'm curious as how you got this working with HardwareSerial, is this a library, how did you get a hold of it and integrate this with libraries that otherwise except a SoftwareSerial object?